AllskyTeam/allsky #3512: Crop user interface

Alex-developer, 2024-03-17:
Add UI for managing image cropping

deece, 2024-11-01:
[allsky_crop_preview.tar.gz](https://github.com/user-attachments/files/17598665/allsky_crop_preview.tar.gz)
While this is being developed, here is a terrible python script that lets you visualise the crops and offsets.

To use it:
```
tar zxf allsky_crop_preview.tar.gz
cd allsky_crop_preview
python3 -m venv venv
. venv/bin/activate
pip install -r requirements.txt
./crop_preview.py <image file>
```

It's slow and clunky, as it shells out to convert to perform the same operations that save_image.sh does, but it's better than trial and error.

EricClaeys, 2024-11-01:
@deece, cool, thanks for doing it.
Can you modify it to use the initial width and height as the max Crop Width and Crop Height?  i.e., don't allow enlarging the image.
I can't enter a number into a field - it changes it to the max.  It's processing every keystroke, rather than waiting until I'm done entering a number.  Can an "Update" button be added that causes the new crop to happen?

There is code to display a "clock", but I don't see it.

FYI, in the next Allsky release the crop values are how many pixels to crop from the top, right, bottom, and left.  Offsets are gone.  It would be nice to change your tool accordingly.

@Alex-developer, this would be a nice addition; weren't you planning to have a "Crop" tab in the Overlay Editor?  A similar "Resize" tool/tab would be nice, or maybe combine the two.

EricClaeys, 2024-11-01:
@deece, it would really be cool to draw a box around the crop area, and have the box resize while the picture remains constant-sized.

Alex-developer, 2024-11-01:
This looks cool. When I am at my laptop I will take a look.

The new webui has a crop and resize tool in it

deece, 2024-11-01:
It's just a dirty hack to use until the new webui is released, I wasn't really intending it to grow, tweaked, or be maintained, and certainly not committed to the repo 😂

Alex-developer, 2024-11-02:
Just had a play and its really nice. I have implemented it a slightly different way but some of your code will be handy. If you dont mind I will use some of your code and add a credit to Allsky for using your code

deece, 2024-11-02:
Sure, not a problem :)

closed 2024-12-15 by Alex-developer