AllskyTeam/allsky #3519: [BUG] allsky service not starting up

WaldperlachFabi, 2024-03-23:
Cam: RPI HQ
PI: Pi 4 4GB
OS: 12.5
Version: v2023.05.01_04

Description:
No Images are taken, but there should be images taken. I think the background service doesnt start up correctly because when i do systemctl status allsky i get an error (attached).
The Web interface is avaible but there arent any photos taken.
The log file is emty to, even if it is set to 4.
Taking images with the libcamera-still commands works well.
config.sh as well as variablees.sh is attacked too 

[log.txt](https://github.com/AllskyTeam/allsky/files/14733737/log.txt)
[variables.sh.txt](https://github.com/AllskyTeam/allsky/files/14733738/variables.sh.txt)

EricClaeys, 2024-03-23:
Please attach `~/allsky/config/logs/install.sh.log`.
My guess is something during installation failed.

WaldperlachFabi, 2024-03-31:
yep, was an issue with the installation

closed 2024-03-31 by WaldperlachFabi

EricClaeys, 2024-03-31:
Would you please attach the file I asked about?  I may be able to keep the problem from happening in the future.