AllskyTeam/allsky #3542: [QUESTION] How to change WiFi network??

ThomasW69, 2024-03-31:
### Environment

* Camera:  RPi - HQ
* OS: Buster
* Allsky version:  v2023.05.01_04
* Pi model4  4 GB)

### Bug Description

I was able to install Allsky at home on the Raspi without any problems. To do this, I set up a wifi connection to my wifi network (SSID: TWPRIVAT5). Then I wanted to install the camera in my observatory and connect it to the Wifi there (SSID: WAVLINK_N). I did not succeed! All available Wifi networks and their SSID are displayed, and I can also enter the corresponding passwords in the "Configure-Wifi" dialog, but the Raspi does not establish a connection. Even editing the wpa-supplicant.conf does not change anything. 
It is also interesting that the Allsky camera dialed back into my old WiFi network when I returned home, even though I deleted it in the "Configure-Wifi" dialog. The WiFi passphrase no longer appears there either. Nevertheless, a connection is established. I cannot even switch to other WiFi networks at home. It seems as if my SSID "TWPRIVAT5" is permanently and unchangeably burned into the system.

So how can I change a WiFi network? 

### Log / configuration files

[image]
[image]

lignodelago, 2024-04-23:
Hi ThomasW69

I´m just new to Raspberry and Allsky. I run the Allsky software on a 64-Bit-Bookworm-OS on a Raspberry Zero 2 W (Timelapse, Startrails and Keogram I disabled due to performance issues). I set (one after another) different WIFI credentials within raspi-config connected with ssh to the raspi instead of using the Allsky-web-interface. No problems so far. May you try this? Be carefull what you are typing, if you miss the right credencial ... everythings lost;-)

login ssh user@ipaddress + password
sudo raspi-config
option 1 System Options
option S1 Wireless LAN

EricClaeys, 2024-04-24:
@ThomasW69,  there's a bug in the WebUI interface so I suggest not using it.
You should also be able to click on the Wi-Fi icon on the taskbar to connect to an SSID:
[image]

FYI, the next release of Allsky will be the last one to support Buster, so you may want to consider using Bookworm 64-bit.  It also allows you to configure Wi-Fi via the icon on the taskbar.

closed 2024-05-31 by EricClaeys