AllskyTeam/allsky #3570: [BUG] Pi HQ Cam binned causes keogram and startrails generation to fail

bendenajones, 2024-04-16:
Pi HQ Camera 4056x3040 resolution; binned is 2028x1520

So I have the PiCapture.cpp "fix" in for binning and since enabled binning 2x2 support for both Day and Night Keogram and Startrails fail. Timelapses (mini and full) generate ok.

In the log file:
[image]

I see the cmd line to generate the said outputs are specifying the unbinned resolution. Me thinks this is the issue.

EricClaeys, 2024-04-16:
Do you specify the Image Width and Height in the WebUI?  If so, set them to 0.  If they are both non-zero then the `-s` option is added to the command line, and any image not that size is rejected.
Also, FYI, the keogram and startrails commands don't like having images of multiple sizes.

EricClaeys, 2024-04-16:
Are you aware of the `generateForDay.sh` script that can be used to manually create a timelapse, keogram, or startrails?

bendenajones, 2024-04-16:
Crap! I left those set from the binn testing. Reverted them to 0.

I can just run those scripts w/o inputs?

bendenajones, 2024-04-16:
Ok the generateForDay script is working correctly and thus my reported issue is not an issue. thanks.

EricClaeys, 2024-04-16:
Run `generateForDay.sh  --help` for a list of options.
Glad you figured things out.