AllskyTeam/allsky #3611: [ERROR] Fresh install. Python Dependencies [photutils]

mrgrosser, 2024-05-09:
*** ERROR: 
**********
Python_dependencies [photutils] failed
**********

* The full log file is in /home/user/allsky/config/logs/Python_dependencies.10.log
The end of the file is:
        Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
        Collecting photutils
          Downloading photutils-1.12.0.tar.gz (1.2 MB)
             ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.2/1.2 MB 5.4 MB/s eta 0:00:00
        ERROR: photutils from https://files.pythonhosted.org/packages/10/b6/2ecd1ddebf269aa78103959a99ebb2c2ca9070f392cf10ac767fc4176b2a/photutils-1.12.0.tar.gz (from -r /tmp/package (line 1)) does not appear to be a Python project: neither 'setup.py' nor 'pyproject.toml' found.

CabinHal, 2024-05-12:
I have just encountered this identical error.

drakula1981, 2024-05-12:
Sadly, I have exactly the same problem with a brand new install

morrowwm, 2024-05-12:
Same error on a fresh install of allsky on top of a brand new debian 12.5 on a 1GB RAM Raspberry Pi 3. 

I ventured over to https://photutils.readthedocs.io/en/stable/install.html to try building photutils from source. But that also failed trying to build rasterio. 

I'll keep poking at it.

mrgrosser, 2024-05-12:
Not ideal, but I was able to install by making sure pip was installed, then did 'pip install photutils'. Once that was done, I re-ran the installer and it went through ok. Definitely not a solution though.

mrgrosser, 2024-05-12:
> > Not ideal, but I was able to install by making sure pip was installed, then did 'pip install photutils'. Once that was done, I re-ran the installer and it went through ok. Definitely not a solution though.
> 
> Did you do that inside the allsky venv?

I SSH'd onto the Raspberry Pi. Did 'sudo apt install pip' and then the pip command to install photutils.

Alex-developer, 2024-05-13:
Can you please try the installation now. I have removed the requirement for photutils as its not actually used,

morrowwm, 2024-05-13:
> Can you please try the installation now. I have removed the requirement for photutils as its not actually used,

Yesterday, I removed it from requirements.txt myself and the installation succeeded. I hope to be doing another installation later this week, so will try from bare silicon then.

CabinHal, 2024-05-13:
I did complete the installation this afternoon without any problem. Thanks so very much for your prompt action. What a great service you do for the community.

   Hal McAlister

CAUTION: This email was sent from someone outside of the university. Do not click links or open attachments unless you recognize the sender and know the content is safe.

EricClaeys, 2024-05-19:
Closing - issue resolved in current release and next release.

closed 2024-05-19 by EricClaeys