AllskyTeam/allsky #3622: Transition Auto Stretch Amount

EricClaeys, 2024-05-11:
### Discussed in https://github.com/AllskyTeam/allsky/discussions/2942

<div type='discussions-op-text'>

<sup>Originally posted by **AlaskanAstro** August 29, 2023</sup>
Perhaps I have missed it but it would be nice to either offset the time of day that the auto stretch kicks in or have it ramp the amount, similar to how gain works for a day and night value. 
Currently if a time lapse starts fairly early in the evening the initial auto stretch may be too aggressive for the first hour or so. 

Thanks! Currently running the _03 release updated last night so auto stretch does actually work. </div>

AlaskanAstro, 2024-05-11:
I'm realizing that the original question about ramping the STRETCH value from the config file through sunrise and sunset is different than what a true "auto stretch" would be. 

So the idea of "auto stretch" is something that changes the stretch amount and/or midpoint to the raw image to get a target median brightness in the saved jpeg. It would help preserve the dynamic range of the image by not blowing out highlights but still bringing up the dark area of the histogram. Exactly the same idea as the built in STRETCH settings in the config, but adjusting on the fly as needed. For instance the desired stretch amount is rarely the same on a full moon night as a dark winter night with barely any milky way. Similar to what PixInsight does with the screen transfer function. Basically the STRETCH value is not usually a one-size-fits-all throughout the month even with the auto-exposure working correctly. I hope that helps clear up the idea? I probably rambled too long.

EricClaeys, 2024-05-11:
That makes sense, thanks. Do you know if the `convert` command can do an auto stretch?

AlaskanAstro, 2024-05-12:
I really have no idea, sorry. The biggest difference I see is that the currently implemented setting called "auto stretch" is essentially a fixed value, not an automatic recalculation of what the stretch should be.