AllskyTeam/allsky #3632: RPi HQ Camera - "No camera detected" during installation [BUG]

seafurymike, 2024-05-15:
### Environment

* Camera:  RPi HQ
* OS: Bookworm 64bit
* Allsky version:v2024.05.01_04
* Pi model and amount of memory (Version 4B, 4GB)

### Bug Description
Installation fails, the installer sees the camera and then fails later to load it

xxxx@AllSkyCam:~/allsky $ ./install.sh
* Continuing installation.  Steps already performed will be skipped.
   The last status was: Error encountered - No camera detected

**************************************************************
*** Welcome back to the Allsky Installer - v2023.05.01_04! ***
**************************************************************

* RPi camera found.

*** NOTICE: 
**********
The following steps can take up to an hour depending on the speed of your Pi
and how many of the necessary dependencies are already installed.
You will see progress messages throughout the process.
At the end you will be prompted again for additional steps.
**********

* Setting up WebUI options and default settings for RPi cameras.
* Making new settings file '/home/seafurymike/allsky/config/settings.json'.

*** ERROR: 
**********
No camera detected - installation aborted.
**********

The camera is present and the OS sees it::

xxxx@AllSkyCam:~/allsky $ libcamera-hello --list-cameras
Available cameras
-----------------
0 : imx296 [1456x1088 10-bit RGGB] (/base/soc/i2c0mux/i2c@1/imx296@1a)
    Modes: 'SRGGB10_CSI2P' : 1456x1088 [60.38 fps - (0, 0)/1456x1088 crop]

### Log / configuration files

Follow the instructions for Reporting Issues in the Wiki, then ATTACH a copy of the file(s) above.
Do NOT copy/paste them into this Issue.

EricClaeys, 2024-05-15:
@seafurymike,
Allsky doesn't support the imx296 sensor.  The next release will provide a message saying that versus saying "camera not found".
Please run the following:
~~~
libcamera-still --list-cameras -v
libcamera-still -v --nopreview --timeout 1 --metadata - --shutter 1 --output /dev/null
~~~
and post the output.  Also, include a URL for camera information and I can look into supporting it in the next release.

seafurymike, 2024-05-15:
Here is the output requested::

AllSkyCam:~ $ libcamera-still --list-cameras -v
Available cameras
-----------------
0 : imx296 [1456x1088 10-bit RGGB] (/base/soc/i2c0mux/i2c@1/imx296@1a)
    Modes: 'SRGGB10_CSI2P' : 1456x1088 [60.38 fps - (0, 0)/1456x1088 crop]

    Available controls for 1456x1088 SRGGB10_CSI2P mode:
    ----------------------------------------------------
    AeConstraintMode : [0..3]
    AeEnable : [false..true]
    AeExposureMode : [0..3]
    AeFlickerMode : [0..1]
    AeFlickerPeriod : [100..1000000]
    AeMeteringMode : [0..3]
    AnalogueGain : [1.000000..251.188644]
    AwbEnable : [false..true]
    AwbMode : [0..7]
    Brightness : [-1.000000..1.000000]
    ColourGains : [0.000000..32.000000]
    Contrast : [0.000000..32.000000]
    ExposureTime : [29..0]
    ExposureValue : [-8.000000..8.000000]
    FrameDurationLimits : [16562..15534444]
    HdrMode : [0..4]
    NoiseReductionMode : [0..4]
    Saturation : [0.000000..32.000000]
    ScalerCrop : [(0, 0)/64x64..(0, 0)/1456x1088]
    Sharpness : [0.000000..16.000000]
    StatsOutputEnable : [false..true]

AllSkyCam:~ $ libcamera-still -v --nopreview --timeout 1 --metadata - --shutter 1 --output /dev/null
[0:01:43.688918622] [2074]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+120-eb00c13d
[0:01:43.729296168] [2077]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:01:43.731446592] [2077]  INFO RPI vc4.cpp:446 Registered camera /base/soc/i2c0mux/i2c@1/imx296@1a to Unicam device /dev/media1 and ISP device /dev/media0
[0:01:43.731544352] [2077]  INFO RPI pipeline_base.cpp:1102 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Options:
    verbose: 2
    info_text:
    timeout: 1ms
    width: 0
    height: 0
    output: /dev/null
    post_process_file: 
    preview: none
    qt-preview: 0
    transform: identity
    roi: all
    shutter: 1us
    metering: centre
    exposure: normal
    ev: 0
    awb: auto
    flush: false
    wrap: 0
    brightness: 0
    contrast: 1
    saturation: 1
    sharpness: 1
    framerate: 30
    denoise: auto
    viewfinder-width: 0
    viewfinder-height: 0
    tuning-file: (libcamera)
    lores-width: 0
    lores-height: 0
    autofocus-range: normal
    autofocus-speed: normal
    autofocus-window: all
[... 100 lines cut ...]
JPEG size is 183257
EXIF data len 298
Saved image 1456 x 1088 to file /dev/null
{
    "ExposureTime": 29,
    "AnalogueGain": 11.885022,
    "SensorBlackLevels": [ 3840, 3840, 3840, 3840 ],
    "FrameDuration": 16577,
    "Lux": 321.316742,
    "AeLocked": true,
    "DigitalGain": 1.009674,
    "ColourGains": [ 2.563697, 2.082299 ],
    "ColourTemperature": 4000,
    "FocusFoM": 213,
    "ColourCorrectionMatrix": [ 2.045146, -0.739536, -0.305616, -0.471715, 2.133571, -0.661861, -0.108284, -0.576562, 1.684846 ],
    "ScalerCrop": "(0, 0)/1456x1088",
    "SensorTimestamp": 104709006000
}Closing RPiCam application(frames displayed 0, dropped 0)
Camera stopped!
Tearing down requests, buffers and configuration
Camera closed

The following is a link to the datasheet for the camera I have connected::

https://www.farnell.com/datasheets/3929135.pdf

EricClaeys, 2024-05-15:
Thanks.  I think you are going to find it doesn't make a very good allsky camera since it is limited to 15 second exposures.

seafurymike, 2024-05-15:
Damn, didnt realise. Thought using the HQ camera would be a good option. Will have to rethink what i should do for a camera then. Is there any details on good options and potentially outputs from those options?

EricClaeys, 2024-05-16:
I always recommend the HQ camera.  It's a good one and a lot less expensive than ZWO.  And you can get lenses that give 180 FOV.  T Module 3 and other cameras with an attached lens are 80 - 120 FOV which might be fine if you can't see a lot of the sky.

EricClaeys, 2024-05-31:
Closing Issue.
Will not support - this sensor has a short maximum exposure so will not work as an allsky camera.

closed 2024-05-31 by EricClaeys

Polar-Axis, 2024-07-24:
I swapped from the HQ IMX477 camera to the IMX296, for a sky plate solving application, and the latter is superior for sensitivity due to the much larger pixel size. Stars are bright with just 1s exposure. I would think supporting any RPi compatible camera should be possible without too much effort?

Polar-Axis, 2024-07-24:
Aha! Excellent, I see other cameras will be supported in a future release. I have installed branch 2024.xx.xx and it recognises other cameras. Thanks!

EricClaeys, 2024-07-24:
v2024.xx.xx doesn't support the imx296 though.  Is a 15 second Max exposure good enough?

Polar-Axis, 2024-07-24:
I think 15s should be good enough for the imx296.

I've tried the IMX219 briefly with v2024.xx.xx, it shows brighter stars (Big Dipper, Lyra etc) but it a bit marginal for use because at night with high gain and 11.76s exposure it is quite noisy. I've swaped back to the IMX477 HQ camera temporarily now, but I want to use that camera for another purpose eventually.

I have another IMX296 on order and it should be with me in a few days time.  I have one working well with this software:
https://github.com/githubdoe/skysolve
With that software the 1s exposures show a significant number of stars (16mm f1.2 lens) so I think that camera sensor will work well in an allsky application too (if high resolution is not needed as it is only 1.6Mpix).

Don't add that sesnor just for me though, I could just get another HQ camera which seems to work well despite the small pixel size.

EricClaeys, 2024-07-24:
@Polar-Axis, Alan,
Thanks for the info.  I've spent more time telling people that the 296 probably won't make a good camera than it would have taken me to add it.
Would you please run `allsky-config` with the 296 connected and choose the last option?  Then upload the file it creates.  And let me know how that went - allsky-config is new and we're hoping to add a lot more to it going forward.

Thanks - Eric

Polar-Axis, 2024-07-24:
Great, will do as soon as the camera arrives.