AllskyTeam/allsky #3633: [BUG] Maybe its a BUG  Version: v2023.05.01_02 installed but [URL]/allsky gives 404

WheresWally, 2024-05-15:
Hopefully this will be easy
I want to put my camera on the Web page for all to enjoy
BUT the [URL]/allsky doesnt work just gives a 404
[URL]/public.php works fine but doesnt show all the timelapse/video nice links
What is the correct url and extension to use
or do I need to configure these  links somehow/
Help
WAlly

# Is this a QUESTION or FEATURE REQUEST?

If so, please add a Discussion item by clicking on the Discussions link above;  do NOT create an Issue. 

&nbsp; 
## Look in the Wiki before submitting this Issue
The Wiki has lots of information including troubleshooting tips for the vast majority of known issues.

 &nbsp; 

### Environment

* Camera:  ZWO  or  RPi - include the model
* OS: Buster or Bullseye
* Allsky version:
     * The newest software includes a file called  ~/allsky/version
     * If it does not exist, run:       grep Software /var/log/allsky.log
* Pi model and amount of memory (512 MB, 1, 2, 4, or 8 GB)

### Bug Description

* Include a clear and concise description of what the bug is.
* Can it be reproduced?  If so, how?
* Did anything change?  Any settings?  Most issues occur after a change was made.
* Please include the exact messages or a screenshot.
* If this is a configuration-related issue (or you are not sure), attach these files, appending ".txt" to their names first:
  * Allsky problems:
    * ~/allsky/config/config.sh
    * ~/allsky/config/settings.json
    * ~/allsky/config/ftp-settings.sh (only for upload-related problems)
  * Allsky Website problems:
    * ~/allsky/html/allsky/configuration.json (local Website)
    * ~/allsky/config/remote_configuration.json (remote Website)

If you have program output or multi-line messages to include,
add it like this so it formats correctly (note the lines with tildas):

~~~
output/message line 1
output/message line 2
~~~

### Log / configuration files

EricClaeys, 2024-05-15:
@WheresWally,
Is this on your Pi or a remote server?  And are you trying to display the Allsky Website or the WebUI, and on a local network or on the Internet?
public.html is part of the WebUI. You don't want to make the full WebUI available to anyone unless it's on a private network and you trust everyone on the network.

WheresWally, 2024-05-15:
I want the URL I should use for the public facing link on the MAP.
I an setting up the camera you should now see as a blue marker in Perth Australia.
problem is https://perthallsky.net/allsky  gives me 404 not found.
it is on a RPi 4 and I want to show the correct page that allows the public to see the time lapses and keograms etc.
I just maybe skipped something in the install when I did it a few months ago.
thanks for the help

EricClaeys, 2024-05-16:
I suggest not making the WebUI available on the Internet.  If someone guesses your login/password or otherwise hacks into the page (which wasn't made to be on the Internet) they can cause problems. 

The AllSky Website is what 99.9% of the people make available on the Internet and on the Allsky Map. It uses the [URL]/allsky link on the Pi, and whatever link you set up on a remote server.

If you install the Allsky Website on your Pi people will be able to see what you want.  What mist people do is install it on a remote server that has Internet access.  Either way, see the documentation page for Installing the Allsky Website.

WheresWally, 2024-05-16:
>I suggest not making the WebUI available on the Internet. 
Totally agree.

OK so why cant I see anything when I try https://perthallsky.net/allsky 
?
Or how do I fix that?
Wally

EricClaeys, 2024-05-16:
Does https://perthallsky.net/ point to your Pi?  If so, is the Allsky Website installed on your Pi?
Eric

WheresWally, 2024-05-16:
yes it does point to the RPi
And I thought it was.
Should I try reinstalling allsky website?
Wally

WheresWally, 2024-05-16:
ah it seems I never run 
website/install.sh

it would appear to be working now
duh
thanks
Wally

EricClaeys, 2024-05-31:
Issue resolved, closing.

closed 2024-05-31 by EricClaeys