AllskyTeam/allsky #3650: Split DAYS_TO_KEEP for images and videos

EricClaeys, 2024-05-21:
### Discussed in https://github.com/AllskyTeam/allsky/discussions/3406

<div type='discussions-op-text'>

<sup>Originally posted by **schoerg** February  2, 2024</sup>
Hi,
example: on the night of 2024-01-30 the camera took 2465 images, which results in 16.9GB of used space. The created video takes only 100 MB. While it's not feasible to keep every image for all eternity, 100 MB don't hurt overall space consumption. If you really want you could extract the single frames of the video.

Feature: split `DAYS_TO_KEEP` for images and videos, maybe this:
`DAYS_TO_KEEP_IMAGES=14` which would delete images after 14 days
`DAYS_TO_KEEP_VIDEOS=30` which would delete videos after 30 days.
</div>

EricClaeys, 2026-09-30:
@schoerg,
Is this feature still needed?  No one else has expressed interest.

Eric

schoerg, 2026-10-04:
No hard feelings if this doesn't get implemented.

closed 2026-10-04 by EricClaeys