AllskyTeam/allsky #3657: [BUG] timeout 1 argument to libcamera-still broken in bookworm with rpicam-apps 1.4.4-1

MichelMoriniaux, 2024-05-26:
### Environment

* Camera:  RPi - HQ ( Arducam Imx477 )
* OS: Bookworm
* Allsky version: v2023.05.01_04
* Pi model 4b 8G

### Bug Description
with rpicam-apps 1.4.4.1 ( current version for Bookworm) _libcamera-still --timeout 1_ forces the exposure time to **1us** regardless of the _--shutter_ setting. This creates a very noisy, underexposed Daytime or Manual image. it seems that the _--immediate_ flag provides the desired result

time libcamera-still --shutter 60000000 --timeout 1 completes in 1s
time libcamera-still --shutter 60000000 --immediate completes in 61s

side effect:
even though the image is taken in 1us, allsky thinks it was 60s so there is an image taken every 30s instead of 90s

possible solutions:
- replace the timeout command in capture_RPi.cpp by --immediate
(do not change in scripts/fucntions.sh as it breaks command detection)

OR
- change the timeout flag logic to be slightly longer than the exposure in capture_RPi.cpp
 

### Log / configuration files
example log lines showing time discrepancies
2024-05-25T16:18:27.188947-07:00 skycam allsky[4885]: -----
2024-05-25T16:18:27.189394-07:00 skycam allsky[4885]: STARTING EXPOSURE at: 2024-05-25 16:18:27   @ 60.0 sec
2024-05-25T16:18:**27**.229334-07:00 skycam allsky[4885]:  > Running: libcamera-still --output '/home/mark/allsky/tmp/image-20240525161827.jpg' --metadata /home/mark/allsky/config/overlay/extra/libcamera.txt --metadata-format txt **--timeout 1** --nopreview --width 4056 --height 3040 --shutter **60000000** --analoggain 1.45151 --awb auto --quality 100
2024-05-25T16:18:**30**.631629-07:00 skycam allsky[4885]:   > Saving DAY image 'image-20240525161827.jpg'
2024-05-25T16:18:30.633062-07:00 skycam allsky[4885]:   > Sleeping: 30.0 sec

after fix in captureRPi.cpp:
2024-05-25T18:36:10.537156-07:00 skycam allsky[10987]: -----
2024-05-25T18:36:10.538206-07:00 skycam allsky[10987]: STARTING EXPOSURE at: 2024-05-25 18:36:10   @ 1.3 sec
2024-05-25T18:36:**10**.637617-07:00 skycam allsky[10987]:  > Running: libcamera-still --output '/home/mark/allsky/tmp/image-20240525183610.jpg' --metadata /home/mark/allsky/config/overlay/extra/libcamera.txt --metadata-format txt **--immediate** --nopreview --width 4056 --height 3040 --shutter 1284414 --analoggain 1 --awb auto --quality 100
2024-05-25T18:36:**17**.919369-07:00 skycam allsky[10987]:   > Saving DAY image 'image-20240525183610.jpg'
2024-05-25T18:36:17.920713-07:00 skycam allsky[10987]:   > Sleeping: 88.7 sec

EricClaeys, 2024-06-17:
@MichelMoriniaux, is rpicam-apps 1.4.4-1 still the latest release?

aaronwmorris, 2024-08-16:
@EricClaeys v1.5.0 is the latest release on bookworm.

EricClaeys, 2024-08-19:
@MichelMoriniaux, @rkinnett, is this still a problem with v1.5.0 libcamera?

MichelMoriniaux, 2025-03-06:
this seems to be resolved, just built a new skycam for my club and there were no issues with stock software

closed 2025-03-06 by MichelMoriniaux

EricClaeys, 2025-03-06:
@MichelMoriniaux,
Thanks for the update.  That's good to hear.