AllskyTeam/allsky #3662: Les modules ne s'affichent plus dans l'onglet "module manager" du WEBUI

chris-astronome, 2024-05-29:
### Discussed in https://github.com/AllskyTeam/allsky/discussions/3656

<div type='discussions-op-text'>

<sup>Originally posted by **chris-astronome** May 24, 2024</sup>
Bonjour à tous
Suite à une mise à jour des modules, ceux ci ne s'affichent plus dans l'onglet "module manager" du WEBUI mais l'overlay fonctionne mis à part openweather qui s'est désactivé et je ne peux plus le réactiver.
quelqu'un à t'il une idée pour remettre tout ça d'aplomb.</div>

EricClaeys, 2024-05-29:
This looks to be identical to #3656.  If so, why did you create this discussion item?

chris-astronome, 2024-05-29:
Bonjour, je pensais que l'on pouvait déplacer une discussion qui reste sans réponse vers un problème.
J'ai hésité à l'origine poser le problème directement dans Problème.

EricClaeys, 2024-06-17:
Closing this Issue.  It's not clear yet if it's a bug in Allsky or a user configuration problem.
Will track progress in Discussion [3656)[https://github.com/AllskyTeam/allsky/discussions/3656].

closed 2024-06-17 by EricClaeys