AllskyTeam/allsky #3668: Module AllskyAI[BUG]

ettoreribaudo, 2024-06-02:
I write in module one problem but nothing response.
Last week it worked but now it always gives me fake but the photos are uploaded to the allskyAI.com website

I uninstalled and reinstalled the module but still false

Expected behaviour
A clear and concise description of what you expected to happen.

Screenshots
If applicable, add screenshots to help explain your problem.

Allsky Information:

OS: Bullseye
Allsky Version: last
Pi model: PI4 B 4 GB
Camera: imx462 PI
Sensors:
MLX90614
BME280
TLS2591
[image]

EricClaeys, 2024-06-02:
Please run ` grep ERROR /var/log/allsky.log` and let us know what you get.

ettoreribaudo, 2024-06-02:
Thanks Eric:
un  2 07:46:14 allsky allsky[16700]: ERROR: Field 'RPi imx290_60.00_fps' is outside of the image
Jun  2 07:46:14 allsky allsky[16700]: ERROR: Cannot use format ':n' on Number variables like ${CLOUDCOVER} (value=Partial).
Jun  2 07:46:34 allsky allsky[16828]: ERROR: Module allsky_ai.py failed on line 224 - invalid literal for int() with base 10: 'No version file found, probably not created yet'
Jun  2 07:46:34 allsky allsky[16828]: ERROR: Field 'RPi imx290_60.00_fps' is outside of the image
Jun  2 07:46:34 allsky allsky[16828]: ERROR: Cannot use format ':n' on Number variables like ${CLOUDCOVER} (value=Partial).
Jun  2 07:46:56 allsky allsky[16960]: ERROR: Module allsky_ai.py failed on line 224 - invalid literal for int() with base 10: 'No version file found, probably not created yet'
Jun  2 07:46:56 allsky allsky[16960]: ERROR: Field 'RPi imx290_60.00_fps' is outside of the image
Jun  2 07:46:56 allsky allsky[16960]: ERROR: Cannot use format ':n' on Number variables like ${CLOUDCOVER} (value=Partial).
Jun  2 07:47:17 allsky allsky[17093]: ERROR: Module allsky_ai.py failed on line 224 - invalid literal for int() with base 10: 'No version file found, probably not created yet'
Jun  2 07:47:17 allsky allsky[17093]: ERROR: Field 'RPi imx290_60.00_fps' is outside of the image
Jun  2 07:47:17 allsky allsky[17093]: ERROR: Cannot use format ':n' on Number variables like ${CLOUDCOVER} (value=Partial).
Jun  2 07:47:38 allsky allsky[17226]: ERROR: Module allsky_ai.py failed on line 224 - invalid literal for int() with base 10: 'No version file found, probably not created yet'
Jun  2 07:47:38 allsky allsky[17226]: ERROR: Field 'RPi imx290_60.00_fps' is outside of the image
Jun  2 07:47:38 allsky allsky[17226]: ERROR: Cannot use format ':n' on Number variables like ${CLOUDCOVER} (value=Partial).
Jun  2 07:47:59 allsky allsky[17351]: ERROR: Module allsky_ai.py failed on line 224 - invalid literal for int() with base 10: 'No version file found, probably not created yet'
Jun  2 07:47:59 allsky allsky[17351]: ERROR: Field 'RPi imx290_60.00_fps' is outside of the image
Jun  2 07:47:59 allsky allsky[17351]: ERROR: Cannot use format ':n' on Number variables like ${CLOUDCOVER} (value=Partial).
Jun  2 07:48:20 allsky allsky[17478]: ERROR: Module allsky_ai.py failed on line 224 - invalid literal for int() with base 10: 'No version file found, probably not created yet'
Jun  2 07:48:20 allsky allsky[17478]: ERROR: Field 'RPi imx290_60.00_fps' is outside of the image
Jun  2 07:48:20 allsky allsky[17478]: ERROR: Cannot use format ':n' on Number variables like ${CLOUDCOVER} (value=Partial).
Jun  2 07:48:41 allsky allsky[17597]: ERROR: Module allsky_ai.py failed on line 224 - invalid literal for int() with base 10: 'No version file found, probably not created yet'
Jun  2 07:48:41 allsky allsky[17597]: ERROR: Field 'RPi imx290_60.00_fps' is outside of the image
Jun  2 07:48:41 allsky allsky[17597]: ERROR: Cannot use format ':n' on Number variables like ${CLOUDCOVER} (value=Partial).
Jun  2 07:49:02 allsky allsky[17722]: ERROR: Module allsky_ai.py failed on line 224 - invalid literal for int() with base 10: 'No version file found, probably not created yet'
Jun  2 07:49:02 allsky allsky[17722]: ERROR: Field 'RPi imx290_60.00_fps' is outside of the image
Jun  2 07:49:02 allsky allsky[17722]: ERROR: Cannot use format ':n' on Number variables like ${CLOUDCOVER} (value=Partial).
Jun  2 07:49:23 allsky allsky[17855]: ERROR: Module allsky_ai.py failed on line 224 - invalid literal for int() with base 10: 'No version file found, probably not created yet'
Jun  2 07:49:23 allsky allsky[17855]: ERROR: Field 'RPi imx290_60.00_fps' is outside of the image
Jun  2 07:49:23 allsky allsky[17855]: ERROR: Cannot use format ':n' on Number variables like ${CLOUDCOVER} (value=Partial).
Jun  2 07:49:44 allsky allsky[17980]: ERROR: Module allsky_ai.py failed on line 224 - invalid literal for int() with base 10: 'No version file found, probably not created yet'
Jun  2 07:49:44 allsky allsky[17980]: ERROR: Field 'RPi imx290_60.00_fps' is outside of the image
Jun  2 07:49:44 allsky allsky[17980]: ERROR: Cannot use format ':n' on Number variables like ${CLOUDCOVER} (value=Partial).
Jun  2 07:50:06 allsky allsky[18113]: ERROR: Module allsky_ai.py failed on line 224 - invalid literal for int() with base 10: 'No version file found, probably not created yet'
Jun  2 07:50:06 allsky allsky[18113]: ERROR: Field 'RPi imx290_60.00_fps' is outside of the image
Jun  2 07:50:27 allsky allsky[18252]: ERROR: Module allsky_ai.py failed on line 224 - invalid literal for int() with base 10: 'No version file found, probably not created yet'
Jun  2 07:50:27 allsky allsky[18252]: ERROR: Field 'RPi imx290_60.00_fps' is outside of the image
Jun  2 07:50:48 allsky allsky[18385]: ERROR: Module allsky_ai.py failed on line 224 - invalid literal for int() with base 10: 'No version file found, probably not created yet'
Jun  2 07:50:48 allsky allsky[18385]: ERROR: Field 'RPi imx290_60.00_fps' is outside of the image
Jun  2 07:51:09 allsky allsky[18503]: ERROR: Module allsky_ai.py failed on line 224 - invalid literal for int() with base 10: 'No version file found, probably not created yet'
Jun  2 07:51:09 allsky allsky[18503]: ERROR: Field 'RPi imx290_60.00_fps' is outside of the image
Jun  2 07:51:30 allsky allsky[18642]: ERROR: Module allsky_ai.py failed on line 224 - invalid literal for int() with base 10: 'No version file found, probably not created yet'
[... cut ...]
Jun  2 10:54:30 allsky allsky[7288]: ERROR: Cannot use format ':n' on Number variables like ${CLOUDCOVER} (value=Cloudy).
Jun  2 10:54:51 allsky allsky[7414]: ERROR: Field 'RPi imx290_60.00_fps' is outside of the image
Jun  2 10:54:51 allsky allsky[7414]: ERROR: Cannot use format ':n' on Number variables like ${CLOUDCOVER} (value=Cloudy).
Jun  2 10:55:12 allsky allsky[7546]: ERROR: Field 'RPi imx290_60.00_fps' is outside of the image
Jun  2 10:55:12 allsky allsky[7546]: ERROR: Cannot use format ':n' on Number variables like ${CLOUDCOVER} (value=Cloudy).
grep: /var/log/allsky.log: il file binario corrisponde

EricClaeys, 2024-06-02:
For the first error, edit `~/allsky/config/overlay/config/overlay.json` and set the "x", "y", "tlx" and "tly" to 500 then reload the Overlay Editor and move the field to where you want.

For the second error fix the Format for that field.

For the 3rd error I suggest installing the newest allsky_ai module, and seeing if it gives the same ERROR.  If it does contact the module author Christian Kardach at https://www.allskyai.com

ettoreribaudo, 2024-06-02:
"label": "${CAMERA_TYPE} ${CAMERA_MODEL}",
            "x": 419,
            "y": 3002,
            "id": "oe-field-0",
            "fontsize": 44,
            "strokewidth": 0,
            "sample": "RPi HQ",
            "tlx": 10,
            "tly": 2980
This is in Overlay.json

i modify to:
"label": "${CAMERA_TYPE} ${CAMERA_MODEL}",
            "x": 500,
            "y": 500,
            "id": "oe-field-0",
            "fontsize": 44,
            "strokewidth": 0,
            "sample": "RPi HQ",
            "tlx": 500,
            "tly": 500
            
            Is correct?
            
            For second point i corrected n to text.
            Regards.
            Ettore

ettoreribaudo, 2024-06-02:
AllSkyAi is the last.
Ettore

ettoreribaudo, 2024-06-02:
I corrected 1 and 2, i send email to Christian.
Thanks.
Ettore

ettoreribaudo, 2024-06-02:
> For the first error, edit `~/allsky/config/overlay/config/overlay.json` and set the "x", "y", "tlx" and "tly" to 500 then reload the Overlay Editor and move the field to where you want.
>
> For the [...]

I resolved error, i upgrade pillow with: pip install --upgrade Pillow, are is ok
[image]

closed 2024-06-02 by ettoreribaudo

EricClaeys, 2024-06-02:
Did you let Christian know how you fixed the problem?  He'll probably need to change the module.

You made the overlay.json changes correctly.  Note the "y" values was 3002 which is wider than the imx290 sensor at 1920, which is why that field was off the image.

In your initial post you said you have a `imx462` camera but the log files say you have the `imx290 60.00 fps` which I believe is an Arducam model.  Why the difference?

ettoreribaudo, 2024-06-02:
Yes, I notified Christian as soon as I had solved it.
The model is IMX462 innomaker, you have to use it bullseye sees it as IMX290

EricClaeys, 2024-06-02:
Do you have a URL to the camera?

Would you please run `libcamera-still --list-cameras` ?

Thanks

ettoreribaudo, 2024-06-02:
libcamera-still --list-cameras
Available cameras
-----------------
0 : imx290 [1920x1080 12-bit RGGB] (/base/soc/i2c0mux/i2c@1/imx290@1a)
    Modes: 'SRGGB10_CSI2P' : 1280x720 [30.00 fps - (65535, 65535)/65535x65535 crop]
                             1920x1080 [30.00 fps - (65535, 65535)/65535x65535 crop]
           'SRGGB12_CSI2P' : 1280x720 [30.00 fps - (65535, 65535)/65535x65535 crop]
                             1920x1080 [30.00 fps - (65535, 65535)/65535x65535 crop]