AllskyTeam/allsky #3670: [BUG] No camera detected and scrip issue

YoShi-2019, 2024-06-04:
Hello,

first - thank you for putting this project together.

I was trying to install it on RPi 5 with bookworm 64bit. I have a ASI676MC from ZWO

The camera is working (I cheched on a different computer through the native software) but the install scrip cannot find it. I tried to run the code provided but I get an error. See screenshot below.

thank you for your help

[image]

EricClaeys, 2024-06-04:
The current release of Allsky doesn't support the newest ZWO cameras like yours.
You have two options:
1. Wait for the next Allsky release.  I hope to have a test version in a month or two.
2. Install the newest ZWO SDK.  Someone did this recently so you should be able to search the Issues and Discussions in GitHub to find the thread.  It may be closed.

spex357, 2024-06-12:
I'm getting the same with a PIHQ camera on a pi5 and Bookworm, it finds the camera and then doesn't after the settings file is created.

EricClaeys, 2024-06-12:
@spex357, please run `libcamera-still  --list-cameras`.  If that shows the camera but Allsky doesn't see it, then:
~~~
sudo systemctl stop allsky
sudo truncate  -s 0 /var/log/allsky.log 
# set Debug Level to 3 in the WebUI 
~~~
After Allsky fails, attach /var/log/allsky.log

spex357, 2024-06-13:
libcamera-still  --list-cameras
 : imx296 [1456x1088 10-bit] (/base/axi/pcie@120000/rp1/i2c@88000/imx296@1a)
    Modes: 'SBGGR10_CSI2P' : 1456x1088 [60.38 fps - (0, 0)/1456x1088 crop]
The webUI doesn't load I get "Camera Type' not defined in config.sh. Please update it."
allsky.log has nothing in it.
I put my camera in config.sh, CAMERA_MODEL="$(settings '. IMX477')" but the webui still says the same

EricClaeys, 2024-06-14:
@spex357, Brian, you don't have the RPi HQ camera - you have the **imx296** camera, which Allsky doesn't support and probably won't.  It has a maximum exposure time of about 15 seconds, which means it won't make a very good allsky camera.

The current release of Allsky doesn't handle unsupported cameras very well, as you noticed.  The next release will tell you during installation that the camera isn't supported.

EricClaeys, 2024-08-04:
Closing - original post was answered.

closed 2024-08-04 by EricClaeys