AllskyTeam/allsky #3679: [BUG] Allsky stopps completely working after creating timelapse

charira83, 2024-06-14:
### Environment

* Camera:  RPi with HQ camera
* OS: Bullseye
* Allsky version:  v2023.05.01_04
* Pi 4 with 2 GB 

### Bug Description

- Allsky stopps working after creating timelapse.
- Night images are captured, startrails, keograms and timelapse are created.
- As soon as timelapse is created (endOfNight.sh: ===== Timelapse complete shown in the log file), allsky stopps completely working. No more logs are created in the logfile. 
- The webserver runs fine all the time. The last image shown in the webserver is the one, that was taken, when allsky stopped working.
- I have to restart allsky via webserver and everything runs fine until the next morning, when timelapse is created.
- Debug Level 4 is running for a few days now.
- Daytime pictures are captured but not saved
- All Modules in module manager are disabled.
- Image overlays are disabled but the issue still remains

Here is an exmaple of the logfile:
At 04:38 it is capturing a daytime image. 
Next entry is "Timelapse complete" at 08:27. 
After  "INFO: nightday flow Complete..." allsky stopps working, no more log entries 
At 14:15:20 i restarted allsky manually via webbrowser

Jun 11 04:38:26 allsky-pi allsky[31134]: INFO: ----------------------- Running Module allsky_saveimage.py -----------------------
Jun 11 04:38:26 allsky-pi allsky[31134]: INFO: Attempting to load allsky_saveimage
Jun 11 04:38:26 allsky-pi allsky[31134]: INFO: Image /home/harira/allsky/tmp/image-20240611043813.jpg Saved, quality 95
Jun 11 04:38:26 allsky-pi allsky[31134]: INFO: Ignoring watchdog for module saveimage
Jun 11 04:38:26 allsky-pi allsky[31134]: INFO: day flow Complete...
Jun 11 04:38:28 allsky-pi allsky[20337]: Executing sunwait poll exit angle -6.0000 52N 6.91E > /dev/null
Jun 11 04:38:28 allsky-pi allsky[20337]: -----
Jun 11 04:38:28 allsky-pi allsky[20337]: STARTING EXPOSURE at: 2024-06-11 04:38:28   @ 2.8 sec
Jun 11 04:38:28 allsky-pi allsky[20337]:  > Running: libcamera-still --output '/home/harira/allsky/tmp/image-20240611043828.jpg' --timeout 1 --nopreview --width 4056 --height 3040 --shutter 2828427 --analoggain 1 --quality 100
Jun 11 08:27:51 allsky-pi allsky[29287]: endOfNight.sh: ===== Timelapse complete
Jun 11 08:27:52 allsky-pi allsky[3406]: INFO: Loading config /home/harira/allsky/config/settings.json
Jun 11 08:27:52 allsky-pi allsky[3406]: INFO: Running nightday flow...
Jun 11 08:27:52 allsky-pi allsky[3406]: INFO: nightday flow Complete...
Jun 12 14:15:20 allsky-pi allsky[20337]: /home/harira/allsky/bin/capture_RPi: Got SIGTERM to exit.
Jun 12 14:15:20 allsky-pi allsky[20337]: Calling system(/home/harira/allsky/scripts/copy_notification_image.sh --expires 2 NotRunning &)
Jun 12 14:15:23 allsky-pi allsky[20337]:      ***** Stopping AllSky *****
Jun 12 14:15:29 allsky-pi allsky[6016]:      ***** Starting AllSky *****

[Allsky.log.txt](https://github.com/user-attachments/files/15835798/Allsky.log.txt)
[config.sh.txt](https://github.com/user-attachments/files/15835805/config.sh.txt)

EricClaeys, 2024-07-11:
@charira83, any update?

charira83, 2024-07-11:
Actually i'am still struggeling with this issue
The entry "Got SIGTERM... " in the allsky log was actually at the time, when i stopped allsky processing via allsky Webserver.

I found out, it has to do with the creation of timelapse Videos. If i turn off timelapse creation, then everything works fine and no crashes at all.

In the /var/log/messages log of the Linux System there are no entrys for allsky at All.

In the Webserver config saving locations of timelapses etc. Is set to "local" 

I have no clue, what could cause this Problem.

EricClaeys, 2024-07-13:
@charira83, I'm not sure what this means:
> In the Webserver config saving locations of timelapses etc. Is set to "local"

Could you please attach a screenshot?

Also, please do the following:
~~~
sudo systemctl stop allsky
sudo truncate -s 0 /var/log/allsky.log
# In the WebUI, set the Debug Level to 3
~~~
The next time Allsky stops, before restarting it, attach /var/log/allsky.log.

That log file should have what I need to figure out what's going on.

Are the timelapse files being created correctly and can you view them?

charira83, 2024-07-13:
"In the Webserver config saving locations of timelapses etc. Is set to "local" "- > i meant in the ftp-config file the saving locations are All "local" and the files are not saved in any external Servers etc.

I got it working now. 

All i did was cropping the images and reducing the size of the timelapses to 1920x1080 Pixel. As i am using the raspberry Pi HQ Cam with 12MP, the timelapses creation perhaps took too much resources or was struggeling with the Video file in full Resolution.

EricClaeys, 2024-08-04:
Closing - user resolved Issue.

closed 2024-08-04 by EricClaeys