AllskyTeam/allsky #3683: Camera stops capturing images

ChimorinNL, 2024-06-17:
### Environment

* Camera:  RPI_Version1
* OS: Woodworm
* Allsky version: v2023.05.01_04
* Pi model and amount of memory: RPI3B+ with 1GB ram, 32GB SD

### Bug Description

The camera stops capturing after an amount of time, this morning i saw it was stopped at 19:44 in the evening (here in The Netherlands) and i restarted it at 7:23 the next morning. After 3 minutes, at 7:26, the camera was stuck again. 

I did not change significant settings in the camera, no color balance or anything, only auto-gain (maybe it is an issue with this camera, its from aliexpress). Maybe someone can see what happens from the logfiles?

### Log / configuration files

[image]

[image]

ChimorinNL, 2024-06-24:
Hi! Updated my issue with the attached logfiles.

EricClaeys, 2024-06-25:
@ChimorinNL, Tom,  The attachments aren't there so I assume you attached them in an email?  Unfortunately GitHub doesn't accept email attachments so you'll need to login to GitHub and attach them.

EricClaeys, 2024-07-11:
@ChimorinNL, Tom, any update?

ChimorinNL, 2024-07-11:
Hi Sorry, i had some other thins to do (vacation, work) so my allsky project got some delay. No it is not solved yet. Every morning my camera stops capturing images. It does so everytime around 05:00 or 07:00. 

I've tried solving it by setting up a cronjob that restarts my pi at around 06:00, but that wont work (maybe more a problem of mine, but it works when the allsky camera is also working, so i think its just freezes the whole pi or something. Weird thing is that i can reach the WebUI of allsky and restart the pi from there).

Long story short, no, i havent solved this yet. If i can give you any information, please let me know where i can find that.

EricClaeys, 2024-07-11:
@ChimorinNL, if you attach /var/log/allsky.log using Debug Level 3, I can take a look.

ChimorinNL, 2024-07-14:
Sure, but how can i do Debug level 3?

ChimorinNL, 2024-07-14:
Here you go.
[allsky.log](https://github.com/user-attachments/files/16216545/allsky.log)

EricClaeys, 2024-07-14:
@ChimorinNL Tom, to get rid of all the "outside of image" errors, do this:
~~~
cd ~/allsky/config/overlay/config
ls -lt overlay*
~~~
One of the files should be newer than the others.  Edit it and look for this group of lines:
[image]

Change the x, y, tlx, and tly numbers to 200.  Yours will be numbers larger than the camera's resolution which is causing that error.

The timelapse took 2.5 hours to complete which is extremely long, even for a Pi 3B+.  Please do the following:
~~~
cd ~/allsky/images
ls 20240713 | wc -l
~~~
This will show how many images are in the timelapse.  Your daytime delay looks like it's around a couple seconds which will create ~ 8000 images, and your night delay is about 10 seconds which will create another few thousand.  Do you need such short delays?  It's causing your system to run very slowly, which _might_ be the cause of the camera stopping.  Today's stop occurred while the timelapse was being created.

The log file wasn't very helpful since the Debug Level wasn't high enough.  Please change it to 3 in the WebUI, then attach /var/log/allsky.log after the next stop.

ChimorinNL, 2024-07-15:
My guess is that it is the overlay-RPI.json file, it has indeed very big numbers. I changed that to 200.

[image]

I also changed my debug level (sorry, didnt know it was in the ui) and restarted the pi, so i hope i have some more information tommorow. 

I dont need that little delay between pictures no, do you maybe have a suggestion for settings?

EricClaeys, 2024-07-15:
The next release fixes the out-of-bounds numbers in the overlay file.  The numbers are good for the RPi HQ camera, but not for all the cameras with lower resolution.

There is no right or wrong for delay, but typically people have a longer daytime delay, like a minute or so, because they aren't looking for anything in the pictures, like meteors.  People who look for meteors usually have a very short nighttime delay so they minimize the time not taking a picture.  I have a minute since I only want to know if it's cloudy or not, and hence whether or not I can take astro images.

ChimorinNL, 2024-07-16:
Changed my settings to 70 seconds for daytime captures (60 seconds delay, 10 seconds exposure) and my camera is still running after reboot yesterday (as of 09:53)!

I've uploaded the logs with debug level 3, maybe you can see something what helped with this.

[allsky.log](https://github.com/user-attachments/files/16244938/allsky.log)

EricClaeys, 2024-07-17:
Tom, it appears you attached the wrong log file.  It doesn't have Debug Level 3 messages and still has "ERROR: Field 'RPi Version_1' is outside of the image" messages.
Please do the following:
~~~
sudo systemctl stop allsky
sudo truncate -s 0 /var/log/allsky.log
grep debuglevel ~/allsky/config/settings.json   # It should be 3, if not, change in the WebUI.
sudo systemctl start allsky
~~~
Then after the next night-to-day transition, upload `/var/log/allsky.log`.  No need for a reboot.

ChimorinNL, 2024-08-15:
Hi, for now things are going better. My pi sometime stops, but that may be do to the fact it is under voltaged. For now i would say it is working fine and i'll wait on that new update you mentioned.

EricClaeys, 2024-08-16:
Closing Issue since it appears to not be Allsky-related.

closed 2024-08-16 by EricClaeys