AllskyTeam/allsky #3684: Unable to remove allsky for a compelte re-install

WirthmU, 2024-06-18:
### Environment

* Camera:  RPi - HQCam
* OS: Bullseye
* Allsky version: n/a
 * Pi model and amount of memory: RPI 4 Model B Rev 1.2 / 4 GB

### Bug Description

I would want to make a complete re-install. I could halt allsky service and remove most of the file in the allsky-directory.
However it seems that the tmp directory still is blocking complete removal.

```
(foobar) pi@meteotux:~ $ sudo rm -R allsky
rm: cannot remove 'allsky/tmp': Device or resource busy
```

I already dit a reboot but still get the same message.

Any help to this?

EricClaeys, 2024-06-19:
Run
~~~
cd
sudo umount allsky/tmp
rm -fr allsky
~~~

WirthmU, 2024-06-21:
It worked - perfect!
Thank you

closed 2024-06-21 by WirthmU