AllskyTeam/allsky #3710: Upgrade issue with sessions

Alex-developer, 2024-06-28:
When upgrading to the 2024 branch you are unable to save the settings 

[image]

Clearing the sites cookies, containing the php session id allows the settings to be saved

Need to find out whats going on

EricClaeys, 2024-06-29:
@Alex-developer, how long was the WebUI page up before you saved the settings.

Do you have `Use Login` enabled or not?  I assume "yes" since that's when it creates a session.

Alex-developer, 2024-08-02:
In version v2023.05.01_04 lighttpd is running as root so the php sessions are created with as a root user

root@pi5:/home/pi/allsky/html# ll /var/lib/php/sessions
total 4
-rw------- 1 root root 83 Aug  2 22:01 sess_h5nejbkmlecu7dqbatvvua36ea

In v2024.xx.xx lighttpd is running as www-data so lightpd cannto access the previous sessions

EricClaeys, 2024-08-03:
Are you sure v2023.05.01_04 lighttpd ran as root?  I'm pretty sure it's always run as www-data for as long as I've been involved in Allsky.
Either way, my /var/lib/php/sessions directory is owned by root and has nothing in it.  I have "Require WebUI Login" disabled so have no sessions.

Alex-developer, 2024-08-03:
> Are you sure v2023.05.01_04 lighttpd ran as root? I'm pretty sure it's always run as www-data for as long as I've been involved in Allsky. Either way, my /var/lib/php/sessions directory is owned by root and has nothing in it. I have "Require WebUI Login" disabled so have no sessions.

Yes, the v2023.05.01_04 doesnt specify a user, they are commented out in the config so the server will start as root.

#server.username             = "www-data"
#server.groupname            = "www-data"

This will also be a problem for saving the settings as the csrf token will be regenerated on every page request

EricClaeys, 2024-08-03:
Good detective work @Alex-developer.

EricClaeys, 2024-08-03:
@Alex-developer, do we also need to make sure /var/lib/php/sessions/ is writable by the web server?
Mine isn't.

Alex-developer, 2024-08-03:
> @Alex-developer, do we also need to make sure /var/lib/php/sessions/ is writable by the web server?
> 
> Mine isn't. 

Probably, I am testing a fix at the moment but need to do a final one to check the permissions on the session directory

closed 2024-08-21 by Alex-developer