AllskyTeam/allsky #3719: [BUG] Segmentation Fault during compile

hemna, 2024-07-03:
Trying to install allsky on raspi4 bullseye.   

```
waboring@allsky-home:~/allsky/src $ git status
HEAD detached at v2023.05.01_04

waboring@allsky-home:~/allsky/src $ g++ --version
g++ (Raspbian 10.2.1-6+rpi1) 10.2.1 20210110
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

waboring@allsky-home:~/allsky/src $ uname -a
Linux allsky-home 6.1.61-v8+ #1696 SMP PREEMPT Thu Nov  2 16:44:46 GMT 2023 aarch64 GNU/Linux
```

```
waboring@allsky-home:~/allsky/src $ make
Building capture_ZWO.o ...
In file included from /usr/include/opencv4/opencv2/core/cvdef.h:183,
                 from /usr/include/opencv4/opencv2/core.hpp:52,
                 from capture_ZWO.cpp:1:
/usr/include/c++/10/limits:1669:7: internal compiler error: Segmentation fault
 1669 |       min() _GLIBCXX_USE_NOEXCEPT { return __FLT_MIN__; }
      |       ^~~
0xf78178ff ???
	../sysdeps/unix/sysv/linux/arm/sigrestorer.S:64
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <file:///usr/share/doc/gcc-10/README.Bugs> for instructions.
make: *** [Makefile:135: capture_ZWO.o] Error 1
```

EricClaeys, 2024-07-03:
Did you do the standard `git clone ...` then `./install.sh` to download and install Allsky?  Have you modified Bullseye at all?

This is a problem with the compiler, not Allsky.  Something on your system is corrupted.  It could possibly be a bad SD card.

aaronwmorris, 2024-07-03:
You can run `sudo dpkg --verify` to validate the files that are installed through the package manager on your system.  This can take a while to run.  Keep in mind that there will be some findings that are normal like config files that have changed.  You are primarily interested in binaries and libraries.

If you find a corrupted file, you can use `sudo apt-get install packagename --reinstall` to reinstall the files.

EricClaeys, 2024-07-11:
@hemna, Walter, any update?

hemna, 2024-07-11:
Yah.  So I went through the process of trying to reinstall some broken packages and although it worked piecemeal, it seemed that the SD card was continuing to corrupt the filesystem.  I've since gotten a new SD card and as well as a USB ssd which houses my / partition.   Thanks for the help!

closed 2024-07-11 by hemna