AllskyTeam/allsky #3730: [BUG] Unable to access allsky.local with different browser

njrusty, 2024-07-09:
### Environment

* Camera:  ZWO678MC
* OS: Bullseye
* Allsky version:
     * The newest software includes a file called  ~/allsky/version
     * If it does not exist, run:       grep Software /var/log/allsky.log
* RP4 - 8 GB

### Bug Description

I reinstalled ALLSKY yesterday. For some reason, I was no longer able to access allsky.local so I decided to reinstall. Got everything up and running and use a specific user/password combination.  Using my default web browser (Brave) I was able to access my allsky.local webpage and everything up and running.

However; when I tried to login into my allsky.local using Chrome, Firefox or Edge, the user/password screen appears but it will not allow me access. I definitely am entering my users/password correctly on all screens but these browsers not allowing access.

EricClaeys, 2024-07-10:
Are you entering the "admin" login and "secret" password (or whatever you may have changed it to)?  If so, what happens when you try?  A screenshot would be nice.

Try forcing a refresh in the browsers that don't work via Shift-F5.

njrusty, 2024-07-10:
Thank you Eric. I did not know the user/password is browser specific. Although, when I created my ALLSKY OS and had a specific username/password created, and then used my specific browser (Brave) to access allsky.local with my user/password combo it worked. However; as I had stated above, no other browser would work. However/ if using the admin/secret combo in ANY browser, it does work.  Appreciate your help!

EricClaeys, 2024-07-10:
The WebUI username and password are NOT browser-specific.  It's possible the non-working browsers has the old values cached.

EricClaeys, 2024-07-10:
Keep in mind that "admin" and "secret" let you view the WebUI. You access the Pi itself (e.g., with `ssh`) with the Linux username and password which is whatever you set them to.

EricClaeys, 2024-07-21:
@njrusty, Russ,
Is this still a problem?  If not, please close the Issue.  Thanks

EricClaeys, 2024-08-04:
Closing Issue - no response from user in 2 weeks.
The Allsky WebUI works with many browsers, and Brave will be one of them in the next release of Allsky.

closed 2024-08-04 by EricClaeys