AllskyTeam/allsky #3746: [ENHANCEMENT] - Allow easy changing of ALLSKY_IMAGES location

EricClaeys, 2024-07-17:
Many people have an SSD attached to their Pi and they want to use it to store images.

A proposed solution:

Add a "Change location of saved images" menu item.  It will:
- Prompt the user for AN EXISTING location.
- Prompt the user if they want to move all existing images to the new location.
- Create `uservariables.sh` if not already there.  Either way, set ALLSKY_IMAGES to the new location.
- Stop Allsky
- Update `/etc/lighttpd/lighttpd.conf` and `~/allsky/html/includes/allskyDefines.inc` with the new location.
- Optionally move existing images to new location.
- Start Allsky

EricClaeys, 2024-09-03:
Probably want an "allsky_images" setting.

EricClaeys, 2024-12-12:
@Temppu76, when changing the location of ALLSKY_IMAGES after installing Allsky there are a few other things to change, which are summarized in the first post:
1. Stop Allsky: `sudo systemctl stop Allsky`
2. Undo the change you made to `variables.sh` and instead put that line in a new file called `~/allsky/config/uservariables.sh`.  When you upgrade Allsky a new `variables.sh` file is created which wipes out your changes.  Your prior `uservariables.sh` file is reused however.
3. You need to use `sudo` to edit `/etc/lighttpd/lighttpd.conf` and replace the current location of the images (often `/home/pi/allsky/images`) with the new location.  Please create a directory called `images` somewhere on your SSD and put the images there.
4. Run `sudo systemctl stop lighttpd ; sudo systemctl start lighttpd` to have the system reread the file.
5. Edit `~/allsky/html/includes/allskyDefines.inc` and do the same thing, but do NOT use `sudo`.
6. Restart Allsky: `sudo systemctl start allsky`
7. Make sure things work - images are saved, can be viewed, end of night processing works (timelapse, etc.), modules work, etc.

Note that I have never done this and don't know if anyone has upgraded Allsky with this change, so I am counting on your feedback.

Temppu76, 2024-12-12:
@EricClaeys Thanks for the guidance, I did the changes and images are saved to the SSD location, but the images view in the WebUI still does not show the saved images.

Any other ideas what could help here?

I'll do re-installation to SSD, but I can try some tricks before that to find out how this could be done :)

EricClaeys, 2024-12-12:
In the Images page of the WebUI, try forcing a refresh of your browser by pressing Shift-F5.

EricClaeys, 2024-12-12:
If that doesn't work, double check that `~/allsky/html/includes/allskyDefines.inc` points to the new location - if it does, on the web page, right-click and select "View page" or equivalent.  That will bring up the html for the page.  Please paste it in your response.

Temppu76, 2024-12-12:
Hi,
All seems to be as you instructed. I refreshed the page with shift+f5 and also cleared the browser cache.
Attached the page source.
[Images - WebUI.txt](https://github.com/user-attachments/files/18118863/Images.-.WebUI.txt)

EricClaeys, 2024-12-13:
Good to hear.  I will add that to the instructions.

ova103, 2025-01-14:
Hello

I wanted to save the tmp image and day images on the SSD while allsky is running on a SD card.

So I followed the instructions above:

I edited variables.sh  (Note I did not upgrade allsky so I edited varables.sh instead of creating uservariables.s)

	# ALLSKY_TMP="${ALLSKY_HOME}/tmp"
	ALLSKY_TMP="/mnt/ssd/home/Olivier/allsky/tmp"

	# ALLSKY_IMAGES="${ALLSKY_HOME}/images"
	ALLSKY_IMAGES="/mnt/ssd/home/Olivier/allsky/images"

I edited the lighttpd.conf
   
[image]

I edited allskyDefines.inc

      //define('ALLSKY_TMP', '/home/Olivier/allsky/tmp');
      define('ALLSKY_TMP', '/mnt/ssd/home/Olivier/allsky/tmp');
      // define('ALLSKY_IMAGES',  '/home/Olivier/allsky/images');
      define('ALLSKY_IMAGES',  '/mnt/ssd/home/Olivier/allsky/images');s: 

I have attached those files: 

[allskyDefines.txt](https://github.com/user-attachments/files/18412831/allskyDefines.txt)
[lighttpd.txt](https://github.com/user-attachments/files/18412832/lighttpd.txt)
[variables.txt](https://github.com/user-attachments/files/18412833/variables.txt)

The image file are written on the ssd (as well on tmp flder as on image folder), the minitimelaps too. 
So that's great.
I can see the image on the remote website. The timelapse as well. So that's great too.

But I don't see an image on the local website, nor the  timelapse (error 404 Not found). 

And more annoying, I don't see in the editor the "life view": 
[image]

and I cannot edit the overlay
[image]

I have attached the allsky log file
[allsky.log](https://github.com/user-attachments/files/18413352/allsky.log)

I probably need to update something in those 3 files but don"t know what.

Can you help me on this

Thanks a lot.

EricClaeys, 2025-01-14:
Is there a reason you didn't follow the directions at the link I provided?

You need to undo the changes to ALLSKY_TMP and anything else you did that wasn't at that link.

~/allsky/tmp is in memory, not on a disk.  This eliminates most of the reads and writes to the SD card which improves performance.

The web server has an alias called `current` that points to the ~/allsky directory.  It expects "image.jpg" to be at `/current/tmp/image.jpg`  Your additional changes broke that link.

ova103, 2025-01-14:
Hello

I'm not sure I understand. Allsky/tmp is a folder, no? I see files in there, And I see every new image that is being saved there?
So am I wrong on this?

My idea was to have this 'tmp' on SSD to avoid wearing out the SD

But if it's indeed not a folder but in memory I'll undo it.

ova103, 2025-01-14:
Hello Eric,

I removed what was not on your link and it's working again fine, meaning, the images are saved on the SSD, and the life view and overlay editor are working correctly. Also it's ok for the local website. 
So sorry for having not followed exactly your link.

But now of course the images are also being written on the tmp folder on the SD, which I want to avoid to reduce wear on teh SD card.  
So I wanted to put the tmp in memory and followed the instuctions from the FAQ from the documentation.

I did "allsky-config  recheck_tmp"; nothing happened  (I checked tmp folder, image is still saved there)
I did "allsky-config", so I got now a menu and I choose to move tmp to memory. Still image being written on the tmp folder.
I always stopped allsky for doing this. 

I rebooted, still the same.

So what do I do wrong here? 
And how can you check the tmp file is really moved to memory? Any command for that?

Thanks again for your support

EricClaeys, 2025-01-15:
@ova103,
You are making things hard on yourself be doing things that weren't in the instructions.

You are correct that `/home/Olivier/allsky/tmp` is a folder with files in it.  However, instead of pointing to a location on a disk, it points to memory, so reads and writes to it only go to memory, never to a disk.  We did that for performance and to save writes to the SD card.
When the Pi boots it allocates memory and points the `/home/Olivier/allsky/tmp` directory to it (that process is called "mounting" allsky/tmp).  When the Pi shuts down or is rebooted, `/home/Olivier/allsky/tmp` is converted back into a "regular" directory and the prior contents deleted (called "unmounting").  That's OK because it only contains temporary files (hence the "tmp") that are only needed when Allsky is running.

You can verify that `/home/Olivier/allsky/tmp` is in memory by running `mount | grep allsky`.  That shows all the items with "allsky" in their name that are mounted.  Any line with `tmpfs` in it is a "temporary filesystem" in memory.
If `/mnt/ssd/home/Olivier/allsky/tmp` is in the list, let me know and I'll tell you the commands to remove it.

A couple other people got everything working by following the instruction, so I'm confident we'll get you working as well.

Eric

ova103, 2025-01-15:
Hello Eric

Thanks a lot for those explanations. So if I understand well, even if I see files being written on the folder, they are not written on the SD card but rather to memory, right?

Here is the output of "mount | grep allsky":
[image]

So I guess this is ok which means the image is not written on the SD card but to memory.

A last question: if allsky/tmp point to memory, why is there still something mentionned in the FAQ concenring the wear out of the SD card? Was this valid for previous Allsky versions? Would you not mention instead that wear out of the card is minimised because Allsky points the tmp images alsways to memory? 

Thanks again

EricClaeys, 2025-01-15:
@ova103,
You are correct that file systems in memory are never written to disk.

Your screenshot is exactly what you want to see.

Regarding the FAQ, it says:
> During Allsky installation you were prompted to do this - if you did you can ignore this FAQ item.

The FAQ entry only applies to people who, for whatever reason, said "no" during installation when asked if the wanted to put allsky/tmp in memory.
In a future release we might not ask and instead, always put it in memory.  The installation would be slightly easier and we could remove the FAQ item.
I suspect anyone that says "no" probably doesn't know what they are doing.

EricClaeys, 2025-01-16:
@ova103,
my pleasure.  I actually enjoy helping people get their Allsky working.  It also gives me a good idea where people struggle, which often leads to Allsky changes to minimize the struggle.
For example, several people have asked about moving the Allsky images to an SSD, so I wrote the instructions above with the intent that late this year I'd make a script that did all the work.  But after seeing you struggle a lot, I wrote the script yesterday and plan to have it in the Feb 1 release.

After the startrails, keogram, and timelapse are created they are saved in the "images" directory for the current day.
If the user also specified to upload them:
* If the user has a REMOTE Website, they are uploaded there.
* If the user has a remote SERVER, they are uploaded there.
* If the user has a LOCAL Website, i.e., on the Pi, they are COPIED to ~/allsky/html/allsky/startrails, keograms, and videos.

&nbsp;
> still the timelapse, keogram, startrail etc are being written on the SD card.

**Where on the SD card?  ~/allsky/images/YYYYMMDD, or ~/allsky/html/allsky/... ?**

&nbsp;
> would it be better to move allsky from sd to ssd

YES!  Other than not wanting to create a new Linux image on an SSD, I'm not sure why anyone who has an SSD wouldn't boot off it and hence use it for everything.  When that's done, you can remove the SD card.

The Pi 5 supports NVMe disks which are smaller than SSDs and 10+ times faster.  I boot off them on my Pi 5's and don't use an SD card.  It's actually pretty easy to do, but because you're creating a new image of Linux, you'll need to reinstall whatever programs you installed and re-apply all your settings.  For people that only use the Pi for Allsky, this is pretty easy and is described in [here](https://htmlpreview.github.io/?https://raw.githubusercontent.com/AllskyTeam/allsky/master/html/documentation/explanations/imageSDcard.html).  Replace "SD card" with "SSD" in the instructions.

ova103, 2025-01-16:
Thanks Eric. 

Concerning booting on SSD, for the moment I'm reluctant to do it as my SSD card is a bit unprotected in the housing of the allsky. It's an M2 SSD, so quite long and it fits only when putting it's positionned at an angle.  Not so good as it's loose in the box. 

I have by the way a PI  4 on which a NVME disk is not supported.

I might move the whole allsky on the SSD, but I don't know how to make the path identical as the sd card. The path on SSD is now: 
mnt/ssd/home/olivier/allsky
On the sd its /Olivier/allsky.
So how do I make the path identical?

Thanks for your support.

EricClaeys, 2025-01-17:
@ova103,
makes sense not booting off the SSD until it's in a more stable position.  I learn something new every day.

Is your login named `Olivier` or `olivier`?
Linux is case-sensitive so those two logins are different.
If it's `Olivier`, then in the instructions below change `olivier` to `Olivier`.

Instead of moving just the `allsky` directory to the SSD, I suggest moving `/home/olivier`.  When Allsky runs, it assumes the `allsky` directory is in the HOME directory of whoever installed it.  The `olivier` login's HOME directory is `/home/olivier` so Allsky assumes it's installed in `/home/olivier/allsky`.

To move `/home/olivier` to the SSD, do the following.
This assumes the Allsky images are in `/mnt/ssd/home/olivier/allsky`.  If that is NOT correct, do NOT do the stuff below.

~~~
echo $HOME    # it should return     /home/olivier
sudo systemctl stop allsky
cd /mnt/ssd/home/olivier
mv allsky allsky-images
sudo usermod --home /mnt/ssd/home/olivier --move-home
~~~

Leave the current window as is and log into the Pi in a SECOND window.  Run
~~~
echo $HOME    # it should return   /mnt/ssd/home/olivier
~~~
If that works, close the first window since it's still using the old location which no longer exists.

Allsky will need to be re-installed so it picks up the new location.
I suggest using the release that's coming out February 1.  It has some improvements that may help you.
To do so:
~~~
mv allsky-OLD  allsky-SAVED   # ignore any error message
mv allsky/config/uservariables.sh  allsky/config/uservariables.sh-SAVED
mv allsky  allsky-OLD
git clone --recursive --branch v2024.12.06_01 --depth=1 https://github.com/AllskyTeam/allsky.git
cd allsky
./install.sh   # tell it to use your prior version
~~~

If there is anything in your Allsky settings with `/home/olivier/allsky`, update to `/mnt/ssd/home/olivier/allsky`.
Check your other settings, then press the "Save changes" button.

Move your old images: `cd; mv allsky-images/* allsky/images; rmdir allsky-images`.

Make sure Allsky's working and can see your images.

ova103, 2025-01-18:
Thanks

> [@ova103](https://github.com/ova103), makes sense not booting off the SSD until it's in a more stable position. I learn something new every day.

I can imagine you see a ton of strange things happening ...
 
> Is your login named `Olivier` or `olivier`? Linux is case-sensitive so those two logins are different. If it's `Olivier`, then in the instructions below change `olivier` to `Olivier`.

Olivier. I'll change the instructions 

 
> To move `/home/olivier` to the SSD, do the following. This assumes the Allsky images are in `/mnt/ssd/home/olivier/allsky`. If that is NOT correct, do NOT do the stuff below.

yes it is correct 

Before doing all this let me say first:
- my SSD disk is not "empty". I have copied a few days ago the allsky folder from the SD to the SSD (to mnt/ssd/home/Olivier/) but of course it did not work from the SSD. It still works from the SD
-
- however the images are currently already saved on the SSD (and so not on the SD) thanks to the instructions you described in this issue and the support you gave me last week.

So my question is: can I just follow these instructions of moving /home/olivier you gave in your last reaction (ok, after the next release) or should I first delete all these folders on the SSD. Or maybe empty them.

Thanks again

EricClaeys, 2025-01-18:
Is anything in `/mnt/ssd/home/Olivier/allsky` used other than the `images` directory?

ova103, 2025-01-19:
Hello

> Is anything in `/mnt/ssd/home/Olivier/allsky` used other than the `images` directory?

no

closed 2025-02-08 by EricClaeys

njrusty, 2025-02-22:
Based on the following "an SSD: using a USB-to-SSD adapter, plug the USB end of the cable into a USB 3 port on the Pi (or USB 2 port if your Pi doesn't have USB 3).
IS IT MOUNTED AUTOMATICALLY, AND IF SO, WHERE?"
The where part is confusing. If you attached the SSD and the following instructions are to copying files; exactly 'where' do you copy the files too?? Meaning, how do you find the location of the SSD??

EricClaeys, 2025-02-22:
LOL.  I don't know where, which is why I wrote "IS IT MOUNTED AUTOMATICALLY, AND IF SO, WHERE?"  That's really a comment to myself to fix and the page should have said that.

When I wrote that page I didn't have access to a USB drive, but I do now so I'll try it.  I should be able to get back to you later today.
Eric

EricClaeys, 2025-02-22:
@njrusty, in the meantime, please run the following and past the output:
~~~
mount
echo "============================="
df -h
~~~

EricClaeys, 2025-02-23:
@njrusty,

I just added a bunch of stuff to the next release, v2024.12.06_02, that fixes the documentation and IMHO makes it pretty easy to determine "WHERE" the SSD is on the Pi.

If you want to upgrade to that release (which isn't official yet, but works well), you can then follow the improved instructions and provide feedback.
Let me know if interested.

njrusty, 2025-02-23:
Definitely interested

Yahoo Mail - Email Simplified

EricClaeys, 2025-02-23:
Follow the standard installation instructions, EXCEPT for the `git clone ...` command, use
~~~
git clone --depth=1 --recursive --branch v2024.12.06_02 https://github.com/AllskyTeam/allsky.git 
~~~

Once installed, look at the `Explanations -> Image an SD card, SSD, or NVMe` Documentation page for how to use your SSD.

If you have any problems with the installation, attach `~/allsky/config/logs/install.log`.

If you have a remote Allsky Website, run `remoteWebsiteInstall.sh` after you have Allsky working.

njrusty, 2025-02-23:
* Found ZWO ASI678MC
*** NOTICE:**********The following steps can take up to an hour depending on the speed ofyour Pi and how many of the necessary dependencies are already installed.You will see progress messages throughout the process.**********
* Preparing the web server.* Creating new /var/log/lighttpd/error.log.* Installing dependencies.* Preparing Allsky commands.
*** ERROR:**********Compile failed**********
* The full log file is in /home/allsky/allsky/config/logs/make_all.logThe end of the file is:        Building capture_RPi.o ...        Building mode_mean.o ...        2025-02-23 10:54:26 Building capture_RPi program...        2025-02-23 10:54:26 Done.        make[1]: Leaving directory '/home/allsky/allsky/src'        make[1]: Entering directory '/home/allsky/allsky/config_repo'        2025-02-23 10:54:26 nothing to do for all        make[1]: Leaving directory '/home/allsky/allsky/config_repo'        make[1]: *** notification_images: No such file or directory.  Stop.        make: *** [Makefile:26: all] Error 2

EricClaeys, 2025-02-23:
@njrusty, sorry about that.  This is why the release isn't official yet...

Please run:
~~~
cd ~/allsky
mv Makefile Makefile-SAVED
# Copy the attached file here
mv Makefile.txt Makefile
~~~
Then re-run the installation.  It should skip the steps it already performed.

[Makefile.txt](https://github.com/user-attachments/files/18931708/Makefile.txt)

njrusty, 2025-02-23:
Appears the folder does not contain Makefile???

Linux allsky 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64
The programs included with the Debian GNU/Linux system are free software;the exact distribution terms for each program are described in theindividual files in /usr/share/doc/*/copyright.
Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extentpermitted by applicable law.Last login: Sun Feb 23 12:41:17 2025 from ***@***.***:~ $ cd ~/allskymv Makefile Makefile-SAVED# Copy the attached file heremv Makefile.txt Makefilemv: cannot stat 'Makefile': No such file or directorymv: cannot stat 'Makefile.txt': No such file or directory

EricClaeys, 2025-02-23:
It looks like you weren't in `~/allsky` when the other commands were executed.  It couldn't even find Makefile.txt, which you just copied there.

Run the commands one at a time, getting a prompt after each one.
If you get a "No such file..." message after any command, run `pwd; ls` to make sure you are in the correct place.