AllskyTeam/allsky #3766: Allsky website folders seem to be made in the wrong place

broulston, 2024-07-24:
### Environment

* Camera:  RPi HQ
* OS: Bookworm
* Allsky version: v2023.05.01_04
* Pi 5 8 GB

### Bug Description

I seem to have a recent issue with my Allsky website. I had it working, and two nights of Timelapse, keograms, and star trails were uploaded fine. They were visible and could be viewed!

However, starting on 07/22/2024, the new files failed to show. When I went in and looked, the files were created alright, but in the log (attached) they were failing to upload. This is the line form the log (attached):
~~~
2024-07-23T05:02:13.309714-04:00 raspberrypi5 allsky[2646584]: ===== Uploading '/home/cuastropi5/allsky/images/20240722/keogram/keogram-20240722.jpg'
2024-07-23T05:02:13.433085-04:00 raspberrypi5 allsky[2646607]: Upload output from '/home/cuastropi5/allsky/images/20240722/keogram/keogram-20240722.jpg:
2024-07-23T05:02:13.433445-04:00 raspberrypi5 allsky[2646607]:    cp: cannot create regular file '/home/cuastropi5/allsky/html/keograms/keogram-20240722.jpg': No such file or directory
~~~

In my `ftp-settings.sh` I have the following set
```
	# How will files be uploaded?
PROTOCOL="local"

	# The directory where the current image should be copied to.
IMAGE_DIR="/home/cuastropi5/allsky/html/"
WEB_IMAGE_DIR=""

	# The directory where the timelapse video should be uploaded to.
VIDEOS_DIR="/home/cuastropi5/allsky/html/videos"
VIDEOS_DESTINATION_NAME=""
WEB_VIDEOS_DIR=""

	# The directory where the keogram image should be copied to.
KEOGRAM_DIR="/home/cuastropi5/allsky/html/keograms"
KEOGRAM_DESTINATION_NAME=""
WEB_KEOGRAM_DIR=""

	# The directory where the startrails image should be copied to.
STARTRAILS_DIR="/home/cuastropi5/allsky/html/startrails"
STARTRAILS_DESTINATION_NAME=""
WEB_STARTRAILS_DIR=""
```

This is the same setup I had that had been working. However, now when I go into those folders, the subfolders for `videos` (and the like) are nested under another Allsky folder. Like so `/home/cuastropi5/allsky/html/allsky/videos/`. I tried reinstalling via `cuastropi5@raspberrypi5:~/allsky/website $ ./install.sh` and it ran fine saying my old website was  moved and to check my settings. But with this new install, the path is still `/home/cuastropi5/allsky/html/allsky/videos/`. 

So is this the true path that should be used? If so, any thoughts on why did my website originally work with folders in the path `/home/cuastropi5/allsky/html/videos/`?

### Log / configuration files

[allsky.log](https://github.com/user-attachments/files/16362647/allsky.log)

EricClaeys, 2024-07-24:
All the local Website files are in ~/allsky/html/allsky, so your ftp-settings.sh file need to be fixed.

Look in the [Allsky Website settings](https://htmlpreview.github.io/?https://raw.githubusercontent.com/AllskyTeam/allsky/master/html/documentation/settings/allskyWebsite.html) page for an example - it's at the end of the "ftp-settings.sh settings" section.

Would you please run `cd ~/allsky/html;  ls -FC` ?    I'd like to see what else is in that directory.

broulston, 2024-07-24:
Ahh ok, I can update those. Is there a way to force previously made videos/trails etc to upload if they exist? or do you have to manually add the ones that failed to upload?

Are there any thoughts why it worked the first two nights but not since?

I've listed the contents of the directory below:
```
cuastropi5@raspberrypi5:~ $ cd ~/allsky/html;  ls -FC
allsky/  allsky-OLD/  cgi-bin/  config/  css/  documentation/  favicon.ico  image.jpg  includes/  index.php  js/  lib/  public.php
```

EricClaeys, 2024-07-24:
Interesting, I don't see any of the startrails, keograms, etc. in ~/allsky/html.  Did you delete them?

The easiest way to upload the older ones would have been to move them from ~/allsky/html/startrails/* to ~/allsky/html/allsky/startrails.

To "upload" prior files:
~~~
generateForDay.sh --upload DATE
~~~
replacing "DATE" with one of the directory names in ~/allsky/images, e.g.,
~~~
generateForDay.sh --upload 20240723
~~~
would upload (really, just a copy in your case) yesterday's files.

EricClaeys, 2024-07-26:
Glad you got it working!
This will be easier in the next release - there will be 2 settings for local websites - "days to keep" and "Use local website".  All the directory names that cause a LOT of issues like yours are gone!  And there's no need to "install" it.

I'm closing this Issue since it's resolved.

closed 2024-07-26 by EricClaeys