AllskyTeam/allsky #3778: [BUG]  MQTT as part of module "Clear Sky Alarm" protocol error with Mosquitto MQTT in home assistant

HenkUyttenhove, 2024-07-30:
### Environment

* Camera:  RPi - IMX290
* OS: Bookworm
* Allsky version:  v2023.05.01_04  
* Pi model and amount of memory Raspberry Pi 4 Model B Rev 1.5 4MB

### Bug Description

* Include a clear and concise description of what the bug is.
When the module "Clear Sky Alarm" is used together with Mosquitto MQTT of Home Assistant, you get every update an alarm
"mosquite broker Client <unknown> disconnected due to protocol error"

* Can it be reproduced?  If so, how?
Fails every time

### Log / configuration files
No specific logs, no config files besides MQTT parameters in module

EricClaeys, 2024-08-05:
@Alex-developer, any ideas?

Alex-developer, 2024-08-12:
Not off the top of my head but I will take a look

Alex-developer, 2024-08-13:
Found a bug so I have migrated the logic from the allsky_publish module. Tested on Homeassistant with a non SSL conenction and Hivemq with an SSL conenction

[image]

[image]

The change is currently awaiting approval but hopefully that wont take too long

Cheers

Alex

closed 2024-08-13 by EricClaeys