AllskyTeam/allsky #3783: sunset data is 3 days old: PWD: Operation not supported

benhartwich, 2024-08-01:
Hi,

I got the error at allsky website: 
`WARNING: sunset data is 3 days old.
See the 'Troubleshooting > Allsky Website' documentation page for how to resolve this.
`
The debug command gave me the following output:
`allsky@allsky:~/allsky/scripts $ ./postData.sh --debug
WEBSITES=remote, WEBSITES_TO_TO=
Uploading '/home/allsky/allsky/config/settings.json' to htdocs/allsky/viewSettings
Upload output from '/home/allsky/allsky/config/settings.json:
   quote: PWD: Operation not supported
drwxr-xr-x    3 root     root         4096 Jul  5  2020 .
drwxr-xr-x    3 root     root         4096 Jul  5  2020 ..
drwxr-xr-x    6 1001     1002         4096 Jul 29 03:21 htdocs
In DIRECTORY=htdocs/allsky/viewSettings/:
drwxr-xr-x    2 1001     1002         4096 Aug  1 03:00 .
drwxr-xr-x   11 1001     1002         4096 Aug  1 07:18 ..
-rw-r--r--    1 1001     1002          115 Jul 28 18:14 README.txt
-rw-r--r--    1 1001     1002        19207 Jun  1 17:56 allskySettings.php
-rw-r--r--    1 1001     1002        16880 Jun  1 17:56 custom.css
-rw-r--r--    1 1001     1002        28091 Jun  1 18:39 options.json
-rw-r--r--    1 1001     1002         2462 Jul 28 20:32 settings.json

Uploading '/home/allsky/allsky/tmp/data.json' to htdocs/allsky
Upload output from '/home/allsky/allsky/tmp/data.json:
   quote: PWD: Operation not supported
`

What does PWD: operation not supported mean?

EricClaeys, 2024-08-01:
@benhartwich, you can ignore that message.
`PWD` is a command that some, but not all FTP servers support.  It means Print Working Directory - in other words, tell me what directory I am currently in.  It's helpful when debugging upload problems.  The next Allsky releases uses a different method to get the current Directory.
Linux has the same command, but lowercase - `pwd`.  Try it in a terminal window.

Are you still getting the "sunset" message?  If so, what is the date of the `data.json` file on the server?
And what are the contents?
That file is updated daily at the end of night.

EricClaeys, 2024-08-12:
Closing Issue - no response from user in a couple weeks.
The "sunset data is 3 days old" message is almost always due to incorrect ftp-settings.sh entries.  No know Allsky bugs would cause that.

closed 2024-08-12 by EricClaeys