AllskyTeam/allsky #3807: Help with wew heater status under System information

Alex-developer, 2024-08-12:
### Discussed in https://github.com/AllskyTeam/allsky/discussions/3805

<div type='discussions-op-text'>

<sup>Originally posted by **gostnghost** August 11, 2024</sup>
I have a on and off button for my dew heater under the system information window. like this

[image]

The status updates when I click the ON and OFF buttons. The problem is that if the Dew heater module changes the value, the status doesn't update here. Is it possible to read the dew heater status (AS_DEWCONTROLHEATER) in the allskydew.json file and update the System information when I enter this page without clicking ON or OFF?

/Göran</div>

EricClaeys, 2024-10-30:
@Joshude, see [this documentation page](https://htmlpreview.github.io/?https://raw.githubusercontent.com/AllskyTeam/allsky/master/html/documentation/explanations/SystemPageAdditions.html) for instructions on adding buttons and other items to the WebUI's "System" page.

EricClaeys, 2024-11-28:
@Joshude, did you see the documentation page describing how to add the buttons?

closed 2024-12-15 by Alex-developer