AllskyTeam/allsky #3830: [ENHANCEMENT] Additions to checkAllsky.sh

EricClaeys, 2024-08-22:
1.
Since quite a few people have problems with missing, corrupted, and/or incorrect entries in Module and Overlay files, I suggest adding checks to `checkAllsky.sh` to find these problems.
We can start off small (looking for problems people have already identified), and as new problems arise, add checks for those as well.

2.
`checkAllsky.sh` should probably be run automatically once a week (?) if the settings file was updated in the last week.
This will catch problems from # 1 as well as any new problems that appeared.

EricClaeys, 2024-11-28:
@Alex-developer, comments?

What's the best way to run `checkAllsky.sh` weekly?  The Periodic Modules are geared more toward "seconds between runs", not days.

Alex-developer, 2024-11-28:
The main issue with the flows being corrupted is fixed in 2024.12.06 so I am not entirely sure what this is going to check for?

However if we do want a 'checking' system then I would have thought crond would be the best way to run it periodically

EricClaeys, 2024-11-30:
Ok, let's see what, if any, corruption exists with v2024.12.06; if none, we can close this issue.