AllskyTeam/allsky #3831: Error SAI ZWO

WirthmU, 2024-08-22:
# Is this a QUESTION or FEATURE REQUEST?

If so, please add a Discussion item by clicking on the Discussions link above;  do NOT create an Issue. 

sky allsky[877]: *** /home/pi/allsky/bin/capture_ZWO: WARNING: ASISetControlValue() for control 1, value=4482, camNum=0, makeAuto=0 failed: ASI_ERROR_GENERAL_ERROR (16)
Aug 21 20:09:23 allsky allsky[877]:   > /home/pi/allsky/bin/capture_ZWO: ERROR: Failed getting image: ASI_ERROR_TIMEOUT (with 0.8 exposure = YES) (11)
Aug 21 20:09:25 allsky allsky[877]: STARTING EXPOSURE at: 2024-08-21 20:09:25   @ 4.48 ms
Aug 21 20:09:25 allsky allsky[877]: *** /home/pi/allsky/bin/capture_ZWO: WARNING: ASISetControlValue() for control 1, value=32, camNum=0, makeAuto=0 failed: ASI_ERROR_GENERAL_ERROR (16)
Aug 21 20:09:25 allsky allsky[877]: *** /home/pi/allsky/bin/capture_ZWO: ERROR: flushBufferedImages() setControl() returned ASI_ERROR_GENERAL_ERROR (16)
Aug 21 20:09:25 allsky allsky[877]: *** /home/pi/allsky/bin/capture_ZWO: WARNING: ASISetControlValue() for control 1, value=4482, camNum=0, makeAuto=0 failed: ASI_ERROR_GENERAL_ERROR (16)
Aug 21 20:09:30 allsky allsky[877]:   > /home/pi/allsky/bin/capture_ZWO: ERROR: Failed getting image: ASI_ERROR_TIMEOUT (with 0.8 exposure = YES) (11)
Aug 21 20:09:32 allsky allsky[877]: *** /home/pi/allsky/bin/capture_ZWO: ERROR: Maximum number of consecutive errors of 5 reached; capture program exited.
Aug 21 20:09:35 allsky allsky[877]:      ***** Stopping AllSky *****
Aug 21 20:09:36 allsky allsky[747]: Stopping Allsky: Allsky Stopped!<br>Non-recoverable ERROR found<br>See /var/log/allsky.log.
Aug 21 20:09:36 allsky allsky[747]:      ***** AllSky Stopped *****

&nbsp; 
## Look in the Wiki before submitting this Issue
The Wiki has lots of information including troubleshooting tips for the vast majority of known issues.

 &nbsp; 

### Environment

* Camera:  ZWO  or  RPi - include the model
* OS: Buster or Bullseye
* Allsky version:
     * The newest software includes a file called  ~/allsky/version
     * If it does not exist, run:       grep Software /var/log/allsky.log
* Pi model and amount of memory (512 MB, 1, 2, 4, or 8 GB)

### Bug Description

* Include a clear and concise description of what the bug is.
* Can it be reproduced?  If so, how?
* Did anything change?  Any settings?  Most issues occur after a change was made.
* Please include the exact messages or a screenshot.
* If this is a configuration-related issue (or you are not sure), attach these files, appending ".txt" to their names first:
  * Allsky problems:
    * ~/allsky/config/config.sh
    * ~/allsky/config/settings.json
    * ~/allsky/config/ftp-settings.sh (only for upload-related problems)
  * Allsky Website problems:
    * ~/allsky/html/allsky/configuration.json (local Website)
    * ~/allsky/config/remote_configuration.json (remote Website)

If you have program output or multi-line messages to include,
add it like this so it formats correctly (note the lines with tildas):

~~~
output/message line 1
output/message line 2
~~~

### Log / configuration files

Follow the instructions for Reporting Issues in the Wiki, then ATTACH a copy of the file(s) above.
Do NOT copy/paste them into this Issue.

EricClaeys, 2024-08-22:
@WirthmU,
ASI_ERROR_GENERAL_ERROR indicates a hardware problem - could be the camera itself, the USB cable, or the Pi.
Is this a new camera?

Things to try:

1. Is the cable plugged in tightly on both ends?
2. Try the other three USB ports on the Pi.
3. Try a new cable.
4. In config.sh near the end are a couple UHUBCTL variables.  One sets the path to the uhubctl program, e.g., /home/pi/uhubctl.  The other is the port the camera is on.  To be honest, I never figured this out and just use "2".
5. Try using the camera on a PC.  You'll probably need to download ZWO's software, but this is a great way to rule out the camera as a problem.

WirthmU, 2024-08-23:
The system has been up and running for almost 3 years and it used to work for months in a row. I do not think that the cables might be the problem.

After installing UHUBCTL and upgrading to the latest Allsky the system has been running for almost 24 hours without any problems before it crashed again. It looks like UHUBCTL makes a slight difference.

EricClaeys, 2024-08-23:
An easy way to check if it's the new Allsky is to try `allsky-OLD` if you still have it.

Other than a new Allsky, what else is different than what you had before?

Cables can go bad for no apparent reason.

WirthmU, 2024-08-23:
The camera is installed in a remote location in a holiday house. I am a bit reluctant to take the system apart for testing. I guess I will rather try with UHUBCTL and look wether this helps.

aaronwmorris, 2024-08-23:
@WirthmU You have to think of your all sky system as being in a more "industrial" environment.  It is subjected to repeated cooling/heating cycles daily of 20-30 degrees C delta.  The metal contacts of the USB connection are physically expanding and contracting due to these changes.  This temperature delta can also mechanically affect the wiring in the connector.  

It is probably subjected to a more humid environment which can induce more oxidation in the connectors.

This can affect both the USB and power connectors.  Reseating the connectors would probably fix both of these problems.

A bad power supply can cause these issues, too.

WirthmU, 2024-08-24:
I totally agree @aaronwmorris these cameras are high tech in an extremely harsh environment. I went through a number of iterations of camera and housing designs to find workarounds between heat and humidity. And even using the best components available the hardware would need service every once in a while - just like a car.

WirthmU, 2024-08-26:
It still looks to me like it rather is a software than a hardware problem.
I observed the loss of connection to the camera for several days always between 9 and 10 pm. and of course the allsky service went down with it.
Now today I did reset the USB port on the PI issuing the uhubctl commands

`uhubctl -l 1-1 -p 1 -a on`

and then about 30 seconds later 

`uhubctl -l 1-1 -p 1 -a off`

The allsky service and current daytime picture immediately came back.

Do you think you can realize a corresponding watchdog routine in allsky or should this rather be run apart from it?

EricClaeys, 2024-08-26:
@WirthmU, what would a watchdog timer do?
If Allsky finds the uhubctl command it will automatically invoke it after 5 USB-related errors.  I think it will do that two times, then give up.  Did you set the UHUBCTL variables in config.sh?

Another think to try is changing the `New Exposure Algorithm` setting in the WebUI.

WirthmU, 2024-08-27:
If it is already implemented there is no need for a watchdog.

I did set the following in config.sh
```
UHUBCTL_PATH="/usr/sbin/uhubctl"
UHUBCTL_PORT=1
```
`New Exposure Algorithm` is set to yes

This is what I got this morning:
```
Aug 27 09:39:41 allsky allsky[127730]: ==========
Aug 27 09:39:41 allsky allsky[127730]: === Starting daytime capture ===
Aug 27 09:39:41 allsky allsky[127730]: ==========
Aug 27 09:39:41 allsky allsky[127730]: STARTING EXPOSURE at: 2024-08-27 09:39:41   @ 100.00 ms (0.10 sec)
Aug 27 09:40:00 allsky allsky[127730]:      ***** Stopping AllSky *****
Aug 27 09:40:01 allsky allsky[127892]:      ***** Starting AllSky *****
Aug 27 09:40:02 allsky allsky[127892]: WARNING: Resetting USB ports looking for a ZWO camera, then restarting.
Aug 27 09:40:02 allsky allsky[127985]:      ***** Starting AllSky *****
Aug 27 09:40:02 allsky allsky[128007]:      ***** Starting AllSky *****
Aug 27 09:40:02 allsky allsky[128037]:      ***** Starting AllSky *****
Aug 27 09:40:07 allsky allsky[128073]:      ***** Starting AllSky *****
Aug 27 09:40:35 allsky allsky[128133]:      ***** Starting AllSky *****
Aug 27 09:40:42 allsky allsky[128254]: *** /home/pi/allsky/bin/capture_ZWO: ERROR: No Connected Camera...
Aug 27 09:40:45 allsky allsky[128254]:      ***** Stopping AllSky *****
Aug 27 09:40:45 allsky allsky[128133]: ***
Aug 27 09:40:45 allsky allsky[128133]: *** After fixing, restart the allsky service. ***
Aug 27 09:40:45 allsky allsky[128133]: ***
Aug 27 09:40:54 allsky allsky[128133]:      ***** AllSky Stopped *****
```

If I do interpret it correctly, the command that is issued by allsky if there is a USB connection failure it is.

The command that I am using in CLI is:
uhubctl -l 1-1 -p 1 -a off

EricClaeys, 2024-08-27:
I have never been able to figure out how uhubctl's "location" and "port" map to what lsusb returns, which is how Allsky determines if a camera exists.  And it doesn't seem to matter what arguments are passed to uhubctl since appears to power cycle everything.  So I think your port of 1 is fine.

Next time the problem happens, run `lsusb -d 03C3:` which will list the ZWO devices.
I still think this is a hardware issue.

aaronwmorris, 2024-08-27:
The location and port mapping is very vague in lsusb, but I think lsusb tries to simplify the USB layout.  The way hubs are chained together makes the tree very complicated.  And a "hub" is not a physical hub, but the virtual hubs that are used to multiplex the ports.

If you plug in another [physical] USB hub, that might actually multiple USB hubs internally.  I have a 10 port USB hub that is actually represented as 8 USB hub devices (4 USB3 and 4 USB2 hubs each with 4 ports) in uhubctl.

I would be very careful trying to do anything with uhubctl automatically.  There is not requirement that a camera be plugged directly to the Pi (it could be connected to an external hub).  Also, there may be other USB devices, such as USB wifi adapters that could get cycled.

I am with @EricClaeys... this is 100% a hardware issue.

EricClaeys, 2024-08-27:
@aaronwmorris, in the next Allsky release allsky.sh only power cycles the port the ZWO camera is on:
~~~
SEARCH="${ZWO_VENDOR}:${ZWO_CAMERA_ID}"   # e.g., 03C3:385b
sudo "${ALLSKY_BIN}/uhubctl" --action off --exact --search "${SEARCH}"
sleep 3		# give it a few seconds, plus, allow the notification images to be seen
sudo "${ALLSKY_BIN}/uhubctl" --action on --exact --search "${SEARCH}"
~~~

Since ZWO's "snapshot" mode will be the default, there will be significantly fewer ASI_ERROR's so uhubctl will likely only be called for hardware problems.

EricClaeys, 2024-11-05:
Closing Issue - we believe it's a hardware problem, not software.  User was able to get his camera working again.