AllskyTeam/allsky #3833: System Messages 'capture program stopped' (every morning)

jsalsburg, 2024-08-23:
### Environment

  Computer:  RPi4
  Allsky Version:  v2023.05.01_04
  Website Version:  v2023.05.01
Linux rpi 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) arch64
8 gigs

### Bug Description
Every morning, Allsky stops capturing and does not create timelapes/keogram/startrails. I can successfully run generateForDay.sh  --timelapse YYYYMMDD etc. If I stop/start Allsky with the 'System' buttons it will stop capture after a few attempts to capture. After rebooting the RPi will continue capture until the next morning.

Repeats every morning.
<img width="1105" alt="System Messages" src="https://github.co
[allsky.log](https://github.com/user-attachments/files/16721274/allsky.log)
m/user-attachments/assets/6f40d0aa-40e5-4499-8ed5-5b586f2b404b">

### Log / configuration files

EricClaeys, 2024-08-23:
@jsalsburg, what's in `/tmp/capture_RPi_debug.txt`?

jsalsburg, 2024-08-23:
This may not be relevant, I have powered down and restarted several time today working on the camera on my workbench.

Mode selection for 2328:1748:12:P
    SRGGB10_CSI2P,1280x720/0 - Score: 5319.24
    SRGGB10_CSI2P,1920x1080/0 - Score: 3319.24
    SRGGB10_CSI2P,2328x1748/0 - Score: 1000
    SRGGB10_CSI2P,3840x2160/0 - Score: 1648.24
    SRGGB10_CSI2P,4656x3496/0 - Score: 2019
Stream configuration adjusted
Mode selection for 4656:3496:12:P
    SRGGB10_CSI2P,1280x720/0 - Score: 13471.2
    SRGGB10_CSI2P,1920x1080/0 - Score: 11471.2
    SRGGB10_CSI2P,2328x1748/0 - Score: 9152
    SRGGB10_CSI2P,3840x2160/0 - Score: 5471.24
    SRGGB10_CSI2P,4656x3496/0 - Score: 1000
Stream configuration adjusted
Still capture image received

jsalsburg, 2024-08-24:
This issue is caused by my Synology which is hosting the AllSky Remote Web Server. When the Synology is transferring files over my Network or is engaged in running its widgets and Applications, apparently the Web Service is not being serviced so the AllSky/RPi cannot push its image files to the Synology Web Server so AllSky times out. I must apologize about reporting this issue before I figured out why it is occurring. It is my fault. As my experience with AllSky increases, I have learned many things, but I am no expert and this issue proves my naivety. This issue necessitates that I create a different Remote Web Server dedicated to AllSky.

EricClaeys, 2024-08-24:
@jsalsburg, what model NAS do you have and how old is it?  My two-year old, 2-bay, one drive NAS doesn't have these problems.  Except when you are viewing a web page on the NAS, the Allsky Website doesn't add any load.  And uploading a file every minute should not cause any problems. 
My NAS can display the CPU time, disk usage, and memory.  Perhaps you can view them when your NAS is "too busy"?

No need to apologize for anything - we are both learning.

jsalsburg, 2024-08-28:
capture program stopped. This appears to happen right at end or night/begin day. Attached allsky.log.
I did not restart the RPi until after I downloaded the debug and log files.
Here is the contents of capture_RPi_debug.txt ...
Mode selection for 2328:1748:12:P
    SRGGB10_CSI2P,1280x720/0 - Score: 5319.24
    SRGGB10_CSI2P,1920x1080/0 - Score: 3319.24
    SRGGB10_CSI2P,2328x1748/0 - Score: 1000
    SRGGB10_CSI2P,3840x2160/0 - Score: 1648.24
    SRGGB10_CSI2P,4656x3496/0 - Score: 2019
Stream configuration adjusted
Mode selection for 4656:3496:12:P
    SRGGB10_CSI2P,1280x720/0 - Score: 13471.2
    SRGGB10_CSI2P,1920x1080/0 - Score: 11471.2
    SRGGB10_CSI2P,2328x1748/0 - Score: 9152
    SRGGB10_CSI2P,3840x2160/0 - Score: 5471.24
    SRGGB10_CSI2P,4656x3496/0 - Score: 1000
Stream configuration adjusted
dmaHeap allocation failure for rpicam-apps0
ERROR: *** failed to allocate capture buffers for stream ***

[allsky.log](https://github.com/user-attachments/files/16783988/allsky.log)

EricClaeys, 2024-08-28:
@jsalsburg, please create a new Issue at the [rpicam GitHub page](https://github.com/raspberrypi/rpicam-apps) and post the Issue number here.  What you are seeing isn't an Allsky Issue - it's either a hardware problem or (more likely) something in the libcamera-still application.
Let them know that a typical command execution is
~~~
libcamera-still --output '/home/rpi/allsky/tmp/image-20240828060449.jpg' --timeout 1 --nopreview --width 4656 --height 3496 --shutter 4077762 --analoggain 1 --quality 100
~~~
and that the same command will work sometimes and not other times.
If you have changed any Pi settings let them know that as well.

That team has always been very responsive.  They will likely ask for additional information.
Other people have reported this problem.

Eric

jsalsburg, 2024-08-29:
https://github.com/raspberrypi/rpicam-apps/issues/707
dmaHeap allocation failure for rpicam-apps0 #707

jsalsburg, 2024-08-30:
Again, I may be causing this issue. To forgo the limitations of the RPi4 SD card, I attached a USB/SSD to the RPi and altered AllSky to store images to the SSD instead. I may have underestimated the capacity of the SSD and included too many DAYS_TO_KEEP=30, so I changes it to 20. We will see if this solves the issue.
This may be a feature to add; a monitor and system alarm/message for /dev/sda* memory value.

EricClaeys, 2024-08-30:
Other than disk space, I can't think of anything that is impacted by a large DAYS_TO_KEEP.  The only thing that looks at the images is the `Images` page on the WebUI. I suppose if you had hundreds or thousands of days that page would take slightly longer to list all the days.

This should have no impact on the RPi buffer problem, unless having a USB disk uses a lot of the buffers mentioned in the error message.

EricClaeys, 2024-11-28:
@jsalsburg,
Hi.  What's the status of this Issue?  Is it still a problem that Allsky needs to address?  Or was it "user error" and can be closed?

I added your suggestion to [Issue 2835](https://github.com/AllskyTeam/allsky/issues/2835) which is for our "new WebUI".

jsalsburg, 2024-11-28:
This issue is user error, user ignorance, and complex hardware configuration. After replacing the RPi4, h a new 128Gig SD Card, and its Bridge Router, I noticed a few new issues that deserve mentioning. The new hardware required several days of study to get it to work reliably with my Web Server (Synology). Luckily I saved the config folder from the old camera/RPi4. This sped along getting the new RPi running but not completely reliably. Recently, the new SD Card filled up, I forgot that I had configured the old RPi to save 20 days which was saving to a USB/SSD. If the SD Card filles up, Allsky will crash. It is not apparent from the "Live View" Page that anything will soon go wrong. I feel it is important that some easy way is devised to show the RPi Memory status on the "Live View" Page, with whatever it takes; a Module/Overlay or whatever. There could be a few other things related to reliability shown on the  "Live View" or "System" Page, like a histogram of send-to-receive time of the image.jpg file written to the Remote Web Site.

nonmatrixhuman, 2025-01-17:
my camera was working fine. 
after adjusting the settings on the webui
this error appeared 
what do I need to do ? 

System Messages
*** capture_RPi: ERROR: maximum number of consecutive errors of 4 reached; capture program stopped. Total errors=4.   (4 occurrences, last on January 17, 3:28:23 pm CET)

/RPI/gedug.txt
[4:02:37.673992211] [6575] [1;31mERROR [1;37mIPARPI [1;34mipa_base.cpp:139 [0mFailed to load tuning data file NO
[4:02:37.674141273] [6575] [1;31mERROR [1;37mRPI [1;34mpipeline_base.cpp:818 [0mFailed to load a suitable IPA library
[4:02:37.674185336] [6575] [1;31mERROR [1;37mRPI [1;34mvc4.cpp:219 [0mFailed to register camera imx708_wide: -22
ERROR: *** no cameras available ***

Terminal
 libcamera-hello
Made X/EGL preview window
[0:29:57.490400216] [2208]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f
[0:29:57.659151101] [2209]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media2 and ISP device /dev/media0
[0:29:57.659261411] [2209]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[0:29:57.660472221] [2208]  INFO Camera camera.cpp:1033 configuring streams: (0) 2048x1152-YUV420
[0:29:57.661112782] [2209]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA

PI3 B+ 
Pi Camera Module 3