AllskyTeam/allsky #3852: Can a newish ZWO camera be added to the list of cameras for allsky?

j0sk0, 2024-09-06:
Hi,

I have been working on building a sky camera and have built all the components and tested it with a coupe ZWO cameras i have, one didn't work zwo asi120mm, but may be some 64 bit issue. the other was a 294mc pro which did work fine but was not the camera I want to use.

I bought an ZWO asi676mc and this camera does not appear to be recognized by allsky. I do not see it in the list of cameras in the library, so need to ask, can this be added, and what would be a turn around on a request like this? Also I'm up for any ideas if this can be got around?

I did test this camera on another device (but it was Microsoft), and it was recognized and seems to work. So I don't believe this is a camera HW issue. 

Thanks, John

Log details:
john@allsky:/tmp $ grep 'ZWO SDK' /var/log/allsky.log
2024-09-01T08:27:54.300661-05:00 allsky allsky[1667]:    ZWO SDK version 1, 32, 0, 0
2024-09-01T08:33:06.115817-05:00 allsky allsky[1713]:    ZWO SDK version 1, 32, 0, 0
2024-09-01T20:09:56.642808-05:00 allsky allsky[2598]:    ZWO SDK version 1, 32, 0, 0
2024-09-01T20:11:14.915276-05:00 allsky allsky[2918]:    ZWO SDK version 1, 32, 0, 0
2024-09-01T20:12:35.696568-05:00 allsky allsky[3271]:    ZWO SDK version 1, 32, 0, 0
2024-09-01T20:15:43.681086-05:00 allsky allsky[3833]:    ZWO SDK version 1, 32, 0, 0
2024-09-01T20:23:29.919928-05:00 allsky allsky[4822]:    ZWO SDK version 1, 32, 0, 0
2024-09-01T20:37:41.187599-05:00 allsky allsky[6446]:    ZWO SDK version 1, 32, 0, 0
2024-09-01T20:41:10.425468-05:00 allsky allsky[7021]:    ZWO SDK version 1, 32, 0, 0
2024-09-01T20:42:19.877188-05:00 allsky allsky[7318]:    ZWO SDK version 1, 32, 0, 0
2024-09-01T20:43:51.433910-05:00 allsky allsky[7664]:    ZWO SDK version 1, 32, 0, 0
2024-09-01T20:44:50.451866-05:00 allsky allsky[7952]:    ZWO SDK version 1, 32, 0, 0
2024-09-01T20:56:23.538708-05:00 allsky allsky[9244]:    ZWO SDK version 1, 32, 0, 0
2024-09-01T21:05:57.423750-05:00 allsky allsky[10356]:    ZWO SDK version 1, 32, 0, 0
2024-09-01T21:20:04.936934-05:00 allsky allsky[11883]:    ZWO SDK version 1, 32, 0, 0
2024-09-01T21:22:47.447731-05:00 allsky allsky[12294]:    ZWO SDK version 1, 32, 0, 0
2024-09-02T06:27:25.660436-05:00 allsky allsky[67927]:    ZWO SDK version 1, 32, 0, 0
2024-09-02T06:30:44.245103-05:00 allsky allsky[1704]:    ZWO SDK version 1, 32, 0, 0
2024-09-03T08:37:15.166972-05:00 allsky allsky[2430]:    ZWO SDK version 1, 32, 0, 0
2024-09-06T10:49:40.729395-05:00 allsky allsky[2456]:    ZWO SDK version 1, 32, 0, 0
john@allsky:/tmp $ uptime
 11:28:27 up 9 min,  3 users,  load average: 0.06, 0.11, 0.09
john@allsky:/tmp $ sudo systemctl status allsky
○ allsky.service - AllSky Camera Service
     Loaded: loaded (/etc/systemd/system/allsky.service; enabled; preset: enabled)
     Active: inactive (dead) since Fri 2024-09-06 11:26:02 CDT; 2min 36s ago
   Duration: 4.154s
    Process: 2236 ExecStart=/home/john/allsky/allsky.sh (code=killed, signal=TERM)
   Main PID: 2236 (code=killed, signal=TERM)
        CPU: 1.998s

Sep 06 11:26:02 allsky allsky[2414]: open: Not connected
Sep 06 11:26:02 allsky allsky[2414]: cd: Not connected
Sep 06 11:26:02 allsky allsky[2236]:      ***** AllSky Stopped *****
Sep 06 11:26:02 allsky sudo[2429]:     john : PWD=/home/john/allsky ; USER=root ; COMMAND=/usr/bin/systemctl stop allsky
Sep 06 11:26:02 allsky sudo[2429]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1000)
Sep 06 11:26:02 allsky sudo[2429]: pam_unix(sudo:session): session closed for user root
Sep 06 11:26:02 allsky systemd[1]: Stopping allsky.service - AllSky Camera Service...
Sep 06 11:26:02 allsky systemd[1]: allsky.service: Deactivated successfully.
Sep 06 11:26:02 allsky systemd[1]: Stopped allsky.service - AllSky Camera Service.
Sep 06 11:26:02 allsky systemd[1]: allsky.service: Consumed 1.998s CPU time.
john@allsky:/tmp $ grep '^6' /tmp/ZWOcameras.txt
6200MC_Pro
6200MM_Pro
662MC
662MC_Pro
664MC_DDR
664MC_Pro
664MM_DDR
664MM_Pro
678MC
678MC_Pro
678MM
john@allsky:/tmp $ grep '2024-09-06T11:2' /var/log/allsky.log
2024-09-06T11:25:58.291010-05:00 allsky allsky[2236]:      ***** Starting AllSky *****
2024-09-06T11:25:59.048776-05:00 allsky allsky[2351]: Commands used are in: /home/john/allsky/tmp/lftp_cmds/NotificationImage.txt
2024-09-06T11:25:59.048992-05:00 allsky allsky[2351]: Upload output from '/home/john/allsky/tmp/image.jpg:
2024-09-06T11:25:59.049024-05:00 allsky allsky[2351]:    *** upload.sh: ERROR, RET=99:
2024-09-06T11:25:59.049041-05:00 allsky allsky[2351]: FILE_TO_UPLOAD='/home/john/allsky/tmp/image.jpg'
2024-09-06T11:25:59.049055-05:00 allsky allsky[2351]: REMOTE_HOST=''
2024-09-06T11:25:59.049070-05:00 allsky allsky[2351]: DIRECTORY=''
2024-09-06T11:25:59.049084-05:00 allsky allsky[2351]: TEMP_NAME='NotificationImage-25577'
2024-09-06T11:25:59.049105-05:00 allsky allsky[2351]: DESTINATION_NAME='image.jpg'
2024-09-06T11:25:59.049120-05:00 allsky allsky[2351]: open: Not connected
2024-09-06T11:25:59.049136-05:00 allsky allsky[2351]: cd: Not connected
2024-09-06T11:25:59.068340-05:00 allsky allsky[2356]: *** /home/john/allsky/bin/capture_ZWO: ERROR: No Connected Camera...
2024-09-06T11:26:02.080180-05:00 allsky allsky[2356]:      ***** Stopping AllSky *****
2024-09-06T11:26:02.086803-05:00 allsky allsky[2236]: ***
2024-09-06T11:26:02.086890-05:00 allsky allsky[2236]: *** After fixing, restart the allsky service. ***
2024-09-06T11:26:02.087012-05:00 allsky allsky[2236]: ***
2024-09-06T11:26:02.423053-05:00 allsky allsky[2414]: Commands used are in: /home/john/allsky/tmp/lftp_cmds/NotificationImage.txt
2024-09-06T11:26:02.423268-05:00 allsky allsky[2414]: Upload output from '/home/john/allsky/tmp/image.jpg:
2024-09-06T11:26:02.423306-05:00 allsky allsky[2414]:    *** upload.sh: ERROR, RET=99:
2024-09-06T11:26:02.423328-05:00 allsky allsky[2414]: FILE_TO_UPLOAD='/home/john/allsky/tmp/image.jpg'
2024-09-06T11:26:02.423343-05:00 allsky allsky[2414]: REMOTE_HOST=''
2024-09-06T11:26:02.423362-05:00 allsky allsky[2414]: DIRECTORY=''
2024-09-06T11:26:02.423377-05:00 allsky allsky[2414]: TEMP_NAME='NotificationImage-13383'
2024-09-06T11:26:02.423404-05:00 allsky allsky[2414]: DESTINATION_NAME='image.jpg'
2024-09-06T11:26:02.423423-05:00 allsky allsky[2414]: open: Not connected
2024-09-06T11:26:02.423443-05:00 allsky allsky[2414]: cd: Not connected
2024-09-06T11:26:02.424233-05:00 allsky allsky[2236]:      ***** AllSky Stopped *****
john@allsky:/tmp $

EricClaeys, 2024-09-06:
@j0sk0. That camera is supported in the next Allsky release which will be out in about a month.
If you can't wait, you have two options:
1. Install the newest ZWO library yourself.  Search the Discussions and Issues - someone posted instructions. 
2. Install the new Allsky version, realizing it is still being tested, although it's working well. You will need to reinstall it when it's official.  Letme know if you want to install it.

j0sk0, 2024-09-07:
Thanks for the quick reply Eric.

I'll give option one a try, if I run into any snags, I'll reach out to you to attempt option two.

j0sk0, 2024-09-07:
Eric,

I looked at Issue #2044 which I think was the one you were referring to in your reply number 1. 

I pulled all of the current *.a files from the lib folder in the repo. and was able to install them, but I have to ask you, are you sure the asi676 is in the current repository? Because I ran strings on each file, and I don't see this camera listed in the current library. Maybe I'm missing something?

Can you confirm, and advise?

EricClaeys, 2024-09-08:
Did you copy the files from the "master" or "v2024.xx.xx" branch?
"master" is the current release so won't support the 676.
"v2024.xx.xx" supports it.

EricClaeys, 2024-09-08:
@j0sk0, did you copy the files from the "master" release or the "v2024.xx.xx" branch?   "master" is the current release, and "v2024.xx.xx" is the new release with the newer libraries.

Here's the end of the list in the new release:
~~~
485MC
485MC_Pro
492MM
492MM_Pro
533MC
533MC_Pro
533MM
533MM_Pro
585MC
585MC_Pro
6200MC_Pro
6200MM_Pro
662MC
662MC_Pro
664MC_DDR
664MC_Pro
664MM_DDR
664MM_Pro
675MC_DDR
675MC_Pro
675MM_DDR
675MM_Pro
676MC_DDR
676MC_Pro
676MM_DDR
676MM_Pro
678MC
678MC_Pro
678MM
715MC_DDR
715MC_Pro
715MM_DDR
715MM_Pro
990MM_Pro
991MM_Pro
~~~

j0sk0, 2024-09-08:
Eric,

Thanks, did not see the branches, I was looking at the "master".

I have reinstalled with the "v2024.xx.xx" library, the camera is there, and the SW did find it, so I'm in business.

Sorry for any inconvenience, appreciate the help on this.

John

EricClaeys, 2024-09-08:
@j0sk0, glad you got it working.
Eric

mattg8892, 2024-10-24:
in need of help on this same camera, im going to sound like a complete moron here haha i have limited knowledge of linux so bear with me please. 
im tring to get this running but cant make it past install because it doesnt recognize the camera it starts to install then gives me a no camera error. 
so reading above i understand that i need to somehow pull the library from the 2024.xx.xx branch but i cant not figure the process out how to replace and delete what i need to. ive been at this for about 5 hours now and have given up and hope the new version releases soon or someone can help a limited knowledge person out thank you!

closed 2024-11-05 by EricClaeys