AllskyTeam/allsky #3869: ERROR Overlay

Kilis71, 2024-09-14:
Today after a blackout, the allskycam after the reboot no longer displays the overlays, if I try to go to the overlay module, it tells me: Error generating sample data. Please ensure the overlay module is enabled.
but the modules are enabled

Alex-developer, 2024-09-14:
I will have a look into it for you

Alex-developer, 2024-09-16:
What kernel version are you running, type

uname -a

It does seem that there is an issue with a Kernel update

BobW55, 2024-09-18:
uname -a
Linux allsky 6.6.47+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.47-1+rpt1 (2024-09-02) aarch64 GNU/Linux

Kilis71, 2024-09-29:
not being able to solve the problem, I reinstalled everything and did not do the updates

EricClaeys, 2024-09-30:
@Kilis71, did the reinstall fix things?

Kilis71, 2024-09-30:
yes, because I didn't do the OS updates

EricClaeys, 2024-10-01:
Closing Issue, problem solved by reinstalling.

closed 2024-10-01 by EricClaeys