AllskyTeam/allsky #3876: [BUG] upgrade.sh fails on getBranch command

richiejarvis, 2024-09-20:
### Camera model

HD Picam

### Operating System

Bullseye

### Allsky version

v2023.05.01_05

### Pi and amount of memory (512 MB, 1, 2, 4, or 8 GB)

Pi 4 8GB

### Bug Description

When running upgrade.sh, the following error occurs:

richie@allsky:~/git/allsky $ ./upgrade.sh
./upgrade.sh: line 115: get_branch: command not found
./upgrade.sh: line 131: getBranch: command not found

EricClaeys, 2024-09-20:
Do not run `upgrade.sh`.  It is not finished yet.
Please see the documentation for how to upgrade.

EricClaeys, 2024-10-03:
@richiejarvis, any update?
If the problem is resolved, please close this Issue.
Thanks - Eric

EricClaeys, 2024-11-05:
Closing Issue.  The "upgrade.sh" command isn't completed.

closed 2024-11-05 by EricClaeys