AllskyTeam/allsky #3884: [BUG] Image status at top of live view screen incorrect;

BobW55, 2024-09-25:
### Camera model

ZWO 178mc

### Operating System

Bullseye

### Allsky version

V2024.xx.xy

### Pi and amount of memory (512 MB, 1, 2, 4, or 8 GB)

4b 8gb

### Bug Description

Status at top of live screen =
Daytime images updated every 4 seconds, nighttime every 13 seconds

It is night time, camera currently taking 60 second images.  there is a 3000 or 3 second delay set 
it should show nighttime every 63 seconds.

[settings.txt](https://github.com/user-attachments/files/17124550/settings.txt)

BobW55, 2024-09-25:
Yes I know it will vary .

EricClaeys, 2024-09-26:
@BobW55, with "Consistent Delay" disabled, the best I can do is append the word "minimum" to the message.  For example:
[image]

I just changed the code for when Auto-Exposure is enabled and Consistent Delays is disabled (your case).  Instead of using the manual exposure or Max Auto-Exposure, it uses 1 second for daytime and 10 seconds for nighttime, and says "minimum" like the screenshot.  Daytime exposure are usually less than a second, and nighttime exposures are usually at least 10 seconds, so I used those numbers.  In an ideal world the code would use what the actual length of the last exposure, but that's too much work and I think a lot of people will turn that message off anyway.

closed 2024-09-26 by EricClaeys

BobW55, 2024-09-26:
That is fine.   I dont really pay attention to that header anyway.