AllskyTeam/allsky #3906: [BUG] cc1plus: error: ‘-mfloat-abi=hard’: selected architecture lacks an FPU

eat-sleep-code, 2024-10-07:
### Camera model

Raspberry Pi HQ

### Operating System

Bookworm

### Allsky version

latest?

### Pi and amount of memory (512 MB, 1, 2, 4, or 8 GB)

Pi Zero W

### Bug Description

After an SD card failure, I am reinstalling AllSky on a fresh card.   As it is a new install, I went with Bookworm (32-bit).  I am using the instructions here: https://htmlpreview.github.io/?https://raw.githubusercontent.com/thomasjacquin/allsky/master/html/documentation/installations/Allsky.html

However... install fails with...

```
* Installing the web server.
* Installing dependencies.
* Preparing Allsky commands.

*** ERROR:
**********
Compile failed
**********

* The full log file is in /home/pi/allsky/config/logs/make_all.log
The end of the file is:
        make[1]: Entering directory '/home/pi/allsky/src'
        Building capture_ZWO.o ...
        cc1plus: error: ‘-mfloat-abi=hard’: selected architecture lacks an FPU
        make[1]: *** [Makefile:135: capture_ZWO.o] Error 1
        make[1]: Leaving directory '/home/pi/allsky/src'
        make: *** [Makefile:24: all] Error 2
```

EricClaeys, 2024-10-08:
@Alex-developer, any ideas?

@eat-sleep-code, please run `uname -a`.

eat-sleep-code, 2024-10-11:
@EricClaeys 

Linux allsky 6.6.51+rpt-rpi-v6 #1 Raspbian 1:6.6.51-1+rpt2 (2024-10-01) armv6l GNU/Linux

EricClaeys, 2024-10-11:
@eat-sleep-code, would you please run `cat /etc/os-release` ?
I'm trying to find something that will indicate your Pi needs different options.

Alex-developer, 2024-10-11:
Will give this a test on a zero 2 and see if I can fix it

Alex-developer, 2024-10-12:
I have tested on the following kernels on a Pi zero 2 W

64 Bit kernels
Linux zero 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux - OK
Linux zero 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux - OK

32 Bit Kernels
Linux zero 6.6.31+rpt-rpi-v7 #1 SMP Raspbian 1:6.6.31-1+rpt1 (2024-05-29) armv7l GNU/Linux - OK
Linux zero 6.6.51+rpt-rpi-v7 #1 SMP Raspbian 1:6.6.51-1+rpt3 (2024-10-08) armv7l GNU/Linux - OK

I'll need to do some more testing.

@eat-sleep-code Is this a clean install of 32bit Bookwork? Also is it Lite or desktop?

Thanks

Alex

eat-sleep-code, 2024-10-12:
Mine is a Pi Zero W (first-gen) running a 32-bit version of Raspberry Pi OS Lite (Bookworm)

EricClaeys, 2024-10-12:
@eat-sleep-code, we don't recommend that Pi - you won't be able to make a timelapse and maybe not even startrails or keograms.
The rest of Allsky will probably work, although very slowly.

eat-sleep-code, 2024-10-12:
I will order a second generation and see if I get any further

Alex-developer, 2024-10-12:
> I will order a second generation and see if I get any further

The installation and capture side all ran fine for me a on a Zero 2 w. I did not test any of the timelapse or startrail generation.

It was VERY slow, not a great surprise, at doing anything

Alex

closed 2024-10-25 by Alex-developer