AllskyTeam/allsky #3907: [BUG] Error when using GPIO on Rasberry PI 5

splanquart, 2024-10-08:
### Camera model

Rpi HD

### Operating System

Bookworm

### Allsky version

v2023.05.01_05

### Pi and amount of memory (512 MB, 1, 2, 4, or 8 GB)

rpi 5 with 8Go of RAM

### Bug Description

I try to use module dewheater without success. In other hand i was some probleme to read temperature with DHT11 too (but it is not the subject of this thread, juste i think it can be corralated by the issue with GPIO).

I found in logfile `/var/log/allskyperiodic.log` : 
`ERROR: Module allsky_dewheater.py failed on line 224 - Cannot determine SOC peripheral base address`
In this file, at this line there is nothing about SOC ou GPIO.

I search some reason about this bug and not well understand that but rasberry pi 5 was changed lot of stuff and now we need to change som library to use GPIO on rasberry pi 5.

For example I can switch on my drew by this code : 
```
import gpiod
chip = gpiod.Chip('gpiochip0')
pin=18
drew_line = chip.get_line(pin)
drew_line.request(consumer="Drew", type=gpiod.LINE_REQ_DIR_OUT)
drew_line.set_value(1)
```

In other way I found some stuff about this error in other allsky projet : https://github.com/aaronwmorris/indi-allsky/wiki/Dew-Heater-Support#raspberry-pi-5
Where they recommande to uninstall `RPi.GPIO`and  `rpi.lgpio` and reinstall `rpi.lgpio`.

But my `venv` for allsky don't use this `rpi.lgpio` therefore I don't know if this trick can work good for allsky.
```
(venv) steph@sky:~ $ pip freeze|grep gpio
gpiozero==2.0
lgpio==0.2.2.0
pigpio==1.78
```
 
There noting more in log when i passe to debug level 4.

_Note: I know that my problem occur on module but when I read the code I understand that GPIO was supported by the core of allsky projet. It's why I post this big here._

Alex-developer, 2024-10-11:
Was this a clean installation on the pi5? There should be no reason to change anything as the installation is designed to work on a pi5.

Could you do the following

source ~/allsky/venv/bin/activate
pip list

And paste the output

Thanks

Alex

splanquart, 2024-10-12:
Yes it is a clean install. I was this problem. To try to found a fix I make an install of the last version.. And the problem continue.

[list.txt](https://github.com/user-attachments/files/17349711/list.txt)

esenbaytekin, 2024-10-17:
Similarly, I am using Raspberry Pi 5 (4Gb).

Version: v2023.05.01_05
Linux AllSky-RPI5 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64
Pi Revision Raspberry Pi 5 Model B Rev 1.0 (4 GB)
64-Bit Bookworm, no desktop environment

The heater works without any problems. I am adding the venv list. There are differences. I am using is SHT31 module. (i2c 0x44) 
[venv_list.txt](https://github.com/user-attachments/files/17408780/venv_list.txt)

Have you updated RPI other than the normal updates?

splanquart, 2024-10-20:
Thank @esenbaytekin, your venv_list.txt help me to found error.
I don't know why but there is little difference between us two venv : 
- gpiozero 2.0.1  but 2.0.0 for me
- rpi-lgpio 0.6.0 but nothing for me

I found also that I have debian package installed : 
- 

I upgrade my venv and clean my host of gpio debian package: 
```
. ./allsky/venv/bin/activate
pip install gpiozero==2.0.1
sudo apt-get remove python3-rpi.gpio python3-lgpio python3-gpiozero python3-pigpio
pip install rpi.lgpio
```

after that, i reboot and all is good.

EricClaeys, 2024-10-20:
@Alex-developer, do we need to change anything based on this Discussion?

Alex-developer, 2024-10-20:
I'll do some checks on the pi4 and 5. Updating to lgpio makes sense on the pi 5, with the latest kernel, but I have never tested it on older kernels or the pi 4

Alex-developer, 2024-10-23:
I can reproduce this with the latest kernel on the pi5 - Working on a fix now

Alex-developer, 2024-10-25:
Closing this as fixed under https://github.com/AllskyTeam/allsky/issues/3923

closed 2024-10-25 by Alex-developer