AllskyTeam/allsky #3913: [BUG] Documentation outdated: constellation overlay

markusd112, 2024-10-11:
### Camera model

Raspi HQ

### Operating System

Buster

### Allsky version

v2023.05.01_05

### Pi and amount of memory (512 MB, 1, 2, 4, or 8 GB)

Raspi 4, 4GB

### Bug Description

This bug reports is about the AllSky documentation:

https://htmlpreview.github.io/?https://raw.githubusercontent.com/AllskyTeam/allsky/master/html/documentation/explanations/constellationOverlay.html

The information about activating constellation overlay seems not to be up to date: if I go to the editor and try to select the filename in the drop down, there is no "configuration.json" available...

Additionally the link in the following text doesn't work (404 - file not found):

For more information on this, and other Website settings, visit the [Allsky Website Settings](https://htmlpreview.github.io/?https://raw.githubusercontent.com/AllskyTeam/allsky/master/htmlhttps://htmlpreview.github.io/?https://raw.githubusercontent.com/documentation/settings/Website.html) page.

Alex-developer, 2024-10-11:
Thanks we will take a look

markusd112, 2024-10-17:
Some update: I think it was my fault: I tried to activate constellations before pushing the basic website and configure the upload correctly. After that the configuration.json is available in the Web Editor.

The problem with the broken links to other pages of the documentation is still there and occurs on all pages of the documentation with links.

EricClaeys, 2024-10-17:
@markusd112, glad you got it working.
The dead links are fixed in the next release.

I'm closing this Issue since you resolved it.

closed 2024-10-17 by EricClaeys