AllskyTeam/allsky #3919: [FEATURE REQUEST] Add documentation for FTPeS (FTP with explicite SSL)

markusd112, 2024-10-16:
### If this is a new feature, describe what you would like.

Please add some informations about FTPS using explicite SSL in the AllSky documentation

e.g. in the troubleshooting section?
https://htmlpreview.github.io/?https://raw.githubusercontent.com/AllskyTeam/allsky/master/html/documentation/troubleshooting/uploads.html

When setting PROTOCOL="ftps" the upload is done by lftp via FTP with implicite SSL (normally via tcp/990, tcp/989). FTP with implicite SSL is outdated, so you should use FTP with explicite SSL (FTPeS) instead. If you want to upload your data with FTP via explicite SSL you have to configure:
PROTOCOL="ftp"
LFTP_COMMANDS="ftp:ssl-force yes"

FTPeS uses opens the initial session via FTP without encryption. After negotian AUTH_TLS with the FTP-server a TLS session is initiated and your username and password are encrypted. The "ftp:ssl-force yes" option refuses to send password in clear when server does not support SSL.

To check the communication and the use of TLS you can enable debugging via:
LFTP_COMMANDS="debug 999999;ftp:ssl-force yes"

EricClaeys, 2024-11-28:
@markusd112,
Thanks for the suggestion.  I've put it down for v2026.12.06_01, which is the first point release in the major release that comes out next week.

If you'd like to help with this, take a look at the new version of the page [here](https://htmlpreview.github.io/?https://raw.githubusercontent.com/AllskyTeam/allsky/v2024.12.06/html/documentation/troubleshooting/uploads.html), then do a Pull request against the file in the v2024.12.06 branch, including your suggested changes.

One thing to note is the v2024.12.06 got rid of the `config.sh` and `ftp-settings.sh` files and moved all their settings to the Allsky Settings WebUI page.  That page should probably mention adding `set ftp:ssl-force yes` in the "FTP Commands" field (what was LFTP_COMMANDS).  Or maybe it should be there by default?  Does it also work for the "ftp" and "sftp" protocols?

markusd112, 2024-11-28:
@EricClaeys Thanks, I will try to do a pull request with some suggestion, but I am not so familiar with git in the moment. I'll try to learn.

Yes, "ftp:ssl-force" should be set by default, because transferring credentials without encryption shouldn't be used any longer. If someones ftp server doesn't support it, the ftp connection will be rejected.

FTPeS uses FTP for the initiation of the ftp session (without encryption). Then it activates TLS/SSL encryption and sends username/password encrypted. So you have to set "ftp" as protocol for using encrypted FTPeS. If you set "ftps", the lftp software uses FTPiS (implicit SSL) which is outdated. So it is important for the allsky admin that he set protocol to "ftp" and "ftp:ssl-force on" to use encrypted FTPeS.

SFTP doesn't use SSL encryption, it uses SSH instead, so the "ssl-force" command isn't necessary for sftp. It uses ssh encryption by default and only encrypted connections.

markusd112, 2024-11-28:
I have tried to create a "Pull request", but I am not allowed to do: "Pull request creation failed. Validation failed: must be a collaborator "

I have add some lines for using FTP with encryption
https://github.com/AllskyTeam/allsky/blob/v2024.12.06/html/documentation/troubleshooting/uploads.html

I am not allowed to post a .html file here... Find it below in plain text:

`<!DOCTYPE html>
<html lang="en">
<head>
	<meta charset="utf-8">
	<meta http-equiv="X-UA-Compatible" content="IE=edge">
	<meta name="viewport" content="width=device-width, initial-scale=1">
	<script src="../js/documentation.js" type="application/javascript"></script>
	<link href="../css/light.css" rel="stylesheet">
	<link href="../bower_components/bootstrap/dist/css/bootstrap.min.css" rel="stylesheet">
	<script src="../bower_components/jquery/dist/jquery.min.js"></script>
	<script src="../bower_components/bootstrap/dist/js/bootstrap.min.js"></script>
	<style>
		#pageTitle::before {
			content: "Troubleshooting Upload Problems";
		} 
	</style>
	<link href="../css/documentation.css" rel="stylesheet">
	<link href="../documentation-favicon.ico" rel="shortcut icon" type="image/png">
	<script src="../js/all.min.js" type="application/javascript"></script>
	<title>Upload Problems</title>
</head>
<body>
<div w3-include-html="/documentation/pageHeader.html" id="pageHeader"></div>
<div class="Layout">
<div class="Layout-sidebar" w3-include-html="/documentation/sidebar.html" id="sidebar"></div>
<div class="Layout-main markdown-body" id="mainContents">

<p>
<b>Uploading</b> a file means copying it to a different computer.
Allsky supports several ways to upload files including
<code>ftp</code>, <code>scp</code>, and others.
</p>

<h3>Testing an upload</h3>
<p>
After you've configured settings such as
<span class="WebUISetting">Server Name</span>
and
<span class="WebUISetting">Protocol</span>
for a remote Website or remote server,
run one of the commands below, depending on whether you are testing uploads
to a remote Allsky Website or a remote server:
<pre>
testUpload.sh --website
<span class="pl-c"> &nbsp; &nbsp; OR</span>
testUpload.sh --server
</pre>
</p>

<p>
This attempts to upload a file and if it fails,
it tries to determine why it failed and display how to fix it.
</p>
<p>
If that doesn't help, look at the sections below for additional information.
</p>

<h3>Using encryption for FTP</h3>
<p>FTP sends your credentials (username and password) without encryption, so anybody can potentially read it and grant access to your server. To encrypt your username and password you can use the following options:</p>
<h4>FTPiS - FTP with Implicit SSL</h4>
<p>Data- and control-connection between ftp-client and -server are split to different tcp-ports and ssl encrypted. To activate FTPiS set protocol to "ftps":
<pre>
protocol = "ftps"
[... 80 lines cut ...]
	<span class="fileName">~/.lftprc</span> file as above.
</ol>
If your Pi's IP address changed, or the IP address remained but the
machine using the addressed changed, you probably need to remove the previous key from your
<span class="fileName">~/.ssh/known_hosts</span> file.
If the file has only one entry, simply remove it.
If the file has multiple entries, make a backup of the file,
then delete one entry at a time until you find the "bad" one.

<h3>Missing directories on the remote Website or server</h3>
<p>
Review the 
<a allsky="true" external="true"
href="/documentation/installations/AllskyWebsite.html">Allsky Website installation instructions</a>
or the 
<a allsky="true" external="true"
href="/documentation/installations/RemoteServer.html">Remote server installation instructions</a>
to determine how to create the necessary directories.
</p>

</div>
</div>
</body>
</html>
<script> includeHTML(); </script>`

EricClaeys, 2025-01-16:
@aaronwmorris,
@Alex-developer,

Do you agree with the comments in the first post?
That the `ftps` Protocol is "outdated" and `ftp` with `set ftp:ssl-force yes`  should be used instead?

aaronwmorris, 2025-01-16:
Implicit FTP (ftps://) using port 990 and 989 (analogous to 21/20) appears to not be very commonly supported.  Implicit FTP was probably doomed from the start because with unencrypted FTP, firewalls have to actively eavesdrop on the session to dynamically open ports for transfers to work.  There's no way to eavesdrop on an encrypted session.

https://datatracker.ietf.org/doc/html/draft-murray-auth-ftp-ssl-07#appendix-A

EricClaeys, 2025-01-16:
@aaronwmorris,
does that mean you agree with the comments in the first post and that `ftp` should be the default Protocol along with using `set ftp:ssl-force yes` ?

How does Indi-Allsky handle this?

aaronwmorris, 2025-01-16:
At this point, forcing SSL with FTP is probably a safe bet.  Most providers likely support this, although there may be a few edge cases where it might cause a problem.

In indi-allsky, I have all of the file transfer functionality implemented programmatically using pycurl, so I have separate protocol selections for unencrypted and encrypted FTP.  I had to figure out all of the differences between implicit and explicit encrypted FTP which is the only reason I had any understanding of the differences.

markusd112, 2025-02-02:
I have seen today some check message after an upgrade of the allsky software: "Remote Website Protocol set to insecure 'ftp'.
FIX: Using 'ftps' or 'sftp' if possible."

It should be added some information to add "LFTP_COMMANDS="set ftp:ssl-force yes"" to secure the ftp transfer.

[image]

EricClaeys, 2025-02-02:
@markusd112, that's part of what will be updated when we implement the change in this Issue.