AllskyTeam/allsky #3935: [BUG] Can't install even when the camera is detected (IMX283)

krzysztofkuczek, 2024-11-03:
### Camera model

OneInchEye IMX283

### Operating System

Bookworm

### Allsky version

v2023.05.01_05

### Pi and amount of memory (512 MB, 1, 2, 4, or 8 GB)

RPI5 88GB

### Bug Description

The camera is connected and working fine (tested with rpicam-hello).
It's also detected during first phase of installation process, but then the setup fails.

kku@rpi5poe:~/allsky $ ./install.sh

********************************************************
*** Welcome to the Allsky Installer - v2023.05.01_05 ***
********************************************************

* RPi camera found.
* Using RPi camera.
* Size of current swap (200 MB) is sufficient; no change needed.
* /home/kku/allsky/tmp is now in memory.

*** NOTICE:
**********
The following steps can take up to an hour depending on the speed of your Pi
and how many of the necessary dependencies are already installed.
You will see progress messages throughout the process.
At the end you will be prompted again for additional steps.
**********

* Installing the web server.
* Installing dependencies.
* Preparing Allsky commands.
* Modifying locations for WebUI.
* Setting up WebUI options and default settings for RPi cameras.
* Making new settings file '/home/kku/allsky/config/settings.json'.

*** ERROR:
**********
No camera detected - installation aborted.

EricClaeys, 2024-11-03:
Allsky doesn't support the OneInchEye IMX283 camera.
If it's important that it be supported, please run the following:
~~~
echo "=========== libcamera-still --list-cameras output:"
libcamera-still --list-cameras
echo "=========== libcamera-still --metadata output:"
libcamera-still -v --metadata - --immediate 0 --nopreview --thumb none --timeout 1 --shutter 1 --output /dev/null
~~~
and send me the output.

krzysztofkuczek, 2024-11-03:
That's the output, please check attached file.
Thank you in advance
Kris
[imx283_out.txt](https://github.com/user-attachments/files/17611896/imx283_out.txt)

EricClaeys, 2024-11-04:
@krzysztofkuczek, thanks for the info.  Please do the following:
~~~
cd ~/allsky/src
mv ASI_functions.cpp ASI-functions.cpp-SAVED
# copy the attached file to this directory
mv ASI_functions.cpp.txt ASI_functions.cpp
sudo systemctl stop allsky
make && cp capture_RPi ../bin
sudo systemctl start allsky
~~~
Try the installation again.  The attached file has info on the imx283 camera.  Let me know how it goes.

[ASI_functions.cpp.txt](https://github.com/user-attachments/files/17613288/ASI_functions.cpp.txt)

krzysztofkuczek, 2024-11-04:
hi @EricClaeys ,
Thank you for your support. Everything seems to be ok.
Have a nice day.
Kris

EricClaeys, 2024-11-05:
Closing Issue - @krzysztofkuczek has the IMX283 working, and that camera is supported in the next release.

PR #3942 (2024-11-05): Support imx283
Fixes #3935
closed 2024-11-05 by EricClaeys