AllskyTeam/allsky #3940: [BUG] Module allsky_overlay.py failed

timg-ctrl, 2024-11-04:
### Camera model

ASI676MC

### Operating System

Buster

### Allsky version

v2024.xx.xy

### Pi and amount of memory (512 MB, 1, 2, 4, or 8 GB)

8GB

### Bug Description

Error with aalsky_overlay.py

2024-11-04T16:49:29.451947+00:00 allsky allsky[2867]: STARTING EXPOSURE at: 2024-11-04 16:49:29   @ 168.63 ms (0.17 sec)
2024-11-04T16:49:33.039492+00:00 allsky allsky[2994]: ERROR: Module allsky_overlay.py failed on line 209 - Expecting ',' delimiter: line 278 column 25 (char 7669)

EricClaeys, 2024-11-04:
Did this just start happening?  Did it ever work?  If so, did something change?
Can you attach ~/allsky/scripts/modules/allsky_overlay.py?
You'll need to append `.txt` to its name.

Also, please change `Debug Level` in the WebUI to 4, and if it happens again, show the relevant section of the log file.

timg-ctrl, 2024-11-06:
Hi,

Yes, this started happening from a new clone. No changes have been made.

Log
`2024-11-06T11:04:43.520866+00:00 allsky allsky[142237]: INFO: Loading /home/tim/allsky/config/settings.json
2024-11-06T11:04:43.520949+00:00 allsky allsky[142237]: INFO: ===== Running day flow...
2024-11-06T11:04:43.520969+00:00 allsky allsky[142237]: INFO: --------------- Running Module allsky_loadimage ---------------
2024-11-06T11:04:43.520985+00:00 allsky allsky[142237]: INFO: Image /home/tim/allsky/tmp/image-20241106110440.jpg Loaded
2024-11-06T11:04:43.521003+00:00 allsky allsky[142237]: INFO: --------------- Running Module allsky_overlay ---------------
2024-11-06T11:04:43.521018+00:00 allsky allsky[142237]: INFO: Time of day is DAY using overlay overlay1-ZWO_ASI676MC-3552x3552-both.json
2024-11-06T11:04:43.521035+00:00 allsky allsky[142237]: ERROR: Module allsky_overlay.py failed on line 209 - Expecting ',' delimiter: line 278 column 25 (char 7669)
2024-11-06T11:04:43.521054+00:00 allsky allsky[142237]: INFO: --------------- Running Module allsky_saveimage ---------------
2024-11-06T11:04:43.521073+00:00 allsky allsky[142237]: INFO: Image /home/tim/allsky/tmp/image-20241106110440.jpg Saved, quality 95
2024-11-06T11:04:43.521091+00:00 allsky allsky[142237]: INFO: ===== day flow complete.
2024-11-06T11:06:12.383915+00:00 allsky allsky[118088]: Executing sunwait poll exit angle -6.0000 50.3718N 2.2829W > /dev/null
2024-11-06T11:06:12.385399+00:00 allsky allsky[118088]: -----
2024-11-06T11:06:12.385444+00:00 allsky allsky[118088]: STARTING EXPOSURE at: 2024-11-06 11:06:12   @ 983 us (0.983 ms)
2024-11-06T11:06:12.385696+00:00 allsky allsky[118088]:       > Doing initial usleep(500,983) for exposure time 983.
2024-11-06T11:06:13.366426+00:00 allsky allsky[118088]:       > Did usleep(49) 3078 times in loop for total usleep() of 651,805us
2024-11-06T11:06:13.445711+00:00 allsky allsky[118088]:       > Time to take exposure=1,059,680 us, diff_us=1,058,697
2024-11-06T11:06:13.446305+00:00 allsky allsky[118088]:   > GOT IMAGE @ mean 0.512, (auto) gain 0.
2024-11-06T11:06:13.447314+00:00 allsky allsky[118088]:   > Good image: mean 0.512 within range of 0.480 to 0.520 +++++++++
2024-11-06T11:06:13.447348+00:00 allsky allsky[118088]:   > Sleeping: 90.0 sec
2024-11-06T11:06:13.447373+00:00 allsky allsky[118088]:   > Saving DAY image 'image-20241106110612.jpg'
2024-11-06T11:06:13.563098+00:00 allsky allsky[118088]:   > Image took 114.0 ms to save (average 128.6 ms).
2024-11-06T11:06:13.565095+00:00 allsky allsky[142277]: saveImage.sh DAY /home/tim/allsky/tmp/image-20241106110612.jpg DATE=20241106 TIME=110612 AUTOEXPOSURE=0 sAUTOEXPOSURE= EXPOSURE_US=983 sEXPOSURE=983 us (0.983 ms) AUTOGAIN=1 sAUTOGAIN=(auto) GAIN=0 AUTOWB=0 sAUTOAWB= WBR=55 WBB=75 MEAN=0.511719 TEMPERATURE_C=23 TEMPERATURE_F=73 BIN=1 FLIP=none BIT_DEPTH=8 DARKFRAME=0 eOVERLAY=1 AUTOUSB=1 USB=50
2024-11-06T11:06:14.574573+00:00 allsky allsky[142311]: INFO: Loading /home/tim/allsky/config/settings.json
2024-11-06T11:06:14.574666+00:00 allsky allsky[142311]: INFO: ===== Running day flow...
2024-11-06T11:06:14.574687+00:00 allsky allsky[142311]: INFO: --------------- Running Module allsky_loadimage ---------------
2024-11-06T11:06:14.574703+00:00 allsky allsky[142311]: INFO: Image /home/tim/allsky/tmp/image-20241106110612.jpg Loaded
2024-11-06T11:06:14.574720+00:00 allsky allsky[142311]: INFO: --------------- Running Module allsky_overlay ---------------
2024-11-06T11:06:14.574735+00:00 allsky allsky[142311]: INFO: Time of day is DAY using overlay overlay1-ZWO_ASI676MC-3552x3552-both.json
2024-11-06T11:06:14.574750+00:00 allsky allsky[142311]: ERROR: Module allsky_overlay.py failed on line 209 - Expecting ',' delimiter: line 278 column 25 (char 7669)
2024-11-06T11:06:14.574767+00:00 allsky allsky[142311]: INFO: --------------- Running Module allsky_saveimage ---------------
2024-11-06T11:06:14.574783+00:00 allsky allsky[142311]: INFO: Image /home/tim/allsky/tmp/image-20241106110612.jpg Saved, quality 95
2024-11-06T11:06:14.574799+00:00 allsky allsky[142311]: INFO: ===== day flow complete.
2024-11-06T11:07:43.446749+00:00 allsky allsky[118088]: Executing sunwait poll exit angle -6.0000 50.3718N 2.2829W > /dev/null
2024-11-06T11:07:43.448191+00:00 allsky allsky[118088]: -----
2024-11-06T11:07:43.448408+00:00 allsky allsky[118088]: STARTING EXPOSURE at: 2024-11-06 11:07:43   @ 983 us (0.983 ms)
2024-11-06T11:07:43.448495+00:00 allsky allsky[118088]:       > Doing initial usleep(500,983) for exposure time 983.
2024-11-06T11:07:44.434350+00:00 allsky allsky[118088]:       > Did usleep(49) 3085 times in loop for total usleep() of 652,148us
2024-11-06T11:07:44.514525+00:00 allsky allsky[118088]:       > Time to take exposure=1,065,790 us, diff_us=1,064,807
2024-11-06T11:07:44.516026+00:00 allsky allsky[118088]:   > GOT IMAGE @ mean 0.434, (auto) gain 0.
2024-11-06T11:07:44.517231+00:00 allsky allsky[118088]:     > Next exposure change: 181 us (214 pre agression) to 1,164 (* 1.218) [CG.lastExposure_us=983, minAcceptableMean=0.480, CG.lastMean=0.434]
2024-11-06T11:07:44.517364+00:00 allsky allsky[118088]:     >> Retry 1 @ 1,164 us, min=983 us, max=2,000,000,000 us
2024-11-06T11:07:44.523900+00:00 allsky allsky[118088]:       > Doing initial usleep(501,164) for exposure time 1,164.
2024-11-06T11:07:45.510330+00:00 allsky allsky[118088]:       > Did usleep(58) 2959 times in loop for total usleep() of 672,786us
[... cut ...]
2024-11-06T11:07:45.682164+00:00 allsky allsky[142366]: saveImage.sh DAY /home/tim/allsky/tmp/image-20241106110743.jpg DATE=20241106 TIME=110743 AUTOEXPOSURE=0 sAUTOEXPOSURE= EXPOSURE_US=1164 sEXPOSURE=1,164 us (1.164 ms) AUTOGAIN=1 sAUTOGAIN=(auto) GAIN=0 AUTOWB=0 sAUTOAWB= WBR=55 WBB=75 MEAN=0.503906 TEMPERATURE_C=24 TEMPERATURE_F=75 BIN=1 FLIP=none BIT_DEPTH=8 DARKFRAME=0 eOVERLAY=1 AUTOUSB=1 USB=50`

[
[allsky_overlay.txt](https://github.com/user-attachments/files/17646594/allsky_overlay.txt)
](url)

EricClaeys, 2024-11-06:
Please attach, do not copy/paste, the `overlay1-ZWO_ASI676MC-3552x3552-both.json` file.  It should be in `~/allsky/config/overlay/myTemplates`.

timg-ctrl, 2024-11-06:
[overlay1-ZWO_ASI676MC-3552x3552-both.json](https://github.com/user-attachments/files/17652659/overlay1-ZWO_ASI676MC-3552x3552-both.json)

Alex-developer, 2024-11-06:
> [overlay1-ZWO_ASI676MC-3552x3552-both.json](https://github.com/user-attachments/files/17652659/overlay1-ZWO_ASI676MC-3552x3552-both.json)

That file is fine and not causing the issue. Can you please have a look in `~/allsky/config/overlay/config` and see if the file exists there

Alex

EricClaeys, 2024-11-06:
@Alex-developer, what file is causing this error?
~~~
INFO: Time of day is DAY using overlay overlay1-ZWO_ASI676MC-3552x3552-both.json
ERROR: Module allsky_overlay.py failed on line 209 - Expecting ',' delimiter: line 278 column 25 (char 7669)
~~~

Are you thinking there is a corrupted `overlay1-ZWO_ASI676MC-3552x3552-both.json` in ~/allsky/config/overlay/config?

Alex-developer, 2024-11-06:
> @Alex-developer, what file is causing this error?
>
> ```
> INFO: Time of day is DAY using overlay overlay1-ZWO_ASI676MC-3552x3552-both.json
> ERROR: Module allsky_overlay.py failed on line 209 - Expecting [...]

 I suspect the error is invalid json and I suspect that the file is in ~/allsky/config/overlay/config as the file provided is fine.

timg-ctrl, 2024-11-07:
Hi,

Please see the image of the config folder. Do I need overlay1-ZWO_ASI676MC-3552x3552-both.json in the root of this folder?

[image]

EricClaeys, 2024-11-07:
@timg-ctrl, would you please look for the file in `~/allsky/config/overlay/config`, not in `~/allsky/config` ?  If the file's there, please attach it.  We think that may be the problem.

timg-ctrl, 2024-11-07:
Please see image below
[image]

Alex-developer, 2024-11-07:
> Please see image below <img alt="image" width="440" src="https://private-user-images.githubusercontent.com/183402463/383958754-166182a6-13d7-4ecb-b94e-c2b99e9fb9ae.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzEwMDMyNTAsIm5iZiI6MTczMTAwMjk1MCwicGF0aCI6Ii8xODM0MDI0NjMvMzgzOTU4NzU0LTE2NjE4MmE2LTEzZDctNGVjYi1iOTRlLWMyYjk5ZTlmYjlhZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEwN1QxODA5MTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xMDljYjlkYzA2ZmEzOTQ0ZWMxNTUzYzVkYTM3NDQ2MGU1OWEzM2NhMjZiNDRjMDY3ZWIxOWJmMTcxYzAwMzgwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.IuL-xTlek7m9zmaG3x23v7mUYowC9N8tZMPssIVM_7E">

Ok its not in there !

Can you please zip the ~/allsky/config/overlay folder and add it to this issue please. I will then test it here

Thanks

Alex

timg-ctrl, 2024-11-07:
[overlay.zip](https://github.com/user-attachments/files/17669447/overlay.zip)

EricClaeys, 2024-11-09:
@timg-ctrl,
I fixed it.  Some bozo named Eric forgot to add a double quote in a JSON configuration file.

The fix is simple:  Edit `~/allsky/config/overlay/config/fields.json`.  Add a double quote before `10` on line 278 so it looks like this:

[image]

Thanks for reporting this and being patient while we fixed it!

commit 25b3e7d4 (2024-11-09): Update fields.json: add missing double quote

Fixes #3940
closed 2024-11-09 by EricClaeys