AllskyTeam/allsky #3943: [BUG] Exposure won't go above 969.25 ms (0.97 s)

regnets, 2024-11-06:
### Camera model

Raspberry Pi v1.3

### Operating System

Bookworm

### Allsky version

v2023.05.01_05

### Pi and amount of memory (512 MB, 1, 2, 4, or 8 GB)

4

### Bug Description

As the title says my exposure won't go above 969.25 ms.

I tried to use the "Auto Exposure" feature which adjust the exposure during the day, however during night it won't go above 969.25 ms. Even after disabling "Auto Exposure" and fixing the exposure to 15000ms or 30000ms it is still 969.25 ms during night.

### Log / configuration files

[allsky.log](https://github.com/user-attachments/files/17644029/allsky.log)
[settings.json.txt](https://github.com/user-attachments/files/17644036/settings.json.txt)

EricClaeys, 2024-11-06:
@regnets, the Version 1 RPi camera with the ov5647 chip has a maximum exposure of 969249 us which is 0.969249 seconds or 969.249 ms.  This means it's not useful as an allsky camera, and we don't even support it in the next Allsky release.

It may have worked in the past, but I believe some version of the libcamera software, which Allsky uses to control RPi cameras, started enforcing the limit.

Your only choice is a different camera.

closed 2024-11-06 by EricClaeys

regnets, 2024-11-07:
@EricClaeys thanks a lot, thats good news. I already ordered a hq camera. Now this issue is documented for everyone in the future who got the same issue.

EricClaeys, 2024-11-07:
I updated the README.md file in the next Allsky release to indicate that sensor is "NOT RECOMMENDED".