AllskyTeam/allsky #3953: [FEATURE REQUEST] Review Allsky Website Settings Documentation

markusd112, 2024-11-10:
### If this is a new feature, describe what you would like.

I am struggling a little bit with the configuration options of local and remote website and what are the correct settings.

https://htmlpreview.github.io/?https://raw.githubusercontent.com/AllskyTeam/allsky/master/html/documentation/settings/allskyWebsite.html

In the examples in the documentation a variable "${ALLSKY_WEBSITE}" is used, but the documentation doesn't explain what it is for: please add some explanation in the documentation.

For the example no. 3 (local website AND remote server website): shouldn't it be:
IMAGE_DIR="/allsky"   (instead of "/allsky" in the documentation)
WEB_IMAGE_DIR="${ALLSKY_WEBSITE}"     (instead of "" in the documentation)

I had to set the WEB_IMAGE_DIR to "${ALLSKY_WEBSITE}" to see my allsky cam image on the local raspi website. If I set it to "" as in the documentation, the example webcam image of the allsky software is displayed all the time, not my own camera image.

And another thing that I don't understand in the documentations example no. 1: IMAGE_DIR, VIDEOS_DIR, KEOGRAM_DIR and STARTRAILS_DIR are the remote directories, as described in the documentation. Why I have to set it, if I only use the local website on the raspi? Shouldn't it be WEB_IMAGE_DIR, WEB_VIDEOS_DIR, WEB_KEOGRAM_DIR and WEB_STARTRAILS_DIR  in the example no. 1 to be set?

EricClaeys, 2024-11-10:
@markusd112, you bring up very good points.
A LOT of people struggle with this - it's probably the biggest problem we see.
It took me quite a while to figure it out, so I totally re-did it in the next release.  To use a local Allsky Website, you'll simply enable it in the WebUI.  The only setting is how many days of image you want to keep (plus of course everything in the configuration.json file).  All the WEB_IMAGE_DIR and ALLSKY_WEBSITE and other stuff is gone.  There is a separate section in the WebUI for a remote Website and it's simpler as well.

The new release should be out this month.

markusd112, 2024-11-11:
@EricClaeys great, that sounds very good! Thanks for all this work!

closed 2024-11-11 by markusd112