AllskyTeam/allsky #3960: [BUG] Auto exposure is preferred over auto gain control ZWO ASI 533MC

meins1969, 2024-11-12:
### Camera model

ZWO ASI 533MC

### Operating System

Bullseye

### Allsky version

v2023.05.01_03

### Pi and amount of memory (512 MB, 1, 2, 4, or 8 GB)

RPi 4 Model B Rev 1.5 8GB

### Bug Description

Hello, I have the following problem.
I have changed my system from an RPIHQ to a ZWO camera. I don't understand why in night mode the exposure is always set to the maximum auto gain setting and the exposure time is changed to regulate the brightness. I'm used to it the other way round, where you first try to expose with a small gain, then increase the exposure time to max. auto exposure and only then slowly increase the gain. Unfortunately, in my case it's the other way, first the gain is set to the maximum and then the exposure time is reduced. What am I doing wrong?
[allsky.log](https://github.com/user-attachments/files/17721639/allsky.log)

EricClaeys, 2024-11-12:
You aren't doing anything wrong.  The RPi auto-exposure algorithm is smarter than the ZWO algorithm.  RPi does exactly what you said.  ZWO algorithm only handles exposure, so you should set the gain to a manual number.
We plan to port the RPi algorithm to ZWO, but it won't be in the next release.

meins1969, 2024-11-13:
Okay, thanks Eric, that was a quick and plausible answer. Keep it up, I really appreciate your project and your work, what you are doing is great. Thanks to everyone… Michael from Germany

closed 2024-11-13 by meins1969

meins1969, 2024-12-14:
Hello Eric, I have noticed a difference between day and night. During the night, the gain in auto mode is always set to the maximum regardless of whether the mean has already been exceeded. In day mode, however, the gain is set together with the time for the correct exposure. I have the feeling that there are differences in the algorithm here.

EricClaeys, 2024-12-14:
Are you still using auto-gain?  You shouldn't. It uses ZWO's algorithm which we have no control of and don't even know how it works.

meins1969, 2024-12-14:
Hello Eric, thank you, that's interesting information that the algorithm is directly from ZWO. I used Autogain to counteract the full moon, but I haven't been able to test it yet because the sky here in Germany has been thickly overcast for the last 8 weeks.