AllskyTeam/allsky #3981: coordonnées géographiques

CSpivert, 2024-11-25:
### If this is a new feature, describe what you would like.

Hello, I don't understand in which format I should enter the geographic coordinates. I have this error message :
"/home/pi/allsky/bin/capture_RPi: ERROR: Latitude cannot have any spaces.  You entered [ 48.26N].   (November 25, 11:55:14 am CET)	
*** /home/pi/allsky/bin/capture_RPi: ERROR: Longitude cannot have any spaces.  You entered [ 00.48W].   (November 25, 11:55:15 am CET)"
Thank you in advance for your help.

CSpivert, 2024-11-25:
> How did you enter the values?The installation script and the WebUI should
> have checked for valid values when you entered them.The error message
> explains what the problem is, and if you look close at what you entered you
> will see there is a space before the numbers.The WebUI also says how to
> enter the values and gives examples.From: CSpivert
> coordonnées géographiques (Issue #3981)If this is a new feature, describe
> to this thread.

ok thank you, I'm really new to raspberry, do I have to reinstall the app
or is there another way?

CSpivert, 2024-11-25:
sorry, I just found don't take into account the previous message, thank you
in any case

EricClaeys, 2024-11-25:
@CSpivert, welcome to Allsky!  I think you'll find it's fun once you have things working and are comfortable making changes and trying things.

How did you initially enter the values?
We try to make Allsky easy to use for new people so it helps if we can understand where they make mistakes so we can simplify that area.

Eric

CSpivert, 2024-11-26:
I had misinterpreted the coordinate format. I found the example in the
installation a little complicated, well I'm coming a long way, I'm
discovering Raspberry.