AllskyTeam/allsky #3982: [BUG] "Loading" when accessing website via tablet/smartphone

markusd112, 2024-11-27:
### Camera model

Raspi HQ

### Operating System

Buster

### Allsky version

V2023.05.01_05

### Pi and amount of memory (512 MB, 1, 2, 4, or 8 GB)

Raspi 4, 8GB

### Bug Description

When I go to my AllSky website via browser on my tablet (iPad 10) or my smartphone (Samsung Galaxy S23), the page regularly says "Loading" for about 60 seconds, apparently until the page automatically refreshes. When I do a manual reload, the camera image appears immediately. It works perfectly on a desktop computer: the camera image is always displayed immediately. It only happens with mobile devices.

[image]

EricClaeys, 2024-11-27:
Is your Website on the Pi or a remote server on the Internet?  If the later, what's the URL?

Since the delay only occurs with a mobile device, it's probably not a Allsky problem.  What mobile browser are you using?

markusd112, 2024-11-27:
A friend of mine is hosting his allsky website on an internet server too and the same problem occurs with his allsky website, when I open it with my mobile devices, so I think its not a specific problem of my installation/webserver.

EricClaeys, 2024-11-27:
@markusd112, when I went to your page with my Samsung S10 using Samsung's browser, I saw "Loading" for 30-60 seconds (didn't time it).  When I refreshed the page several times "Loading" was only there for 6 seconds.  It's possible that's because I am in the US, but I haven't seen this behavior with other European Allsky sites.  They usually show the picture as it downloads, but your whole image appeared at once.
Your "intervalSeconds" is 30, so if you initially don't see the image for 60 seconds it probably isn't related to that.  But as a test, would you please change the interval to 15 seconds and retest?  I will also then retest.

Samsung's browser used to have Developer Mode, equivalent to F12 on a PC, but I can't find it anymore.  Does the iPad have it?

markusd112, 2024-11-28:
@EricClaeys I've changed the interval to 15 seconds in the moment. Thanks for testing it.

EricClaeys, 2024-12-01:
With intervalSeconds at 15 the image showed up in about 3 -4 seconds.  Try the default of 5 seconds.

EricClaeys, 2024-12-01:
I looked at the code for the page and can't see why there would be a delay, other than typical network speed.  That's not saying there isn't a problem, just that I can't see one.

@Alex-developer, thoughts?

Alex-developer, 2024-12-01:
Which address are you having a problem with? The screenshot shows one address, partly obscured but you posted another one in the comment above.

If its a local website i.e. a .local address I can probably explain it but if its an externally hosted site then I would have to do some testing

Thanks

Alex

markusd112, 2024-12-01:
@Alex-developer It is a public hosted website. I have assigned a temporarily Domainname and wiped out the normally used one, because I don‘t want to publish it here. The problem occurs with both DNS names. It also occurs at the public allsky website of a friend of mine (other hosting provider than me: IONOS), so it doesn‘t seems to be a specific problem with my installation or my hosting provider ( all-inkl.com)

Alex-developer, 2024-12-01:
> @Alex-developer It is a public hosted website. I have assigned a temporarily Domainname and wiped out the normally used one, because I don‘t want to publish it here. The problem occurs with both DNS [...]

Thanks

We will take a look and see if we can reproduce it

Alex

markusd112, 2024-12-09:
> With intervalSeconds at 15 the image showed up in about 3 -4 seconds. Try the default of 5 seconds.

I have set it to 5 seconds today and it needs around 3-5 seconds now to display the allsky image on the mobile devices. 

I followed the recommendation in the documentation and set it to half the exposure time (30 seconds). So setting it to 5 seconds seems to be a good way getting it working, still the cause for this issue is unclear. I believe that the computing power of end devices is so great these days that this shouldn't have a massive impact on CPU usage.

EricClaeys, 2026-10-13:
@markusd112,
I just tried this again on my PC and Samsung phone, and both load in a couple seconds, which is what I'd expect for being in the US.
Is this still an issue for you?  If so, can you upgrade to v2026.10.01 and see if that helps?
Eric

markusd112, 2026-10-21:
It is still an issue here, I tried it just in the moment. I will update to the new version and give feedback. Thanks a lot.

markusd112, 2026-10-22:
The new version v2026.10.01 has the same problem here: it needs around 10-20seconds to display the allsky image, during the waiting time is "loading...." displayed. If you do a page reload in the browser, the Allsky image is displayed immediatly.

Occurs on iOS 27 with Safari and on Windows 11 Pro Firefox 158.0.1

benhartwich, 2026-11-13:
@markusd112 thanks for checking it on v2026.10.01. #5618 fixes one cause. When the page loads while it's hidden (a background or restored tab, or a page Safari preloads from the address bar), the first image is skipped, and "Loading" stays until the next check. A reload shows the image at once because the page is visible then.

That wait is at most `intervalSeconds` from the Website's configuration (5 by default). Could you tell us:

1. What `intervalSeconds` is set to in your `configuration.json` (or `remote_configuration.json` for a remote Website)?
2. Does it happen at night too, or only by day?
3. In Firefox, open the console (F12), load the page and wait for the image. Please paste the lines that appear, especially when `Read data.json …` and the first `=== … imaging starts` line come.

If your `intervalSeconds` is 5 and it still takes 10–20 s, there is a second cause, and the console lines will help find it.

EricClaeys, 2026-11-13:
@benhartwich, Mark's intervalseconds is 20.  Append `/configuration.json` to his URL and you'll see his settings.

Thanks for looking into this.  Should I merge the PR?

benhartwich, 2026-11-13:
@EricClaeys thanks, 20 s explains it: the page loaded hidden skips the first image and waits for the next check, so anywhere from 0 to 20 s, which matches the 10–20 s Mark sees, and why a reload (page visible) shows it at once. With the default 5 s it's barely noticeable, so you don't see it. Yes, please merge #5618 — it gets the image as soon as the page is shown, and also stops the update loop from dying when `data.json` loads slowly.

@markusd112 once it's in, the image should appear right away; please let us know if not.

markusd112, 2026-11-13:
intervalSeconds is set to 20sec. Thanks a lot, I will test it with the update.

EricClaeys, 2026-11-16:
@markusd112,
Have you been able to test the fix?