AllskyTeam/allsky #3998: [BUG] allsky_overlay.py failing

ckambiselis, 2024-11-30:
### Camera model

ASI676MC

### Operating System

Bookworm

### Allsky version

v2024.xx.yy

### Pi and amount of memory (512 MB, 1, 2, 4, or 8 GB)

Raspberry Pi 4 Model B Rev 1.5 (8 GB)

### Bug Description

I am trying to get overlay to work, and after trying different templates in overlay editor, the legacy overlay mode, restarts etc. I checked the log file and I'm getting this error message
`ERROR: Module allsky_overlay.py failed on line 209 - Expecting ',' delimiter: line 278 column 25 (char 7669)`
[allsky.log](https://github.com/user-attachments/files/17967142/allsky.log)
[settings.txt](https://github.com/user-attachments/files/17967143/settings.txt)

Alex-developer, 2024-11-30:
Can you please try it using v2024.12.06, v2024.xx.xx was the pre release version. If it still fails with that version can you attach the config/modules folder (zipped). Please ensure there are no API credentials in any modules before posting it here

Thanks

Alex

ckambiselis, 2024-12-01:
v2024.12.06 fixed the overlay issue.

Thanks

closed 2024-12-01 by ckambiselis