AllskyTeam/allsky #4003: Changes to editor page

Alex-developer, 2024-12-01:
### If this is a new feature, describe what you would like.

Few changes to the page

1. Validate json config in editor before saving as its possible to save invalid json
2. Remove the 'xxx saved' messages as they just build up if you save the config over and over
3. Change the save alert to something a little more friendly

EricClaeys, 2024-12-02:
@Alex-developer,
For # 2, is there a way to display a message that goes away when the user presses "Save"?  That would allow us to keep the message and not have them build up.

Thanks for entering this.

Alex-developer, 2024-12-22:
Changes made

1) Upgraded Codemirror to latest version
 - Includes line numbers
 - Adds lint option
2) Added lint option for json files (Shows errors on editor)
3) Changed message when no files available to edit
4) Validate the json data before its saved to prevent invalid config files from being saved
5) Removed the 'are you sure you want to save' prompt as its pointless
6) Added overlay when saving, the remote config save takes a little while so this gives the user some feedback
7) For not fatal errors the 'saved ok' message is removed after 2 seconds
8) When files are loaded 'XX_NEED_TO_UPDATE_XX' is auto highlighted
9) cleaned up a few minor bits of code

closed 2025-02-08 by Alex-developer