AllskyTeam/allsky #4013: I tried a new install on a clean SD card. Raspberry Pi 4 Model B with an Arducam IMX519. This was a known working camera from a previous working installation of Allsky software where I added the Arducam changes to the OS per https://docs.arducam.com/Raspberry-Pi-Camera/Native-camera/16MP-IMX519/. This installation reported that it could not find a camera.

EricClaeys, 2024-12-07:
I tried a new install on a clean SD card. Raspberry Pi 4 Model B with an Arducam IMX519. This was a known working camera from a previous working installation of Allsky software where I added the Arducam changes to the OS per https://docs.arducam.com/Raspberry-Pi-Camera/Native-camera/16MP-IMX519/. This installation reported that it could not find a camera.

Log file attached.
[install.log](https://github.com/user-attachments/files/18049715/install.log)

_Originally posted by @chicago925 in https://github.com/AllskyTeam/allsky/discussions/3986#discussioncomment-11495846_

EricClaeys, 2024-12-07:
@[EricClaeys](https://github.com/EricClaeys)
If you still have the install.log from when you got the No camera detected - installation aborted. message, please attach it. I can't really Debug without it.
If the installation sees a camera that isn't supported, it should tell you that.

---

@[chicago925](https://github.com/chicago925)
Here is the install log:

[install.log](https://github.com/user-attachments/files/18050014/install.log)

---
@[EricClaeys](https://github.com/EricClaeys)
Thanks, that helps. The second install found the camera so I don't know why it said it didn't. If you can, please rerun the installation but tell it to start over, rather than continue.
Thanks for helping fix this.

---
@[chicago925](https://github.com/chicago925)
The second install was the result of telling it to start over. Note that the only time it finds the camera is when the /boot/firmware/config.txt file contains the Arducam entry.

---
@[EricClaeys](https://github.com/EricClaeys)
Arducam's website made it sound like the imx519 entry has to be in /boot/firmware/config.txt.
The most recent install.log had two installations in it and the 2nd one was a "continuation".

~~~
Sat  7 Dec 13:38:44 CST 2024 * STARTING INSTALLATION OF v2024.12.06.       FIRST
...
Sat  7 Dec 13:39:24 CST 2024 *** ERROR: 
**********
No connected cameras were detected.
**********

Sat  7 Dec 14:25:27 CST 2024 * STARTING INSTALLATION OF v2024.12.06.   SECOND
Sat  7 Dec 14:25:42 CST 2024 * Continuing installation.  Steps already performed will be skipped.
  The last status was: No camera found
...
Sat  7 Dec 14:25:45 CST 2024 * RPi Arducam 16 MP  (imx519) camera found.

...
Sat  7 Dec 14:29:55 CST 2024 *** ERROR: 
**********
No camera detected - installation aborted.
**********
~~~
The first No connected cameras were detected error is to be expected if /boot/firmware/config.txt wasn't configured.
It's possible the 2nd time the error appeared because some setting didn't get passed from the initial install to the "continued" install.

Are you saying you told the second install to "start over", but per the log it continued? If so, that's a bug

Please attach ~/allsky/config/logs/install_status.txt.

chicago925, 2024-12-07:
I told the second install to start over. It says continued in the log so that's probably wrong.

I'm currently re-installing v2023.05.01_04  to make sure the Arducam is functioning as it was. So far it is so this is not a hardware issue. I'll retry the new installation shortly and report in this issue with all logs. The scenario will be with the Arducam entry in /boot/firmware/config.txt. As for now I would say that the new installation script definitely needs to put the Arducam entry in /boot/firmware/config.txt.

chicago925, 2024-12-07:
OK - I reimaged the OD and downloaded the new Allsky. I added the Arducam entry in /boot/firmware/config.txt but forgot to reboot before I started the installation. It errored out quickly not finding the camera. I then rebooted and restarted the install from the beginning. It went through multiple steps, indicated that it sound the camera, but errored out with "No camera detected - installation aborted.". Log files attached:
[install.log](https://github.com/user-attachments/files/18050266/install.log)
[install_status.txt](https://github.com/user-attachments/files/18050268/install_status.txt)

EricClaeys, 2024-12-08:
> I told the second install to start over. It says continued in the log so that's probably wrong.

Sigh.  I thought I fixed that.  I opened [Issue #4016](https://github.com/AllskyTeam/allsky/issues/4016) for this.

It's possible that bug impacted the first problem, and if you `cd ~/allsky; rm -fr config` then re-run the installation, it will truly start over, and might work.

EricClaeys, 2024-12-08:
It's possible the "not starting over when you tell it to" bug impacted the first problem.  Please run:
~~~
cd ~/allsky
mv config config-SAVED
~~~
 then re-run the installation, and if it fails, please attach ~/allsky/config/logs/install.log.

EricClaeys, 2024-12-08:
Also, if it fails again, please run
~~~
makeChanges.sh --debug --force --cameraTypeOnly --fromInstall --addNewSettings  cameranumber 'Camera Number' '' '0' cameramodel 'Camera Model' '' 'Arducam 16 MP' cameratype 'Camera Type' '' 'RPi'
~~~

chicago925, 2024-12-08:
Well I read your note too fast (before you edited it) and did not save the config directory before I re-ran the install. If it helps here is the result (it failed):
[install.log](https://github.com/user-attachments/files/18050286/install.log)

chicago925, 2024-12-08:
-bash: makeChanges.sh: command not found

EricClaeys, 2024-12-08:
The "not starting over when you tell it to" bug is unrelated to the "No camera detected - installation aborted." bug.

Have you rebooted since installing the new version?   If so, it should have found `makeChanges.sh`.  Either way, run
~~~
cd ~/allsky/scripts
./makeChanges.sh --debug --force --cameraTypeOnly --fromInstall --addNewSettings  cameranumber 'Camera Number' '' '0' cameramodel 'Camera Model' '' 'Arducam 16 MP' cameratype 'Camera Type' '' 'RPi'
~~~

chicago925, 2024-12-08:
Ran the above - errored out "No camera detected - installation aborted."
[install.log](https://github.com/user-attachments/files/18050313/install.log)

chicago925, 2024-12-08:
Also - per your question above I rebooted after I changed the /boot/firmware/config.txt to add the Arducam entry.

EricClaeys, 2024-12-08:
Running `makeChanges.sh` outside of the installation script won't write anything to the install.log.  Using `--debug` should have provided some debugging statements to the terminal window.  Was there anything there?

What's in `~/allsky/config/connected_cameras.txt` ?

chicago925, 2024-12-08:
[connected_cameras.txt](https://github.com/user-attachments/files/18050328/connected_cameras.txt)

chicago925, 2024-12-08:
Terminal output from makeChanges.sh:

makeChanges.sh: cameranumber: Old=[], New=[0]
makeChanges.sh: cameramodel: Old=[], New=[Arducam 16 MP]
makeChanges.sh: cameratype: Old=[], New=[RPi]
Re-creating files for cameratype RPi, cameranumber 0
Calling: capture_RPi -cmd rpicam-still -debuglevel 3  -cameranumber 0 -cameramodel 'Arducam 16 MP' -cc_file '/home/brian/allsky/config/cc.json'

EricClaeys, 2024-12-08:
connected_cameras.txt has the camera, which makes sense.  Please attach `~/allsky/config/cc.json`, then run:
~~~
cd ~/allsky/bin
./capture_RPi -cmd rpicam-still -debuglevel 5 -cameranumber 0 -cameramodel 'Arducam 16 MP' -cc_file '/home/brian/allsky/config/cc.json'
~~~

chicago925, 2024-12-08:
Well I don't see a cc.json file in the config directory. I assume that's bad?

Should I tun the script?

chicago925, 2024-12-08:
I did see a status.json file. Here it is.
[status.json](https://github.com/user-attachments/files/18050343/status.json)

EricClaeys, 2024-12-08:
yes, please run `capture_RPi`.  It should create the cc.json file.

chicago925, 2024-12-08:
Did not create the cc.json file. First I had to 'ALLSKY_HOME="/home/brian/allsky" to get to run. Once done here is the result:

brian@allsky-test:~/allsky/bin $ ./capture_RPi -cmd rpicam-still -debuglevel 5 -cameranumber 0 -cameramodel 'Arducam 16 MP' -cc_file '/home/brian/allsky/config/cc.json'
 >>> Parameter [cameranumber             ]  Value: [0]
 >>> Parameter [cameramodel              ]  Value: [Arducam 16 MP]
 >>> Parameter [cc_file                  ]  Value: [/home/brian/allsky/config/cc.json]
Line 1: [RPi	0	imx519]
  cameraType=[RPi], num=0, cameraModel=[imx519]
checking connectedCameraTypes[0], num_connectedCameraTypes=0
    p is [NOT SET], cameraType=[RPi]
  NEW TYPE [RPi], num_connectedCameraTypes=1
Connected camera types: 1, connected RPi cameras: 1
Saving sensor [imx519] from ASICameraInfoArray[4] to RPiCameras[0], ControlCapsArray[8].
capture_RPi: ERROR: camera number 0 not found.
Executing /home/brian/allsky/scripts/addMessage.sh error 'capture_RPi: ERROR: camera number 0 not found.'

EricClaeys, 2024-12-08:
Thanks for the `capture_RPi` output.  I may upload a modified file that produces more debugging output for you to try after I look through the code.

EricClaeys, 2024-12-08:
Not totally finding the Arducam 16 MB imx519 is a separate problem so I created a new [Issue #4019](https://github.com/AllskyTeam/allsky/issues/4019) for it.  Please use that Issue for further posts.

The first problem you mentioned about "no connected cameras" because `libcamera-apps` wasn't installed was correct but the message was misleading (the camera was connected).  I have a change in that will inform the user to check if libcamera-apps is installed.  That will fix this Issue, so I'm locking the Issue since all further comments on the other problems should go in their respective Issues.

Sorry you're seeing these problems, and I appreciate your willingness to help fix them.

PR #4026 (2024-12-08): Say when no RPi command was found
Fixes #4013

If none of rpicam-still, libcamera-still, and raspicam-still exist and the user has an RPi camera, they will see a message saying, "no camera detected", when the real problem is none of those commands exist.

Fix that by telling the user no commands exist.
closed 2024-12-08 by Alex-developer