AllskyTeam/allsky #4019: Arducam imx519 isn't totally found

EricClaeys, 2024-12-08:
_Originally posted by @chicago925 in https://github.com/AllskyTeam/allsky/issues/4013#issuecomment-2525348600_

OK - I reimaged the OS and downloaded the new Allsky. I added the Arducam entry in /boot/firmware/config.txt but forgot to reboot before I started the installation. It errored out quickly not finding the camera. I then rebooted and restarted the install from the beginning. It went through multiple steps, indicated that it found the camera, but errored out with "No camera detected - installation aborted.". Log files attached:
[install.log](https://github.com/user-attachments/files/18050266/install.log)
[install_status.txt](https://github.com/user-attachments/files/18050268/install_status.txt)

---
@ricClaeys
The "not starting over when you tell it to" bug is unrelated to the "No camera detected - installation aborted." bug.

Have you rebooted since installing the new version? If so, it should have found makeChanges.sh. Either way, run
~~~
cd ~/allsky/scripts
./makeChanges.sh --debug --force --cameraTypeOnly --fromInstall --addNewSettings  cameranumber 'Camera Number' '' '0' cameramodel 'Camera Model' '' 'Arducam 16 MP' cameratype 'Camera Type' '' 'RPi'
~~~

---
@chicago925
Ran the above - errored out "No camera detected - installation aborted."
[install.log](https://github.com/user-attachments/files/18050313/install.log)

Also - per your question above I rebooted after I changed the /boot/firmware/config.txt to add the Arducam entry.

---
@EricClaeys
(https://github.com/AllskyTeam/allsky/issues/4013#issuecomment-2525357201)
Running makeChanges.sh outside of the installation script won't write anything to the install.log. Using --debug should have provided some debugging statements to the terminal window. Was there anything there?

What's in ~/allsky/config/connected_cameras.txt ?

---
@chicago925
[connected_cameras.txt](https://github.com/user-attachments/files/18050328/connected_cameras.txt)

Terminal output from makeChanges.sh:
~~~
makeChanges.sh: cameranumber: Old=[], New=[0]
makeChanges.sh: cameramodel: Old=[], New=[Arducam 16 MP]
makeChanges.sh: cameratype: Old=[], New=[RPi]
Re-creating files for cameratype RPi, cameranumber 0
Calling: capture_RPi -cmd rpicam-still -debuglevel 3 -cameranumber 0 -cameramodel 'Arducam 16 MP' -cc_file '/home/brian/allsky/config/cc.json'
~~~

---
@EricClaeys
`connected_cameras.txt` has the camera, which makes sense. Please attach `~/allsky/config/cc.json`, then run:
~~~
cd ~/allsky/bin
./capture_RPi -cmd rpicam-still -debuglevel 5 -cameranumber 0 -cameramodel 'Arducam 16 MP' -cc_file '/home/brian/allsky/config/cc.json'
~~~

---
@[chicago925](https://github.com/chicago925)
Well, I don't see a cc.json file in the config directory. I assume that's bad?

Should I run the script?

I did see a status.json file. Here it is.
[status.json](https://github.com/user-attachments/files/18050343/status.json)

---
@EricClaeys
(https://github.com/AllskyTeam/allsky/issues/4013#issuecomment-2525369338)
yes, please run capture_RPi. It should create the cc.json file.

---
@[chicago925](https://github.com/chicago925)
Did not create the cc.json file. First I had to 'ALLSKY_HOME="/home/brian/allsky" to get to run. Once done here is the result:
~~~
brian@allsky-test:~/allsky/bin $ ./capture_RPi -cmd rpicam-still -debuglevel 5 -cameranumber 0 -cameramodel 'Arducam 16 MP' -cc_file '/home/brian/allsky/config/cc.json'

Parameter [cameranumber ] Value: [0]
Parameter [cameramodel ] Value: [Arducam 16 MP]
Parameter [cc_file ] Value: [/home/brian/allsky/config/cc.json]
Line 1: [RPi 0 imx519]
cameraType=[RPi], num=0, cameraModel=[imx519]
checking connectedCameraTypes[0], num_connectedCameraTypes=0
p is [NOT SET], cameraType=[RPi]
NEW TYPE [RPi], num_connectedCameraTypes=1
Connected camera types: 1, connected RPi cameras: 1
Saving sensor [imx519] from ASICameraInfoArray[4] to RPiCameras[0], ControlCapsArray[8].
capture_RPi: ERROR: camera number 0 not found.
Executing /home/brian/allsky/scripts/addMessage.sh error 'capture_RPi: ERROR: camera number 0 not found.'
~~~

---
@[EricClaeys](https://github.com/EricClaeys)
Thanks for the capture_RPi output. I may upload a modified file that produces more debugging output for you to try after I look through the code.

EricClaeys, 2024-12-08:
@chicago925, changes to /boot/firmware/config.txt are up to the user.  We have to assume the user read the camera manual or web page and know how to physically set it up as well as handle any software changes.  Some of the Arducam cameras require changing config.txt, and some don't, and I think it's related to what OS is used.

EricClaeys, 2024-12-08:
@chicago925, please try the attached file.  To do so:
~~~
cd ~/allsky/src
mv ASI_functions.cpp ASI_functions.cp.SAVED
# copy the attached file here
mv ASI_functions.txt ASI_functions.cpp
sudo systemctl stop allsky
make
cp capture_??? ../bin
~~~
then rerun the new installation.

This new file spells the camera brand "Arducam" ; it used to be "ArduCam" for 3 cameras, including the 519.

Thanks.

chicago925, 2024-12-09:
Success! I was able to complete the installation of the new version. I'm not sure if the above change was the answer as I installed the latest versions of the Arducam bash scripts and libcamera_dev and libcamera_app prior to doing the above. You may be able to answer that. In any case the imx519 is detected and is working and the install completed successfully.

Regarding the need for user getting the Arducam config,.txt changes made prior to the install I agree but would also add that there should be a note in the Pre-Installation documentation that the camera being used should be physically connected to the Raspberry Pi, libcamera must be enabled, and any other required changes needed to support the proper operation of the camera should be completed. It is better to state this up front even though you are adding checks for this. I'll open an enhancement request for this doc change.

Thanks for all your help - much appreciated!

EricClaeys, 2024-12-09:
That's good news.  Was the successful installation of v2024.12.06 with just the changed file I posted, or were there other changes?

closed 2024-12-09 by EricClaeys