AllskyTeam/allsky #4022: Just upgraded, all went smooth. Had to clear command line parameters for Keograms (as they are now in the UI) and do some cleanup of remote upload settings I had but never used.

EricClaeys, 2024-12-08:
_Originally posted by @chvvkumar in https://github.com/AllskyTeam/allsky/discussions/3986#discussioncomment-11491402_

Just upgraded, all went smooth. Had to clear command line parameters for Keograms (as they are now in the UI) and do some cleanup of remote upload settings I had but never used. 

Thank you all for the great work!

---
@EricClaeys
@chvvkumar, Kumar, thanks for the feedback.
Was it hard to understand what to do with the Keogram parameters you cleared? Was the message that the install.sh script produced clear?

---
@chvvkumar
The script message was clear as is the documentation calling out a fresh install is recommended due to the number of changes. I YOLO'd it as I am rebuilding my camera tomorrow so not a big issue if something went wrong. It was not bad at all.

Only thing I noticed is the issues and solutions that show up at the top on the UI right after the upgrade are a bit hard to read. This is due to the fact that in my case there were multiple warnings for having FTP upload enabled without any username or password defined. And the individual issues and instructions to fix them are grouped together and hard to tell each issue apart.

But that is a minor case of adding a blank line after each issue and solution.

One other thing I noticed was the image crop settings did not carry over. I re added the crop pixel amounts under 'Post Capture Settings' and that was the only thing I noticed so far.

---
@EricClaeys
Thanks again for the feedback. By any chance do you have a screenshot of the hard-to-read messages?

Since the cropping settings totally changed, you should have seen this message during installation:

The way to specify cropping images has changed.  You need to reenter your crop settings.
Specify the amount to crop from the top, right, bottom, and left.
Did you not see it?

I would like to have automatically carried over the old crop settings but realized it would have been pretty difficult to do and very error prone, since the old settings depended on the sensor size. So I decided to just have the user manually reset them.

---
@chvvkumar
I somehow missed that warning in the script, but it was not a big task to re-apply the crop. As for the messages, I unfortunately do not have a screenshot of the screen.

This is at the top of the page where warnings show up, I will try to recreate the warnings and see if I can bring them back.

In the meantime, here is my install.log and checkAllsky.sh output:

install.log

pi@allskypi5:~/allsky/scripts $ ./checkAllsky.sh
No issues found.

EricClaeys, 2024-12-08:
@chvvkumar,
I assume you fixed what the hard-to-read messages suggested, and that's why manually running `checkAllsky.sh` didn't find anything?

The first time you go to the WebUI after an Allsky installation, it runs `checkAllsky.sh` and displays the output.  I agree the output is hard to read.  Here's what I got:

[image]

Since I can reproduce there's no need for you to try.  I'll fix this.

closed 2024-12-08 by Alex-developer

chvvkumar, 2024-12-08:
> @chvvkumar, I assume you fixed what the hard-to-read messages suggested, and that's why manually running `checkAllsky.sh` didn't find anything?
>
> The first time you go to the WebUI after an Allsky [...]

That is exactly it!

And yes, I fixed all the errors (not many to be honest, just the FTP settings and Keogram extra params).

This is where I think an extra newline would make telling apart the errors easier:

[image]

EricClaeys, 2024-12-09:
If you re-get the new version again, it has newlines, per your suggestion.

PR #4023 (2024-12-08): Update checkAllsky.sh: Separate messages when in WebUI
When checkAllsky.sh output is displayed in the WebUI there is no space between messages so it's hard to read. Fix by adding a blank line between messages.

Fixes #4022