AllskyTeam/allsky #4028: [BUG] time lapse creation problem after upgrade

sjtaubman, 2024-12-08:
### Camera model

RPi HQ (imx477)

### Operating System

Bookworm

### Allsky version

v2024.12.06

### Pi and amount of memory (512 MB, 1, 2, 4, or 8 GB)

Pi 4 Model B Rev 1.5 (4 GB)

### Bug Description

I installed v2024.12.06 yesterday seemingly without any apparent issues (although I did need to re-install after accidentally deleting the allsky directory when I intended to delete the OLD directory).

I have since discovered that no Timelapse was created this morning and the WebUI is displaying an error: "postToMap.sh: ERROR returned while uploading map data: .   (December 08, 06:53:42 AM)". Perhaps this is an unrelated error?

I do see the following message in the allsky.log file: "*** timelapse.sh: ERROR: ffmpeg failed."

Thanks for your help with this!

### Log / configuration files

[allsky.log](https://github.com/user-attachments/files/18053128/allsky.log)
[settings.json.txt](https://github.com/user-attachments/files/18053137/settings.json.txt)

sjtaubman, 2024-12-08:
I should have mentioned that I also ran an update in the hopes eliminating the very blue images at dusk and dawn (so far this hasn't worked): sudo apt update && sudo apt full-upgrade

Not sure if this could be a contributing factor. I am getting mini-timelapse files create but there is some odd pixelation in the clouds that was not previously present.

EricClaeys, 2024-12-08:
@Alex-developer, any idea what these messages are about?  There are 360 of them in @sjtaubman's snippet of the log file.
~~~
2024-12-08T07:59:41.079204-08:00 allsky allsky[133362]: INFO: Created TensorFlow Lite XNNPACK delegate for CPU.
~~~

Alex-developer, 2024-12-08:
> @Alex-developer, any idea what these messages are about? There are 360 of them in @sjtaubman's snippet of the log file.
> 
> ```
> 2024-12-08T07:59:41.079204-08:00 allsky allsky[133362]: INFO: Created TensorFlow Lite XNNPACK delegate for CPU.
> ```

Not an error I have seen before. Core Allsky does not install Tensorflow but the Allsky_AI module does install and use it

@sjtaubman Do you have the allsky_ai module installed?

sjtaubman, 2024-12-08:
Yes, I do have the allsky_ai module installed.

EricClaeys, 2024-12-08:
@sjtaubman, did you take a look at the `Troubleshooting -> Timelapse` documentation page?  It describes the two most common problems, but in your case, those don't seem to apply.  Your log file only had:
~~~
*** timelapse.sh: ERROR: ffmpeg failed.
     Output: Processed 941 images
~~~
for the error message.
The new release didn't change anything regarding how timelapse are generated, and your settings look fine, so we'll need to figure out what happened.

* Are your mini-timelapse working?
* How many images are in `~/allsky/images/20241207`?  If you don't know how to determine the number, run: `find -type f ~/allsky/images/20241207 | wc -l`.
  * If there are more than 941 then the timelapse likely found a bad image - the Troubleshooting page has tips on that.  Any zero-length files _should_ have been deleted, but to check, run `ls -l ~/allsky/images/20241207/image* -S -r | head -20` and look for any zero-length files.  Also run `removeBadImages.sh --debug 20241207` and let me know the output.  If it found any bad images, do NOT remove them just yet.
  * If there are 941 images, run `generateForDay.sh --timelapse 20241207` to see if it still fails.  If it does, stop Allsky (`sudo systemctl stop allsky`) then run it again - it's possible those "XNNPACK delegate" messages are causing it.

For the `postToMap.sh` error, please run `postToMap.sh --force --debug` and see if it works.  If it doesn't, please attach the output to a NEW Issue so we can track it separately.

Alex-developer, 2024-12-08:
> Yes, I do have the allsky_ai module installed.

@EricClaeys That explains where they are coming from, I'll speak to the module author and see if they can be surpressed

sjtaubman, 2024-12-09:
I did check the Troubleshooting page, but that does not seem to apply here. There do appear to be just 941 files in the directory and they are all non-zero length. 

Mini-timelapses are being created, albeit there is some odd pixelation in the video when there are clouds. I have attached an example.

https://github.com/user-attachments/assets/494d2bf4-22f0-462c-ae36-440c07f674ec

I ran removeBadImages.sh --debug 20241207 and let it go for about an hour, but it appeared to hang and not produce any output.

I ran generateForDay.sh --timelapse 20241207 and got the same ffmpeg failed message:

*** timelapse.sh: ERROR: ffmpeg failed.
    Output: Processed 941 images

Links in '/home/sjtaubman/allsky/tmp/sequence-timelapse.20241207' left for debugging.
Remove them when the problem is fixed.

generateForDay.sh: Command Failed:  '/home/sjtaubman/allsky/scripts/timelapse.sh'  --output '/home/sjtaubman/allsky/images/20241207/allsky-20241207.mp4' '/home/sjtaubman/allsky/images/20241207'

I stopped Allsky and manually ran generateForDay.sh again. It ran with no errors! I am however seen the same odd pixelation effect when there are clouds that I was getting in the mini-timelapse.

sjtaubman, 2024-12-09:
> For the pixelation issue, first try increasing the mini-timelapse Bitrate. Also, did you try this?
> image.png (view on web) <https://github.com/user-attachments/assets/8d244106-70e8-41dd-b0d0-c5fd4da3b42a>
> It's on the Allsky Settings page.
>
> For the daily timelapse failure, try disabling the allsky_ai module, then rerun the timelapse while Allsky is running. I suspect it'll work.

Unfortunately, disabling allsky_ai did not work. I am still able to manually generate the timelapse with Allsky stopped entirely.  

I will try adding -level 3.1 to see if that improves video quality.  Unrelated perhaps, but I am now seeing that when I do generate at timelapse the videos no longer play on my iPad or iPhone, but do play in a browser on my computer.

EricClaeys, 2024-12-09:
Would you please try a timelapse again with Allsky running, and when it fails, run `cat ~/allsky/tmp/timelapseTMP.txt`.
Also, try running with fewer images:
~~~
cd ~/allsky/images
mkdir test
cp $( ls 20241207/image-* | head -20 ) test
generateForDay.sh --timelapse test
~~~
That will copy 20 images from the "20241207" directory to the "test" directory.
If that works, then run the `cp` command with -200, then -400, then -800 images.  I'm curious where it fails.

Re: iPad / iPhone:  Look at the rest of the screenshot, after the part circled in blue.

sjtaubman, 2024-12-10:
Running with 20 images succeeds. Running with 200 fails. When I tried to make the timelapse with 200 I got the following message:

*** timelapse.sh: ERROR: ffmpeg failed.
    Output: Processed 201 images
    [libx264 @ 0x55c429d900] height not divisible by 2 (100x75)
    Error initializing output stream 0:0 -- Error while opening encoder for output stream #0:0 - maybe incorrect parameters such as bit_rate, rate, width or height

Links in '/home/sjtaubman/allsky/tmp/sequence-timelapse.test' left for debugging.
Remove them when the problem is fixed.

generateForDay.sh: Command Failed:  '/home/sjtaubman/allsky/scripts/timelapse.sh'  --output '/home/sjtaubman/allsky/images/test/allsky-test.mp4' '/home/sjtaubman/allsky/images/test'

The timelapseTMP.txt file has the following information in it:
Processed 201 images
[libx264 @ 0x55c429d900] height not divisible by 2 (100x75)
Error initializing output stream 0:0 -- Error while opening encoder for output stream #0:0 - maybe incorrect parameters such as bit_rate, rate, width or height

EricClaeys, 2024-12-10:
Have you been checking to see if the timelapse were created, despite what the error message says?  Try
~~~
ls -l ~/allsky/images/*/*.mp4
~~~

The 100x75 is the size of a thumbnail. By any chance are your `Daily Timelapse Height` and `Width` set to 100 and 75?  Or is there a thumbnail in the "test" directory?  Run `ls -l ~/allsky/images/test` and look for a very small file.

Would you please try again with the attached file:
~~~
cd ~/allsky/scripts
mv timelapse.sh timelapse.sh.SAVED
# copy the attached file here
mv timelapse.txt timelapse.sh
chmod 755 timelapse.sh
generateForDay.sh --timelapse test
~~~

[timelapse.txt](https://github.com/user-attachments/files/18071693/timelapse.txt)

sjtaubman, 2024-12-10:
I had been checking to see if a timelapse was created, and I don't believe that it was (subsequently I have manually created the timelapse's with allsky not running so there are timelapse files in the directories now).

There is a thumbnail in the test directory. If I delete the file thumbnail-test.jpg, I am able to successfully create a timelapse from the set of 200 files that were in the test directory. So, going back to your originally suggestion, I increased the number of images in the test directory while also deleting that thumbnail file before trying to create the timelapse. At 400 images I got the following error message:

*** timelapse.sh: ERROR: ffmpeg failed.
    Output: Processed 400 images
    [libx264 @ 0x5587f2be20] frame MB size (254x190) > level limit (3600)
    [libx264 @ 0x5587f2be20] DPB size (4 frames, 193040 mbs) > level limit (0 frames, 18000 mbs)
    [libx264 @ 0x5587f2be20] MB rate (1206500) > level limit (108000)

Links in '/home/sjtaubman/allsky/tmp/sequence-timelapse.test' left for debugging.
Remove them when the problem is fixed.

generateForDay.sh: Command Failed:  '/home/sjtaubman/allsky/scripts/timelapse.sh'  --output '/home/sjtaubman/allsky/images/test/allsky-test.mp4' '/home/sjtaubman/allsky/images/test'

sjtaubman, 2024-12-10:
Also, I ran generateForDay.sh --timelapse test using the timelapse.txt file you provided. A timelapse was successfully created and I got the following output:

===== Generating Timelapse
+ [[ 0 -gt 0 ]]
+ SCALE=
++ settings .timelapsefflog
++ local DO_NULL=false
++ [[ .timelapsefflog == \-\-\n\u\l\l ]]
++ local M=timelapse.sh
++ local FIELD=.timelapsefflog
++ [[ . != \. ]]
++ local FILE=/home/sjtaubman/allsky/config/settings.json
++ [[ ! -f /home/sjtaubman/allsky/config/settings.json ]]
+++ jq -r .timelapsefflog /home/sjtaubman/allsky/config/settings.json
++ j=warning
++ [[ warning == \n\u\l\l ]]
++ echo warning
++ return 0
+ FFLOG=warning
++ settings .timelapsevcodec
++ local DO_NULL=false
++ [[ .timelapsevcodec == \-\-\n\u\l\l ]]
++ local M=timelapse.sh
++ local FIELD=.timelapsevcodec
++ [[ . != \. ]]
++ local FILE=/home/sjtaubman/allsky/config/settings.json
++ [[ ! -f /home/sjtaubman/allsky/config/settings.json ]]
+++ jq -r .timelapsevcodec /home/sjtaubman/allsky/config/settings.json
++ j=libx264
++ [[ libx264 == \n\u\l\l ]]
++ echo libx264
++ return 0
+ VCODEC=libx264
++ settings .timelapsepixfmt
++ local DO_NULL=false
++ [[ .timelapsepixfmt == \-\-\n\u\l\l ]]
++ local M=timelapse.sh
++ local FIELD=.timelapsepixfmt
++ [[ . != \. ]]
++ local FILE=/home/sjtaubman/allsky/config/settings.json
++ [[ ! -f /home/sjtaubman/allsky/config/settings.json ]]
+++ jq -r .timelapsepixfmt /home/sjtaubman/allsky/config/settings.json
++ j=yuv420p
++ [[ yuv420p == \n\u\l\l ]]
++ echo yuv420p
++ return 0
+ PIX_FMT=yuv420p
++ settings .timelapseextraparameters
++ local DO_NULL=false
++ [[ .timelapseextraparameters == \-\-\n\u\l\l ]]
++ local M=timelapse.sh
++ local FIELD=.timelapseextraparameters
++ [[ . != \. ]]
++ local FILE=/home/sjtaubman/allsky/config/settings.json
++ [[ ! -f /home/sjtaubman/allsky/config/settings.json ]]
+++ jq -r .timelapseextraparameters /home/sjtaubman/allsky/config/settings.json
++ j='-level 3.1'
++ [[ -level 3.1 == \n\u\l\l ]]
++ echo '-level 3.1'
++ return 0
+ EXTRA='-level 3.1'
++ ffmpeg -y -f image2 -loglevel warning -r 25 -i /home/sjtaubman/allsky/tmp/sequence-timelapse.test/%04d.jpg -vcodec libx264 -b:v 5000k -pix_fmt yuv420p -movflags +faststart -level 3.1 /home/sjtaubman/allsky/images/test/allsky-test.mp4
+ X='[swscaler @ 0x557d7e5000] [swscaler @ 0x557d7f1f90] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x557d7e5000] [swscaler @ 0x557d883540] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x557d7e5000] [swscaler @ 0x557d914af0] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x557d7e5000] [swscaler @ 0x557d9a60a0] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x557d7e5000] [swscaler @ 0x557da37650] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x557d7e5000] [swscaler @ 0x557d7f1f90] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x557d7e5000] [swscaler @ 0x557d8844e0] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x557d7e5000] [swscaler @ 0x557da38100] deprecated pixel format used, make sure you did set range correctly
[... 20 lines cut ...]
[swscaler @ 0x557d7e5000] [swscaler @ 0x557d9a60a0] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x557d7e5000] [swscaler @ 0x557da37650] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x557d7e5000] [swscaler @ 0x557d7f1f90] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x557d7e5000] [swscaler @ 0x557d8844e0] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x557d7e5000] [swscaler @ 0x557da38100] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x557d7e5000] [swscaler @ 0x557d915980] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x557d7e5000] [swscaler @ 0x557d9a6f30] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x557d7e5000] [swscaler @ 0x557d7f1f90] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x557d7e5000] [swscaler @ 0x557d8844e0] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x557d7e5000] [swscaler @ 0x557da38100] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x557d7e5000] [swscaler @ 0x557d915980] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x557d7e5000] [swscaler @ 0x557d9a6f30] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x557d7e5000] [swscaler @ 0x557d7f1f90] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x557d7e5000] [swscaler @ 0x557d8844e0] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x557d7e5000] [swscaler @ 0x557da38100] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x557d7e5000] [swscaler @ 0x557d915980] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x557d7e5000] [swscaler @ 0x557d9a6f30] deprecated pixel format used, make sure you did set range correctly
[libx264 @ 0x557d574e20] frame MB size (254x190) > level limit (3600)
[libx264 @ 0x557d574e20] DPB size (4 frames, 193040 mbs) > level limit (0 frames, 18000 mbs)
[libx264 @ 0x557d574e20] MB rate (1206500) > level limit (108000)
	Done

================
If you want to upload the file(s) you just created,
	execute 'generateForDay.sh --upload  --timelapse test'

EricClaeys, 2024-12-10:
Thanks for doing this.
I will need to check why it fails when the thumbnail is there since the script should be looking only for files whose names begin with `image` and the thumbnail begins with `thumbnail`.  To be safe, during testing please always remove the thumbnail image first.

When you used the file I posted was it with 20 images?  Please always use that file until we resolve this Issue.
In the file, delete this line to minimize to test output:
~~~
echo -e "RET=$?, X=${X}"	# XXXXX TESTING
~~~
Did you run with 400 images using the file I posted?  If not, please do.

I believe these messages are due to the `-level 3.1` being added:
~~~
[libx264 @ 0x5587f2be20] frame MB size (254x190) > level limit (3600)
[libx264 @ 0x5587f2be20] DPB size (4 frames, 193040 mbs) > level limit (0 frames, 18000 mbs)
[libx264 @ 0x5587f2be20] MB rate (1206500) > level limit (108000)
~~~
The timelapse succeeded so I assume these are just informatory messages.

sjtaubman, 2024-12-10:
Thank you. I appreciate the help!

I was able to generate a timelapse for last night's images by manually running generateForDay.sh this morning. However, it did not successfully run automatically this morning. I needed to execute the script myself. Not sure why this is?

I just now ran with 400 images in the test folder using the file you posted (removing the echo statement you suggested deleting) and it did generate a timelapse file and the following output:

===== Generating Timelapse
+ [[ 0 -gt 0 ]]
+ SCALE=
++ settings .timelapsefflog
++ local DO_NULL=false
++ [[ .timelapsefflog == \-\-\n\u\l\l ]]
++ local M=timelapse.sh
++ local FIELD=.timelapsefflog
++ [[ . != \. ]]
++ local FILE=/home/sjtaubman/allsky/config/settings.json
++ [[ ! -f /home/sjtaubman/allsky/config/settings.json ]]
+++ jq -r .timelapsefflog /home/sjtaubman/allsky/config/settings.json
++ j=warning
++ [[ warning == \n\u\l\l ]]
++ echo warning
++ return 0
+ FFLOG=warning
++ settings .timelapsevcodec
++ local DO_NULL=false
++ [[ .timelapsevcodec == \-\-\n\u\l\l ]]
++ local M=timelapse.sh
++ local FIELD=.timelapsevcodec
++ [[ . != \. ]]
++ local FILE=/home/sjtaubman/allsky/config/settings.json
++ [[ ! -f /home/sjtaubman/allsky/config/settings.json ]]
+++ jq -r .timelapsevcodec /home/sjtaubman/allsky/config/settings.json
++ j=libx264
++ [[ libx264 == \n\u\l\l ]]
++ echo libx264
++ return 0
+ VCODEC=libx264
++ settings .timelapsepixfmt
++ local DO_NULL=false
++ [[ .timelapsepixfmt == \-\-\n\u\l\l ]]
++ local M=timelapse.sh
++ local FIELD=.timelapsepixfmt
++ [[ . != \. ]]
++ local FILE=/home/sjtaubman/allsky/config/settings.json
++ [[ ! -f /home/sjtaubman/allsky/config/settings.json ]]
+++ jq -r .timelapsepixfmt /home/sjtaubman/allsky/config/settings.json
++ j=yuv420p
++ [[ yuv420p == \n\u\l\l ]]
++ echo yuv420p
++ return 0
+ PIX_FMT=yuv420p
++ settings .timelapseextraparameters
++ local DO_NULL=false
++ [[ .timelapseextraparameters == \-\-\n\u\l\l ]]
++ local M=timelapse.sh
++ local FIELD=.timelapseextraparameters
++ [[ . != \. ]]
++ local FILE=/home/sjtaubman/allsky/config/settings.json
++ [[ ! -f /home/sjtaubman/allsky/config/settings.json ]]
+++ jq -r .timelapseextraparameters /home/sjtaubman/allsky/config/settings.json
++ j='-level 3.1'
++ [[ -level 3.1 == \n\u\l\l ]]
++ echo '-level 3.1'
++ return 0
+ EXTRA='-level 3.1'
++ ffmpeg -y -f image2 -loglevel warning -r 25 -i /home/sjtaubman/allsky/tmp/sequence-timelapse.test/%04d.jpg -vcodec libx264 -b:v 5000k -pix_fmt yuv420p -movflags +faststart -level 3.1 /home/sjtaubman/allsky/images/test/allsky-test.mp4
+ X='[swscaler @ 0x55951ea000] [swscaler @ 0x55951f6f90] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x55951ea000] [swscaler @ 0x5595288540] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x55951ea000] [swscaler @ 0x5595319af0] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x55951ea000] [swscaler @ 0x55953ab0a0] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x55951ea000] [swscaler @ 0x559543c650] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x55951ea000] [swscaler @ 0x55951f6f90] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x55951ea000] [swscaler @ 0x55952894e0] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x55951ea000] [swscaler @ 0x559543d100] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x55951ea000] [swscaler @ 0x559531a980] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x55951ea000] [swscaler @ 0x55953abf30] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x55951ea000] [swscaler @ 0x55951f6f90] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x55951ea000] [swscaler @ 0x55952894e0] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x55951ea000] [swscaler @ 0x559543d100] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x55951ea000] [swscaler @ 0x559531a980] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x55951ea000] [swscaler @ 0x55953abf30] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x55951ea000] [swscaler @ 0x55951f6f90] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x55951ea000] [swscaler @ 0x55952894e0] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x55951ea000] [swscaler @ 0x559543d100] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x55951ea000] [swscaler @ 0x559531a980] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x55951ea000] [swscaler @ 0x55953abf30] deprecated pixel format used, make sure you did set range correctly
[libx264 @ 0x5594f79e20] frame MB size (254x190) > level limit (3600)
[libx264 @ 0x5594f79e20] DPB size (4 frames, 193040 mbs) > level limit (0 frames, 18000 mbs)
[libx264 @ 0x5594f79e20] MB rate (1206500) > level limit (108000)'
+ RET=0
+ set +x
	Done

EricClaeys, 2024-12-11:
We're making progress...
I added fix for the `thumbnail*.jpg` file being included, which it shouldn't be.

If you use this file you'll get slightly less debugging output.
[timelapse.txt](https://github.com/user-attachments/files/18087150/timelapse.txt)

Is it correct to say that the ffmpeg failure is only happening now when `generateForDay.sh` is being run automatically at the end of night?
Please look in `/var/log/allsky.log` for errors when it ran last night.  Search for `endOfNight.sh` then keep looking until you see that `endOfNight.sh` finished.   Or attach the file and I'll look.

sjtaubman, 2024-12-11:
Yes, the ffmpeg failure is now only happening when generateForDay.sh is being run automatically at the end of the night. Here is what I think is the relevant output from /var/log/allsky.log (full file attached as well):

2024-12-10T07:00:24.947104-08:00 allsky allsky[77457]: *** timelapse.sh: ERROR: ffmpeg failed.
2024-12-10T07:00:25.016628-08:00 allsky allsky[80373]:     Output: Processed 1066 images
2024-12-10T07:00:25.016845-08:00 allsky allsky[80373]:     [libx264 @ 0x55a5285e20] frame MB size (254x190) > level limit (3600)
2024-12-10T07:00:25.016924-08:00 allsky allsky[80373]:     [libx264 @ 0x55a5285e20] DPB size (4 frames, 193040 mbs) > level limit (0 frames, 18000 mbs)
2024-12-10T07:00:25.032315-08:00 allsky allsky[80373]:     [libx264 @ 0x55a5285e20] MB rate (1206500) > level limit (108000)
2024-12-10T07:00:25.032492-08:00 allsky allsky[77457]: Links in '/home/sjtaubman/allsky/tmp/sequence-timelapse.20241209' left for debugging.
2024-12-10T07:00:25.032589-08:00 allsky allsky[77457]: Remove them when the problem is fixed.
2024-12-10T07:00:25.302542-08:00 allsky allsky[77452]: generateForDay.sh: Command Failed: nice -n 15 '/home/sjtaubman/allsky/scripts/timelapse.sh'  --output '/home/sjtaubman/allsky/images/20241209/allsky-20241209.mp4' '/home/sjtaubman/allsky/images/20241209'
2024-12-10T07:00:25.716145-08:00 allsky allsky[75884]: endOfNight.sh: ===== Timelapse complete
2024-12-10T07:00:27.176776-08:00 allsky allsky[80383]: WARNING: '/home/sjtaubman/allsky/images/20241209/allsky-20241209.mp4' not found; skipping.

[allsky.log](https://github.com/user-attachments/files/18087641/allsky.log)

EricClaeys, 2024-12-11:
Good job.  Those are exactly the log records.
Sure would be nice if ffmpeg gave a useful message, huh?  We would have figured it out as soon as it happened.

I the WebUI in the Timelapse section change the `Log Level` setting to `INFO + Warnings + Errors` - hopefully it will tell us something.
Also, please make sure the latest file I posted is being used.
Talk tomorrow.

EricClaeys, 2024-12-11:
If it fails again, try setting the timelapse width and height to half the resolution of an image.  That will decrease the memory needed and the resulting file say, plus be quicker to produce.

sjtaubman, 2024-12-11:
Ok, will do. Confirming I am using the timelapse.sh file you posted and I changed the Log Level to Info + Warnings + Errors.

...I was thinking the messages didn't seem very useful, but I figured it was me? ;-)

EricClaeys, 2024-12-11:
Not you.  `ffmpeg` never gives useful messages.

sjtaubman, 2024-12-11:
The timelapse did not get produced automatically once again, and I have attached the output from the log file here. Please let me know if you need additional information.. Thank you!

[output-from-allsky.log](https://github.com/user-attachments/files/18099648/output-from-allsky.log)

EricClaeys, 2024-12-11:
The `RET=137` in the log output means the process was killed.
This is probably the "Killed" problem mentioned on the [`Troubleshooting  -> Timelapse`](https://htmlpreview.github.io/?https://raw.githubusercontent.com/AllskyTeam/allsky/master/html/documentation/troubleshooting/timelapse.html) documentation page, but without the "Killed" message.  Please follow the steps on that page.

sjtaubman, 2024-12-11:
Ok, thanks. So I am assuming I should decrease the timelapse width and height settings to 1920 by 1080 and see if this works.

I did not have this problem prior to the upgrade and it runs when I manually generate the timelapse. Is this likely because my Pi is right on the threshold of having enough memory?

sjtaubman, 2024-12-11:
I had 1920 x 1080 prior to the upgrade (not really sure why I used these numbers). I think with the new install I accepted the default to put 0 in for each the width and height.  I am going to try 2028 x 1520 and hopefully this will solve the problem tomorrow.

Is there a straightforward way to increase the swap space on the Pi? I haven't done this before and do not know how.  I realize this may not be necessary of course. 

Thank you!

sjtaubman, 2024-12-12:
Changing the width/height to 2028 x 1520 did in fact solve the problem The timelapse was created automatically this morning. Thank you for all your help!

EricClaeys, 2024-12-12:
Great!!
I will modify the code to display a better message when RET=137 is found.  It would have saved us a bunch of time.

There is a Documentation page on changing swap space.  It's pretty easy.

PR #4065 (2024-12-13): Update timelapse.sh: check for return code 137
Fixes #4028 

For some reason, ffmpeg can be killed but we don't get a "Killed" message, but the return code is 137.  Check for that.
commit 619e929c (2024-12-13): Update timelapse.sh: check for return code 137

Fixes #4028 

For some reason, ffmpeg can be killed but we don't get a "Killed" message, but the return code is 137.  Check for that.
closed 2024-12-13 by Alex-developer