AllskyTeam/allsky #4030: [BUG] Module allsky_overlay.py failed on line 209 - cannot import name '_imagingft' from 'PIL'

HenkUyttenhove, 2024-12-08:
### Camera model

IMX290

### Operating System

Bookworm

### Allsky version

v2024.12.06

### Pi and amount of memory (512 MB, 1, 2, 4, or 8 GB)

PI3 2 

### Bug Description

ERROR: Module allsky_overlay.py failed on line 209 - cannot import name '_imagingft' from 'PIL'

I tried to manually install PIL/pillow on the distribution but still didn't work.   I noticed in the logs that the PIL library was also included in the VENV environment.

So, I removed the PIL in the venv evironment and putted a symbolic link and all worked !!!
"ln -s /usr/lib/python3/dist-packages/PIL ./PIL"

Alex-developer, 2024-12-08:
Thanks I will take a look at this

Alex-developer, 2024-12-13:
I have done a few tests on this and cannot reproduce it. Was this a fresh OS install on the pi or did you upgrade to 06 from a previous version?

HenkUyttenhove, 2024-12-14:
It was a new install.  Some days later, I build another Allsky camera and followed the same steps and didn't ran into this problem. I assumed the problem was already solved in code.
If problem can't be reproduced, looks like something went wrong during unpacking or installation.  Ticket can be closed as I also couldn't reproduct.  thx

closed 2024-12-14 by HenkUyttenhove