AllskyTeam/allsky #4031: [BUG] Upgrade fails

laheller, 2024-12-08:
### Camera model

ZWO ASI120MC-S

### Operating System

Bookworm

### Allsky version

v2023.05.01_05

### Pi and amount of memory (512 MB, 1, 2, 4, or 8 GB)

Pi 4B, 8GB

### Bug Description

I am trying to upgrade the current allsky installation to the new version 2024.12.06
Following the official documentation on upgrade.

During upgrade I received the following error message:

> *** ERROR:
> **********
> Unable to save camera capabilities.
> **********
> 
> 
> *** ERROR:
> **********
> 
> ERROR: capture_ZWO: ERROR: Currently connected 'ZWO ASI120MC-S' camera not found in '/home/laheller/allsky/config/connected_cameras.txt'.
> Unable to create cc file '/home/laheller/allsky/config/cc.json'.
> **********
> 

What I did before upgrade:
sudo systemctl stop allsky
sudo systemctl stop lighttpd.service
mv ~/allsky ~/allsky-OLD

Then simple followed the install guide.

Any idea, how to proceed?

EricClaeys, 2024-12-08:
@laheller, please use the attached file:
~~~
cd ~/allsky/scripts
mv functions.sh functions.sh.SAVE
# copy the attached file to this directory
mv functions.txt functions.sh
chmod 755 functions.sh
~~~
Then retry the installation.

Someone else recently had the same problem, but their (really old) ASI120 camera was called `ZWO ASI120MC-S ZWOptical company` in USB, but querying the camera itself returned `ZWO ASI120MC-S`.

Is your camera old?

[functions.txt](https://github.com/user-attachments/files/18054384/functions.txt)

laheller, 2024-12-09:
@EricClaeys 

With the replaced functions.sh I received the same error on upgrade:

> *** ERROR:
> **********
> Unable to save camera capabilities.
> **********
>
>
> *** ERROR:
> **********
>
> ERROR: capture_ZWO: ERROR: Currently connected 'ZWO ASI120MC-S' camera not found in [...]

My ASI 120 MC-S camera is about 6 years old.

laheller, 2024-12-09:
@EricClaeys 

Attached you can find the output as screenshot.
[image]

laheller, 2024-12-09:
@EricClaeys 

> **UPDATE: Do not do this. I looked at your output more closelyand need to make a different fix.** Sorry for the confusion.
> 
> Thanks. It appears like the `functions.sh` file I posted isn't being used. Would you please reinstall the release? If you rename the current one to `allsky-OLD` then do a new `git clone...` the new install should be fairly quick.

Don't do what? I tried to reinstall the release many times and I always ended up with the same error.
Tried with and without your fixed functions.sh file, always the same error as above.

EricClaeys, 2024-12-09:
@laheller, please try this file.  It worked on my Pi.
[functions.txt](https://github.com/user-attachments/files/18068839/functions.txt)

After I posted the last time, I discovered reinstalling a new Allsky wouldn't fix the problem.  That's what I asked you not to do.

laheller, 2024-12-10:
@EricClaeys 

OK, now it works! Thanks for the fix!

EricClaeys, 2024-12-10:
Great!  Thanks for your patience and help.

PR #4046 (2024-12-10): Check for "Planetary Camera" in ZWO model name
Fixes #4031

Another old ASI120 camera that has "ASI120 Planetary Camera" as the USB name but "ASI120" as the name reported by the camera itself.
closed 2024-12-10 by EricClaeys